code-iai/iai_kinect2 #304: Invoking "make -j8 -l8" failed when RUN 'catkin_make -DCMAKE_BUILD_TYPE="Release"

MaxxieCC, 2016-07-08:
When I run 'catkin_make -DCMAKE_BUILD_TYPE="Release"' ,it got the errors as follows. I don't know how to solve it... Any help would be appreciated!

[ 42%] Built target kinect2_registration
[ 57%] Built target kinect2_calibration
Linking CXX executable /home/maxxie/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge
[ 71%] Built target kinect2_bridge_nodelet
[ 85%] Built target kinect2_viewer
/usr/bin/ld: warning: libcudart.so.7.5, needed by /usr/local/lib/libfreenect2.so, not found (try using -rpath or -rpath-link)
collect2: error: ld returned 1 exit status
make[2]: *** [/home/maxxie/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge] Error 1
make[1]: *** [iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

wiedemeyer, 2016-07-08:
Hi,
you need to add CUDA to your environment:
```
export CUDA_HOME="/usr/local/cuda"
export LD_LIBRARY_PATH="${CUDA_HOME}/lib64:${LD_LIBRARY_PATH}" 
export PATH="${CUDA_HOME}/bin:${PATH}"
```

MaxxieCC, 2016-07-09:
Thanks for your help, it worked!

closed 2016-07-09 by MaxxieCC

Pandyaji04, 2018-06-08:
pandyaji@P320:~$ export CUDA_HOME="/usr/local/cuda"
pandyaji@P320:~$ export LD_LIBRARY_PATH="${CUDA_HOME}/lib64:${LD_LIBRARY_PATH}" pandyaji@P320:~$ export PATH="${CUDA_HOME}/bin:${PATH}"
pandyaji@P320:~$ cd ~/catkin_ws/src
pandyaji@P320:~/catkin_ws/src$ catkin_make

Base path: /home/pandyaji/catkin_ws/src
The specified source space "/home/pandyaji/catkin_ws/src/src" does not exist
pandyaji@P320:~/catkin_ws/src$ cd ..
pandyaji@P320:~/catkin_ws$ catkin_make
Base path: /home/pandyaji/catkin_ws
Source space: /home/pandyaji/catkin_ws/src
Build space: /home/pandyaji/catkin_ws/build
Devel space: /home/pandyaji/catkin_ws/devel
Install space: /home/pandyaji/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/pandyaji/catkin_ws/build"
####
####
#### Running command: "make -j8 -l8" in "/home/pandyaji/catkin_ws/build"
####
[  0%] Built target trajectory_msgs_generate_messages_py
[  0%] Built target _industrial_msgs_generate_messages_check_deps_GetRobotInfo
[  0%] Built target _industrial_msgs_generate_messages_check_deps_RobotStatus
[  0%] Built target _industrial_msgs_generate_messages_check_deps_ServiceReturnCode
[  0%] Built target _industrial_msgs_generate_messages_check_deps_CmdJointTrajectory
[  0%] Built target _industrial_msgs_generate_messages_check_deps_TriState
[  0%] Built target _industrial_msgs_generate_messages_check_deps_RobotMode
[  0%] Built target _industrial_msgs_generate_messages_check_deps_StartMotion
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target std_msgs_generate_messages_eus
[  0%] Built target _industrial_msgs_generate_messages_check_deps_SetRemoteLoggerLevel
[  0%] Built target trajectory_msgs_generate_messages_eus
[  0%] Built target trajectory_msgs_generate_messages_lisp
[  0%] Built target _industrial_msgs_generate_messages_check_deps_SetDrivePower
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] Built target _industrial_msgs_generate_messages_check_deps_StopMotion
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target _industrial_msgs_generate_messages_check_deps_DebugLevel
[  0%] Built target _industrial_msgs_generate_messages_check_deps_DeviceInfo
[  0%] Built target trajectory_msgs_generate_messages_cpp
[  0%] Built target geometry_msgs_generate_messages_nodejs
[  0%] Built target trajectory_msgs_generate_messages_nodejs
[  0%] Built target std_msgs_generate_messages_nodejs
[  0%] Built target geometry_msgs_generate_messages_eus
[  0%] Built target geometry_msgs_generate_messages_cpp
[  0%] Built target geometry_msgs_generate_messages_py
[  0%] Built target geometry_msgs_generate_messages_lisp
[  1%] Built target simple_message_dummy
[  1%] Built target node_example_gencfg
[  3%] Built target industrial_robot_client_dummy
[  4%] Built target industrial_utils
[  4%] Building CXX object barrett_hand_sim/barrett_hand_gazebo/CMakeFiles/gazebo_mimic_plugin.dir/src/mimic_plugin.cpp.o
[ 11%] Built target industrial_msgs_generate_messages_py
[ 12%] Built target abb_irb2400_manipulator_moveit_ikfast_plugin
[ 18%] Built target industrial_msgs_generate_messages_eus
[ 24%] Built target industrial_msgs_generate_messages_lisp
[ 24%] Built target _node_example_generate_messages_check_deps_NodeExampleData
[ 29%] Built target industrial_msgs_generate_messages_cpp
[ 35%] Built target industrial_msgs_generate_messages_nodejs
[ 37%] Built target industrial_trajectory_filters
[ 39%] Built target node_example_generate_messages_cpp
[ 39%] Built target node_example_generate_messages_eus
[ 40%] Built target node_example_generate_messages_py
[ 41%] Built target node_example_generate_messages_lisp
[ 41%] Built target node_example_generate_messages_nodejs
[ 41%] Built target industrial_msgs_generate_messages
[ 41%] Built target node_example_generate_messages
[ 42%] Built target talker
[ 43%] Built target listener
[ 56%] Built target simple_message_float64
[... 1414 lines cut ...]
       if (it != this->myDataPtr->connections.end())
              ^
/usr/include/gazebo-7/gazebo/common/Event.hh:662:9: error: request for member ‘push_back’ in ‘((gazebo::event::EventT<void(std::__cxx11::basic_string<char>, std::__cxx11::basic_string<char>)>*)this)->gazebo::event::EventT<void(std::__cxx11::basic_string<char>, std::__cxx11::basic_string<char>)>::myDataPtr->gazebo::event::EventTPrivate<void(std::__cxx11::basic_string<char>, std::__cxx11::basic_string<char>)>::connectionsToRemove’, which is of non-class type ‘int’
         this->myDataPtr->connectionsToRemove.push_back(it);
         ^
/usr/include/gazebo-7/gazebo/common/Event.hh: In instantiation of ‘gazebo::event::ConnectionPtr gazebo::event::EventT<T>::Connect(const boost::function<Signature>&) [with T = void(const gazebo::common::UpdateInfo&); gazebo::event::ConnectionPtr = boost::shared_ptr<gazebo::event::Connection>]’:
/usr/include/gazebo-7/gazebo/common/Events.hh:139:60:   required from ‘static gazebo::event::ConnectionPtr gazebo::event::Events::ConnectWorldUpdateBegin(T) [with T = boost::_bi::bind_t<void, boost::_mfi::mf0<void, MimicPlugin>, boost::_bi::list1<boost::_bi::value<MimicPlugin*> > >; gazebo::event::ConnectionPtr = boost::shared_ptr<gazebo::event::Connection>]’
/home/pandyaji/catkin_ws/src/barrett_hand_sim/barrett_hand_gazebo/src/mimic_plugin.cpp:43:51:   required from here
/usr/include/gazebo-7/gazebo/common/Event.hh:620:11: error: ‘class gazebo::event::EventTPrivate<void(const gazebo::common::UpdateInfo&)>’ has no member named ‘connections’
[... cut ...]
barrett_hand_sim/barrett_hand_gazebo/CMakeFiles/gazebo_mimic_plugin.dir/build.make:62: recipe for target 'barrett_hand_sim/barrett_hand_gazebo/CMakeFiles/gazebo_mimic_plugin.dir/src/mimic_plugin.cpp.o' failed
make[2]: *** [barrett_hand_sim/barrett_hand_gazebo/CMakeFiles/gazebo_mimic_plugin.dir/src/mimic_plugin.cpp.o] Error 1
CMakeFiles/Makefile2:7997: recipe for target 'barrett_hand_sim/barrett_hand_gazebo/CMakeFiles/gazebo_mimic_plugin.dir/all' failed
make[1]: *** [barrett_hand_sim/barrett_hand_gazebo/CMakeFiles/gazebo_mimic_plugin.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed
pandyaji@P320:~/catkin_ws$

Pandyaji04, 2018-06-08:
any help for me.. ?i am not able execute catkin_make.. even i tried what you mentioned @wiedemeyer wiedemeyer

Ammonliu, 2020-08-23:
/opt/ros/melodic/include/ros/internal/condition_variable.h:141:5: error: reference to ‘nanoseconds’ is ambiguous
     nanoseconds d = tp.time_since_epoch();
     ^~~~~~~~~~~
In file included from /usr/include/boost/thread/pthread/timespec.hpp:20:0,
                 from /usr/include/boost/thread/pthread/condition_variable.hpp:9,
                 from /usr/include/boost/thread/condition_variable.hpp:16,
                 from /usr/include/boost/thread/condition.hpp:13,
                 from /opt/ros/melodic/include/actionlib/server/simple_action_server.h:40,
                 from /home/ammon/catkin_workspace/src/ros_kortex/kortex_driver/include/kortex_driver/non-generated/pre_computed_joint_trajectory_action_server.h:17,
                 from /home/ammon/catkin_workspace/src/ros_kortex/kortex_driver/include/kortex_driver/non-generated/kortex_arm_driver.h:56,
                 from /home/ammon/catkin_workspace/src/ros_kortex/kortex_driver/src/non-generated/driver/kortex_arm_driver.cpp:13:
/usr/include/boost/chrono/duration.hpp:198:50: note: candidates are: typedef class boost::chrono::duration<long int, boost::ratio<1, 1000000000> > boost::chrono::nanoseconds
     typedef duration<boost::int_least64_t, nano> nanoseconds;    // at least 64 bits needed
                                                  ^~~~~~~~~~~
In file included from /usr/include/c++/7/thread:38:0,
                 from /home/ammon/catkin_workspace/src/ros_kortex/kortex_driver/include/kortex_driver/non-generated/kortex_arm_driver.h:19,
                 from /home/ammon/catkin_workspace/src/ros_kortex/kortex_driver/src/non-generated/driver/kortex_arm_driver.cpp:13:
/usr/include/c++/7/chrono:597:42: note:                 typedef struct std::chrono::duration<long int, std::ratio<1, 1000000000> > std::chrono::nanoseconds
     typedef duration<int64_t, nano>      nanoseconds;
                                          ^~~~~~~~~~~
In file included from /opt/ros/melodic/include/ros/callback_queue.h:39:0,
                 from /opt/ros/melodic/include/actionlib/client/simple_action_client.h:46,
                 from /opt/ros/melodic/include/moveit/move_group_interface/move_group_interface.h:54,
                 from /home/ammon/catkin_workspace/src/ros_kortex/kortex_driver/include/kortex_driver/non-generated/kortex_arm_simulation.h:25,
                 from /home/ammon/catkin_workspace/src/ros_kortex/kortex_driver/include/kortex_driver/non-generated/kortex_arm_driver.h:59,
                 from /home/ammon/catkin_workspace/src/ros_kortex/kortex_driver/src/non-generated/driver/kortex_arm_driver.cpp:13:
/opt/ros/melodic/include/ros/internal/condition_variable.h:142:46: error: ‘d’ was not declared in this scope
     timespec ts = boost::detail::to_timespec(d);
                                              ^
ros_kortex/kortex_driver/CMakeFiles/kortex_arm_driver_implementation.dir/build.make:62: recipe for target 'ros_kortex/kortex_driver/CMakeFiles/kortex_arm_driver_implementation.dir/src/non-generated/driver/kortex_arm_driver.cpp.o' failed
make[2]: *** [ros_kortex/kortex_driver/CMakeFiles/kortex_arm_driver_implementation.dir/src/non-generated/driver/kortex_arm_driver.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:30343: recipe for target 'ros_kortex/kortex_driver/CMakeFiles/kortex_arm_driver_implementation.dir/all' failed
make[1]: *** [ros_kortex/kortex_driver/CMakeFiles/kortex_arm_driver_implementation.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

I have this two errors, anybody can help? I appreciate!