code-iai/iai_kinect2 #306: Roslaunch Error when run the second time

MaxxieCC, 2016-07-09:
Hi all,
When I run 'roslaunch kinect2_bridge kinect2_bridge.launch' and ' rosrun kinect2_viewer kinect2_viewer' after the package was built, everything is right. 

However, when I run the same command again , I got the errors as follows:

process[kinect2_bridge-3]: started with pid [4306]
[ INFO] [1468031794.309511543]: Loading nodelet /kinect2_bridge of type kinect2_bridge/kinect2_bridge_nodelet to manager kinect2 with the following remappings:
[ INFO] [1468031794.311003914]: waitForService: Service [/kinect2/load_nodelet] could not connect to host [maxxie-Z97-HD3:37898], waiting...
[kinect2_points_xyzrgb_sd-4] process has died [pid 4195, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load depth_image_proc/point_cloud_xyzrgb kinect2 rgb/camera_info:=kinect2/sd/camera_info rgb/image_rect_color:=kinect2/sd/image_color_rect depth_registered/image_rect:=kinect2/sd/image_depth_rect depth_registered/points:=kinect2/sd/points __name:=kinect2_points_xyzrgb_sd __log:=/home/maxxie/.ros/log/f342ddc8-457d-11e6-aeab-005a39d36c81/kinect2_points_xyzrgb_sd-4.log].
log file: /home/maxxie/.ros/log/f342ddc8-457d-11e6-aeab-005a39d36c81/kinect2_points_xyzrgb_sd-4*.log
[kinect2_points_xyzrgb_qhd-5] process has died [pid 4196, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load depth_image_proc/point_cloud_xyzrgb kinect2 rgb/camera_info:=kinect2/qhd/camera_info rgb/image_rect_color:=kinect2/qhd/image_color_rect depth_registered/image_rect:=kinect2/qhd/image_depth_rect depth_registered/points:=kinect2/qhd/points __name:=kinect2_points_xyzrgb_qhd __log:=/home/maxxie/.ros/log/f342ddc8-457d-11e6-aeab-005a39d36c81/kinect2_points_xyzrgb_qhd-5.log].
log file: /home/maxxie/.ros/log/f342ddc8-457d-11e6-aeab-005a39d36c81/kinect2_points_xyzrgb_qhd-5*.log
[kinect2_points_xyzrgb_hd-6] process has died [pid 4206, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load depth_image_proc/point_cloud_xyzrgb kinect2 rgb/camera_info:=kinect2/hd/camera_info rgb/image_rect_color:=kinect2/hd/image_color_rect depth_registered/image_rect:=kinect2/hd/image_depth_rect depth_registered/points:=kinect2/hd/points __name:=kinect2_points_xyzrgb_hd __log:=/home/maxxie/.ros/log/f342ddc8-457d-11e6-aeab-005a39d36c81/kinect2_points_xyzrgb_hd-6.log].
log file: /home/maxxie/.ros/log/f342ddc8-457d-11e6-aeab-005a39d36c81/kinect2_points_xyzrgb_hd-6*.log

When I compiled and built the package again, everything was right again... And then ,run the second time, the errors got as said above.

Thanks for any help!

wiedemeyer, 2016-07-11:
Did you stopped the first launch before starting the second one?

MaxxieCC, 2016-07-11:
I closed the terminal and kill the process... What else should I do to stop the first launch?

Thanks for your help!

wiedemeyer, 2016-07-11:
launch it the first time, wait until it is launched, press CTRL+C and wait until it stopped, then launch again.

closed 2016-07-28 by wiedemeyer