code-iai/iai_kinect2 #307: waitForService: Service [/kinect2/load_nodelet] is now available.

muwangchuzhong, 2016-07-10:
when I run "roslaunch kinect2_bridge kinect2_bridge.launch", I got the errors as follows:
lyz@lyz-All-Series:~/catkin_ws$ roslaunch kinect2_bridge kinect2_bridge.launch 
... logging to /home/lyz/.ros/log/3161d15a-4677-11e6-98c8-14dda955f5d5/roslaunch-lyz-All-Series-9314.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://lyz-All-Series:53012/

SUMMARY
========

PARAMETERS
 * /kinect2_bridge/base_name: kinect2
 * /kinect2_bridge/base_name_tf: kinect2
 * /kinect2_bridge/bilateral_filter: True
 * /kinect2_bridge/calib_path: /home/lyz/catkin_...
 * /kinect2_bridge/depth_device: -1
 * /kinect2_bridge/depth_method: default
 * /kinect2_bridge/edge_aware_filter: True
 * /kinect2_bridge/fps_limit: -1.0
 * /kinect2_bridge/jpeg_quality: 90
 * /kinect2_bridge/max_depth: 12.0
 * /kinect2_bridge/min_depth: 0.1
 * /kinect2_bridge/png_level: 1
 * /kinect2_bridge/publish_tf: False
 * /kinect2_bridge/queue_size: 5
 * /kinect2_bridge/reg_device: -1
 * /kinect2_bridge/reg_method: default
 * /kinect2_bridge/sensor: 
 * /kinect2_bridge/use_png: False
 * /kinect2_bridge/worker_threads: 4
 * /kinect2_points_xyzrgb_hd/queue_size: 5
 * /kinect2_points_xyzrgb_qhd/queue_size: 5
 * /kinect2_points_xyzrgb_sd/queue_size: 5
 * /rosdistro: indigo
 * /rosversion: 1.11.20

NODES
  /
    kinect2 (nodelet/nodelet)
    kinect2_bridge (nodelet/nodelet)
    kinect2_points_xyzrgb_hd (nodelet/nodelet)
    kinect2_points_xyzrgb_qhd (nodelet/nodelet)
    kinect2_points_xyzrgb_sd (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [9326]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 3161d15a-4677-11e6-98c8-14dda955f5d5
process[rosout-1]: started with pid [9340]
started core service [/rosout]
process[kinect2-2]: started with pid [9348]
process[kinect2_bridge-3]: started with pid [9358]
process[kinect2_points_xyzrgb_sd-4]: started with pid [9359]
process[kinect2_points_xyzrgb_qhd-5]: started with pid [9360]
process[kinect2_points_xyzrgb_hd-6]: started with pid [9366]
[ INFO] [1468138842.789734981]: Loading nodelet /kinect2_bridge of type kinect2_bridge/kinect2_bridge_nodelet to manager kinect2 with the following remappings:
[ INFO] [1468138842.791866438]: waitForService: Service [/kinect2/load_nodelet] has not been advertised, waiting...
[ INFO] [1468138842.801354095]: Initializing nodelet with 8 worker threads.
[ INFO] [1468138842.812447738]: waitForService: Service [/kinect2/load_nodelet] is now available.
[ERROR] [1468138842.859477119]: Failed to load nodelet [/kinect2_bridge] of type [kinect2_bridge/kinect2_bridge_nodelet] even after refreshing the cache: Failed to load library /home/lyz/catkin_ws/devel/lib//libkinect2_bridge_nodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libcudart.so.7.0: cannot open shared object file: No such file or directory)
[ERROR] [1468138842.859515034]: The error before refreshing the cache was: Failed to load library /home/lyz/catkin_ws/devel/lib//libkinect2_bridge_nodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libcudart.so.7.0: cannot open shared object file: No such file or directory)
[FATAL] [1468138842.859656627]: Failed to load nodelet '/kinect2_bridge` of type `kinect2_bridge/kinect2_bridge_nodelet` to manager `kinect2'
[FATAL] [1468138842.960630907]: Failed to load nodelet '/kinect2_points_xyzrgb_sd` of type `depth_image_proc/point_cloud_xyzrgb` to manager `kinect2'
[FATAL] [1468138842.960631512]: Failed to load nodelet '/kinect2_points_xyzrgb_qhd` of type `depth_image_proc/point_cloud_xyzrgb` to manager `kinect2'
[FATAL] [1468138842.960681669]: Failed to load nodelet '/kinect2_points_xyzrgb_hd` of type `depth_image_proc/point_cloud_xyzrgb` to manager `kinect2'
[kinect2-2] process has died [pid 9348, exit code -11, cmd /opt/ros/indigo/lib/nodelet/nodelet manager __name:=kinect2 __log:=/home/lyz/.ros/log/3161d15a-4677-11e6-98c8-14dda955f5d5/kinect2-2.log].
log file: /home/lyz/.ros/log/3161d15a-4677-11e6-98c8-14dda955f5d5/kinect2-2*.log
[kinect2_bridge-3] process has died [pid 9358, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load kinect2_bridge/kinect2_bridge_nodelet kinect2 __name:=kinect2_bridge __log:=/home/lyz/.ros/log/3161d15a-4677-11e6-98c8-14dda955f5d5/kinect2_bridge-3.log].
log file: /home/lyz/.ros/log/3161d15a-4677-11e6-98c8-14dda955f5d5/kinect2_bridge-3*.log
[kinect2_points_xyzrgb_sd-4] process has died [pid 9359, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load depth_image_proc/point_cloud_xyzrgb kinect2 rgb/camera_info:=kinect2/sd/camera_info rgb/image_rect_color:=kinect2/sd/image_color_rect depth_registered/image_rect:=kinect2/sd/image_depth_rect depth_registered/points:=kinect2/sd/points __name:=kinect2_points_xyzrgb_sd __log:=/home/lyz/.ros/log/3161d15a-4677-11e6-98c8-14dda955f5d5/kinect2_points_xyzrgb_sd-4.log].
log file: /home/lyz/.ros/log/3161d15a-4677-11e6-98c8-14dda955f5d5/kinect2_points_xyzrgb_sd-4*.log
[kinect2_points_xyzrgb_qhd-5] process has died [pid 9360, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load depth_image_proc/point_cloud_xyzrgb kinect2 rgb/camera_info:=kinect2/qhd/camera_info rgb/image_rect_color:=kinect2/qhd/image_color_rect depth_registered/image_rect:=kinect2/qhd/image_depth_rect depth_registered/points:=kinect2/qhd/points __name:=kinect2_points_xyzrgb_qhd __log:=/home/lyz/.ros/log/3161d15a-4677-11e6-98c8-14dda955f5d5/kinect2_points_xyzrgb_qhd-5.log].
log file: /home/lyz/.ros/log/3161d15a-4677-11e6-98c8-14dda955f5d5/kinect2_points_xyzrgb_qhd-5*.log
[kinect2_points_xyzrgb_hd-6] process has died [pid 9366, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load depth_image_proc/point_cloud_xyzrgb kinect2 rgb/camera_info:=kinect2/hd/camera_info rgb/image_rect_color:=kinect2/hd/image_color_rect depth_registered/image_rect:=kinect2/hd/image_depth_rect depth_registered/points:=kinect2/hd/points __name:=kinect2_points_xyzrgb_hd __log:=/home/lyz/.ros/log/3161d15a-4677-11e6-98c8-14dda955f5d5/kinect2_points_xyzrgb_hd-6.log].
log file: /home/lyz/.ros/log/3161d15a-4677-11e6-98c8-14dda955f5d5/kinect2_points_xyzrgb_hd-6*.log
[kinect2_bridge-3] restarting process
process[kinect2_bridge-3]: started with pid [9462]
[kinect2_points_xyzrgb_sd-4] restarting process
process[kinect2_points_xyzrgb_sd-4]: started with pid [9463]
[kinect2_points_xyzrgb_qhd-5] restarting process
process[kinect2_points_xyzrgb_qhd-5]: started with pid [9464]
[kinect2_points_xyzrgb_hd-6] restarting process
process[kinect2_points_xyzrgb_hd-6]: started with pid [9465]
[ INFO] [1468138843.228260177]: Loading nodelet /kinect2_bridge of type kinect2_bridge/kinect2_bridge_nodelet to manager kinect2 with the following remappings:
[ INFO] [1468138843.231950773]: waitForService: Service [/kinect2/load_nodelet] could not connect to host [lyz-All-Series:44971], waiting...

Thanks for any help!

wiedemeyer, 2016-07-11:
you need to add CUDA to your environment. Add the following lines to ~/.bashrc:
```
export CUDA_HOME="/usr/local/cuda"
export LD_LIBRARY_PATH="${CUDA_HOME}/lib64:${LD_LIBRARY_PATH}" 
export PATH="${CUDA_HOME}/bin:${PATH}"
```

closed 2016-07-28 by wiedemeyer

jasoninsh, 2017-01-08:
@wiedemeyer Thanks for the reply,I am working with ROS now when I reinstall the computer system.Nodelet stop working.So it may be a problem with the system.

QinZiwen, 2018-02-06:
My cuda environment variables no problem, But I have same problem
```
[FATAL] [1517901231.882893659]: Failed to load nodelet '/kinect2_points_xyzrgb_hd` of type `depth_image_proc/point_cloud_xyzrgb` to manager `kinect2'
[FATAL] [1517901231.882894244]: Failed to load nodelet '/kinect2_points_xyzrgb_sd` of type `depth_image_proc/point_cloud_xyzrgb` to manager `kinect2'
[FATAL] [1517901231.883005570]: Failed to load nodelet '/kinect2_points_xyzrgb_qhd` of type `depth_image_proc/point_cloud_xyzrgb` to manager `kinect2'
[FATAL] [1517901231.883287914]: Failed to load nodelet '/kinect2_bridge` of type `kinect2_bridge/kinect2_bridge_nodelet` to manager `kinect2'
[kinect2-2] process has died [pid 10096, exit code -11, cmd /opt/ros/indigo/lib/nodelet/nodelet manager __name:=kinect2 __log:=/home/turtlebot/.ros/log/47caeb8c-0b0d-11e8-a42b-c8d3ffd835e0/kinect2-2.log].
log file: /home/turtlebot/.ros/log/47caeb8c-0b0d-11e8-a42b-c8d3ffd835e0/kinect2-2*.log
[kinect2_bridge-3] process has died [pid 10101, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load kinect2_bridge/kinect2_bridge_nodelet kinect2 __name:=kinect2_bridge __log:=/home/turtlebot/.ros/log/47caeb8c-0b0d-11e8-a42b-c8d3ffd835e0/kinect2_bridge-3.log].
log file: /home/turtlebot/.ros/log/47caeb8c-0b0d-11e8-a42b-c8d3ffd835e0/kinect2_bridge-3*.log
[kinect2_points_xyzrgb_sd-4] process has died [pid 10102, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load depth_image_proc/point_cloud_xyzrgb kinect2 rgb/camera_info:=kinect2/sd/camera_info rgb/image_rect_color:=kinect2/sd/image_color_rect depth_registered/image_rect:=kinect2/sd/image_depth_rect depth_registered/points:=kinect2/sd/points __name:=kinect2_points_xyzrgb_sd __log:=/home/turtlebot/.ros/log/47caeb8c-0b0d-11e8-a42b-c8d3ffd835e0/kinect2_points_xyzrgb_sd-4.log].
log file: /home/turtlebot/.ros/log/47caeb8c-0b0d-11e8-a42b-c8d3ffd835e0/kinect2_points_xyzrgb_sd-4*.log
[kinect2_points_xyzrgb_qhd-5] process has died [pid 10103, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load depth_image_proc/point_cloud_xyzrgb kinect2 rgb/camera_info:=kinect2/qhd/camera_info rgb/image_rect_color:=kinect2/qhd/image_color_rect depth_registered/image_rect:=kinect2/qhd/image_depth_rect depth_registered/points:=kinect2/qhd/points __name:=kinect2_points_xyzrgb_qhd __log:=/home/turtlebot/.ros/log/47caeb8c-0b0d-11e8-a42b-c8d3ffd835e0/kinect2_points_xyzrgb_qhd-5.log].
log file: /home/turtlebot/.ros/log/47caeb8c-0b0d-11e8-a42b-c8d3ffd835e0/kinect2_points_xyzrgb_qhd-5*.log
[kinect2_points_xyzrgb_hd-6] process has died [pid 10111, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load depth_image_proc/point_cloud_xyzrgb kinect2 rgb/camera_info:=kinect2/hd/camera_info rgb/image_rect_color:=kinect2/hd/image_color_rect depth_registered/image_rect:=kinect2/hd/image_depth_rect depth_registered/points:=kinect2/hd/points __name:=kinect2_points_xyzrgb_hd __log:=/home/turtlebot/.ros/log/47caeb8c-0b0d-11e8-a42b-c8d3ffd835e0/kinect2_points_xyzrgb_hd-6.log].
log file: /home/turtlebot/.ros/log/47caeb8c-0b0d-11e8-a42b-c8d3ffd835e0/kinect2_points_xyzrgb_hd-6*.log
[kinect2_bridge-3] restarting process
process[kinect2_bridge-3]: started with pid [10253]
[kinect2_points_xyzrgb_sd-4] restarting process
process[kinect2_points_xyzrgb_sd-4]: started with pid [10254]
[kinect2_points_xyzrgb_qhd-5] restarting process
process[kinect2_points_xyzrgb_qhd-5]: started with pid [10255]
[kinect2_points_xyzrgb_hd-6] restarting process
[ INFO] [1517901232.086211545]: Loading nodelet /kinect2_bridge of type kinect2_bridge/kinect2_bridge_nodelet to manager kinect2 with the following remappings:
process[kinect2_points_xyzrgb_hd-6]: started with pid [10268]
[ INFO] [1517901232.092379379]: waitForService: Service [/kinect2/load_nodelet] could not connect to host [localhost:49090], waiting...
```