code-iai/iai_kinect2 #308: kinect2_bridge is not working

boyangfu, 2016-07-13:
Hey guys, I'm running my kinect on a Jetson TK1 and the kinect2_bridge is not working. Could someone help me?

(gdb) run
Starting program: /home/ubuntu/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
[New Thread 0xb0bd3230 (LWP 11116)]
[New Thread 0xb0196230 (LWP 11117)]
[New Thread 0xaf996230 (LWP 11118)]
[New Thread 0xaefff230 (LWP 11119)]
[ERROR] [registerPublisher] Failed to contact master at [localhost:11311].  Retrying...
^C
Program received signal SIGINT, Interrupt.
0xb6a39d8e in nanosleep () at ../sysdeps/unix/syscall-template.S:81
81	../sysdeps/unix/syscall-template.S: No such file or directory.
(gdb) quit
A debugging session is active.

	Inferior 1 [process 11112] will be killed.

Quit anyway? (y or n) y

wiedemeyer, 2016-07-14:
Did you tried to start a roscore? If you are new to ROS, try http://wiki.ros.org/ROS/Tutorials.

boyangfu, 2016-07-14:
Hey wiedemeyer,
Thanks for the help! Yes, I'm new to ros and your advice helped me get past the "failed to connect to master" error. However, it seems like my kinect2_bridge is still not working, could you provide some insight? I really appreciate it. Here is my new debug log:

(gdb) run
Starting program: /home/ubuntu/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
[New Thread 0xb0bd3230 (LWP 4407)]
[New Thread 0xb0196230 (LWP 4409)]
[New Thread 0xaf7ff230 (LWP 4410)]
[New Thread 0xaefff230 (LWP 4411)]
[New Thread 0xae7ff230 (LWP 4416)]
[New Thread 0xad615230 (LWP 4417)]
[ INFO] [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: default
        fps_limit: -1
       calib_path: /home/ubuntu/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: cuda
     depth_device: -1
       reg_method: opencl
       reg_device: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4
[Error] [VaapiRgbPacketProcessorImpl] vaDisplayIsValid(display) failed
[Info] [CudaDepthPacketProcessorImpl] device 0: GK20A @ 852MHz Memory 1892MB
[Info] [CudaDepthPacketProcessorImpl] selected device 0
[New Thread 0xab192230 (LWP 4442)]
[New Thread 0xa688c230 (LWP 4443)]
[New Thread 0xa608c230 (LWP 4444)]
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 7 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:19 with serial 352006534147
[Info] [Freenect2Impl] found 1 devices
[ INFO] [Kinect2Bridge::initDevice] Kinect2 devices found: 
[ INFO] [Kinect2Bridge::initDevice]   0: 352006534147 (selected)
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[ INFO] [Kinect2Bridge::initDevice] starting kinect2
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
[ INFO] [Kinect2Bridge::initDevice] device serial: 352006534147
[ INFO] [Kinect2Bridge::initDevice] device firmware: 4.0.3916.0
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] canceling rgb transfers...
[Info] [Freenect2DeviceImpl] canceling depth transfers...
[Info] [Freenect2DeviceImpl] stopped
[ WARN] [Kinect2Bridge::initCalibration] using sensor defaults for color intrinsic parameters.
[ WARN] [Kinect2Bridge::initCalibration] using sensor defaults for ir intrinsic parameters.
[ WARN] [Kinect2Bridge::initCalibration] using defaults for rotation and translation.
[ WARN] [Kinect2Bridge::initCalibration] using defaults for depth shift.
[ INFO] [DepthRegistration::New] Using OpenCL registration method!
[ INFO] [DepthRegistration::New] Using OpenCL registration method!
[ERROR] [DepthRegistrationOpenCL::init] [depth_registration_opencl.cpp](180) cl::Platform::get(&platforms) failed: -1001
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] releasing usb interfaces...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed
[ERROR] [Kinect2Bridge::start] Initialization failed!
[Thread 0xae7ff230 (LWP 4416) exited]
[Thread 0xb0196230 (LWP 4409) exited]
[Thread 0xaf7ff230 (LWP 4410) exited]
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] already closed, doing nothing
[Thread 0xa688c230 (LWP 4443) exited]
[Thread 0xa608c230 (LWP 4444) exited]
[Thread 0xad615230 (LWP 4417) exited]
[Thread 0xb0bd3230 (LWP 4407) exited]
[Thread 0xaefff230 (LWP 4411) exited]
[Thread 0xab192230 (LWP 4442) exited]
[Inferior 1 (process 4403) exited normally]

wiedemeyer, 2016-07-15:
Could it be that the Jetson TK1 does not support OpenCL. I think I heard something like that in the past. If I am correct, start it without OpenCL: `rosrun kinect2_bridge kinect2_bridge reg_method:=cpu`.

boyangfu, 2016-07-15:
Hey wiedemeyer,

I seem to be getting a similar error running it with reg_method:=cpu. With regards to OpenCL, I looked into this before and from my preliminary research it seems like there's no easy way to install it in an ARM7 architecture, so you could be right about the lack of support. Would it be possible to use iai_kinect2 without OpenCL? Thank you for your fast replies!

Warm regards,
boyangfu

ubuntu@tegra-ubuntu:~/catkin_ws/devel/lib/kinect2_bridge$ rosrun kinect2_bridge kinect2_bridge reg_method:=cpu
[ INFO] [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: default
        fps_limit: -1
       calib_path: /home/ubuntu/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: cuda
     depth_device: -1
       reg_method: opencl
       reg_device: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4
[Error] [VaapiRgbPacketProcessorImpl] vaDisplayIsValid(display) failed
[Info] [CudaDepthPacketProcessorImpl] device 0: GK20A @ 852MHz Memory 1892MB
[Info] [CudaDepthPacketProcessorImpl] selected device 0
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 7 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:6 with serial 352006534147
[Info] [Freenect2Impl] found 1 devices
[ INFO] [Kinect2Bridge::initDevice] Kinect2 devices found: 
[ INFO] [Kinect2Bridge::initDevice]   0: 352006534147 (selected)
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[ INFO] [Kinect2Bridge::initDevice] starting kinect2
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
[ INFO] [Kinect2Bridge::initDevice] device serial: 352006534147
[ INFO] [Kinect2Bridge::initDevice] device firmware: 4.0.3916.0
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] canceling rgb transfers...
[Info] [Freenect2DeviceImpl] canceling depth transfers...
[Info] [Freenect2DeviceImpl] stopped
[ WARN] [Kinect2Bridge::initCalibration] using sensor defaults for color intrinsic parameters.
[ WARN] [Kinect2Bridge::initCalibration] using sensor defaults for ir intrinsic parameters.
[ WARN] [Kinect2Bridge::initCalibration] using defaults for rotation and translation.
[ WARN] [Kinect2Bridge::initCalibration] using defaults for depth shift.
[ INFO] [DepthRegistration::New] Using OpenCL registration method!
[ INFO] [DepthRegistration::New] Using OpenCL registration method!
[ERROR] [DepthRegistrationOpenCL::init] [depth_registration_opencl.cpp](180) cl::Platform::get(&platforms) failed: -1001
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] releasing usb interfaces...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed
[ERROR] [Kinect2Bridge::start] Initialization failed!
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] already closed, doing nothing

wiedemeyer, 2016-07-15:
sorry, you have to write `_reg_method:=cpu`

boyangfu, 2016-07-15:
No problem - now it's stuck on a different step. I've waited about 5 min for "clients to connect". Thanks! 

ubuntu@tegra-ubuntu:~/catkin_ws/devel/lib/kinect2_bridge$ rosrun kinect2_bridge kinect2_bridge _reg_method:=cpu
[ INFO] [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: default
        fps_limit: -1
       calib_path: /home/ubuntu/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: cuda
     depth_device: -1
       reg_method: cpu
       reg_device: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4
[Error] [VaapiRgbPacketProcessorImpl] vaDisplayIsValid(display) failed
[Info] [CudaDepthPacketProcessorImpl] device 0: GK20A @ 852MHz Memory 1892MB
[Info] [CudaDepthPacketProcessorImpl] selected device 0
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 7 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:8 with serial 352006534147
[Info] [Freenect2Impl] found 1 devices
[ INFO] [Kinect2Bridge::initDevice] Kinect2 devices found: 
[ INFO] [Kinect2Bridge::initDevice]   0: 352006534147 (selected)
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[ INFO] [Kinect2Bridge::initDevice] starting kinect2
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
[ INFO] [Kinect2Bridge::initDevice] device serial: 352006534147
[ INFO] [Kinect2Bridge::initDevice] device firmware: 4.0.3916.0
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] canceling rgb transfers...
[Info] [Freenect2DeviceImpl] canceling depth transfers...
[Info] [Freenect2DeviceImpl] stopped
[ WARN] [Kinect2Bridge::initCalibration] using sensor defaults for color intrinsic parameters.
[ WARN] [Kinect2Bridge::initCalibration] using sensor defaults for ir intrinsic parameters.
[ WARN] [Kinect2Bridge::initCalibration] using defaults for rotation and translation.
[ WARN] [Kinect2Bridge::initCalibration] using defaults for depth shift.
[ INFO] [DepthRegistration::New] Using CPU registration method!
[ INFO] [DepthRegistration::New] Using CPU registration method!
[ INFO] [Kinect2Bridge::main] waiting for clients to connect
^C^Z
[1]+  Stopped                 rosrun kinect2_bridge kinect2_bridge _reg_method:=cpu

rigersrukaj, 2016-07-16:
Hi Wiedemeyer, thanks so much for your help - we figured out we had to run both kinect2_viewer and kinect2_bridge in different terminals and it is working now, albeit very slowly on cpu

closed 2016-07-18 by wiedemeyer

TrinhTUHH, 2017-08-22:
@rigersrukaj I see in the log these lines
```
[ WARN] [Kinect2Bridge::initCalibration] using sensor defaults for color intrinsic parameters.
[ WARN] [Kinect2Bridge::initCalibration] using sensor defaults for ir intrinsic parameters.
[ WARN] [Kinect2Bridge::initCalibration] using defaults for rotation and translation.
[ WARN] [Kinect2Bridge::initCalibration] using defaults for depth shift.
```
Does it mean the kinect_bridge do not load the calibrated files?