code-iai/iai_kinect2 #310: No depth data processed on odroid xu4

srikanths001, 2016-07-15:
This was observed on odroid xu4. The libfreenect2 was installed with all the corresponding dependencies. But since odroid does not support OpenGL processing could not open the viewer with display of the stream from kinect2. But from the output log from libfreenect2 it looks like it has been successfully able to receive the streams. Below is the output from Protonect.
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Info] [TurboJpegRgbPacketProcessor] avg. time: 77.2071ms -> ~12.9522Hz
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Info] [OpenCLDepthPacketProcessor] avg. time: 74.0777ms -> ~13.4993Hz
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
The viewer is turned off. Received 100 frames. Ctrl-C to stop.

Now, after I launch the kinect2_bridge and try to read some topic such as the sd color data, there is no output being published to the topic and this the log from the kinect2_bridge

NODES
  /
    kinect2 (nodelet/nodelet)
    kinect2_bridge (nodelet/nodelet)
    kinect2_points_xyzrgb_hd (nodelet/nodelet)
    kinect2_points_xyzrgb_qhd (nodelet/nodelet)
    kinect2_points_xyzrgb_sd (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [1791]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to ae4ba1b0-4a72-11e6-918c-7cdd90521254
process[rosout-1]: started with pid [1804]
started core service [/rosout]
process[kinect2-2]: started with pid [1807]
process[kinect2_bridge-3]: started with pid [1808]
process[kinect2_points_xyzrgb_sd-4]: started with pid [1809]
process[kinect2_points_xyzrgb_qhd-5]: started with pid [1810]
process[kinect2_points_xyzrgb_hd-6]: started with pid [1812]
[ INFO] [1468576711.211040188]: Loading nodelet /kinect2_bridge of type kinect2_bridge/kinect2_bridge_nodelet to manager kinect2 with the following remappings:
[ INFO] [1468576711.255780083]: waitForService: Service [/kinect2/load_nodelet] has not been advertised, waiting...
[ INFO] [1468576711.273860724]: Initializing nodelet with 8 worker threads.
[ INFO] [1468576711.284488750]: waitForService: Service [/kinect2/load_nodelet] is now available.
[ INFO] [1468576712.041307763]: [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: default
        fps_limit: -1
       calib_path: /home/odroid/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: default
     depth_device: -1
       reg_method: default
       reg_device: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 5
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4
[Info] [OpenCLDepthPacketProcessorImpl]  devices:
[Info] [OpenCLDepthPacketProcessorImpl]   0: Mali-T628 (GPU)[ARM]
[Info] [OpenCLDepthPacketProcessorImpl]   1: Mali-T628 (GPU)[ARM]
[Info] [OpenCLDepthPacketProcessorImpl] selected device: Mali-T628 (GPU)[ARM]
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[Warning] [OpenCLDepthPacketProcessorImpl] Your ocl-icd has deadlock bugs. Update to 2.2.4+ is recommended.
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 16 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:4 with serial 078982140147
[Info] [Freenect2Impl] found 1 devices
[ INFO] [1468576712.425850594]: [Kinect2Bridge::initDevice] Kinect2 devices found: 
[ INFO] [1468576712.426070844]: [Kinect2Bridge::initDevice]   0: 078982140147 (selected)
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[ INFO] [1468576712.618130405]: [Kinect2Bridge::initDevice] starting kinect2
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
[ INFO] [1468576713.516232760]: [Kinect2Bridge::initDevice] device serial: 078982140147
[ INFO] [1468576713.516511051]: [Kinect2Bridge::initDevice] device firmware: 4.0.3911.0
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] canceling rgb transfers...
[Info] [Freenect2DeviceImpl] canceling depth transfers...
[Info] [Freenect2DeviceImpl] stopped
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Error] [OpenCLDepthPacketProcessorImpl] kernel_filterPixelStage2 = cl::Kernel(program, "filterPixelStage2", &err): -45
[Error] [OpenCLDepthPacketProcessor] could not initialize OpenCLDepthPacketProcessor
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[ WARN] [1468576713.916792443]: [Kinect2Bridge::initCalibration] using sensor defaults for color intrinsic parameters.
[ WARN] [1468576713.916994067]: [Kinect2Bridge::initCalibration] using sensor defaults for ir intrinsic parameters.
[ WARN] [1468576713.917162401]: [Kinect2Bridge::initCalibration] using defaults for rotation and translation.
[ WARN] [1468576713.917365025]: [Kinect2Bridge::initCalibration] using defaults for depth shift.
[ INFO] [1468576714.261370247]: [DepthRegistration::New] Using OpenCL registration method!
[ INFO] [1468576714.261701121]: [DepthRegistration::New] Using OpenCL registration method!
[ INFO] [1468576714.319247457]: [DepthRegistrationOpenCL::init] devices:
[ INFO] [1468576714.319538456]: [DepthRegistrationOpenCL::init]   0: Mali-T628
[ INFO] [1468576714.319742164]: [DepthRegistrationOpenCL::init]   1: Mali-T628
[ INFO] [1468576714.320022914]: [DepthRegistrationOpenCL::init] selected device: Mali-T628
[ INFO] [1468576714.800210693]: [DepthRegistrationOpenCL::init] devices:
[ INFO] [1468576714.800506109]: [DepthRegistrationOpenCL::init]   0: Mali-T628
[ INFO] [1468576714.800679317]: [DepthRegistrationOpenCL::init]   1: Mali-T628
[ INFO] [1468576714.800842692]: [DepthRegistrationOpenCL::init] selected device: Mali-T628
[ INFO] [1468576715.435450127]: [Kinect2Bridge::main] waiting for clients to connect
[ INFO] [1468576748.275033788]: [Kinect2Bridge::callbackStatus] client connected. starting device...
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
[Info] [DepthPacketStreamParser] -1 packets were lost
[ INFO] [1468576751.822776727]: [Kinect2Bridge::main] depth processing: ~nanms (~nanHz) publishing rate: ~0Hz
[ INFO] [1468576751.825882223]: [Kinect2Bridge::main] color processing: ~23.5712ms (~42.4247Hz) publishing rate: ~10.6652Hz
[ INFO] [1468576754.828565558]: [Kinect2Bridge::main] depth processing: ~nanms (~nanHz) publishing rate: ~0Hz
[ INFO] [1468576754.829480431]: [Kinect2Bridge::main] color processing: ~16.2065ms (~61.7037Hz) publishing rate: ~10.9762Hz
[ INFO] [1468576757.834223555]: [Kinect2Bridge::main] depth processing: ~nanms (~nanHz) publishing rate: ~0Hz
[ INFO] [1468576757.838277299]: [Kinect2Bridge::main] color processing: ~15.5676ms (~64.2361Hz) publishing rate: ~11.3233Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 69.984ms -> ~14.289Hz

From the log it indicates that there is no depth data being received but there seems to be color data. But even if only color data is being processed there is still no data being published to topics like '/kinect2/sd/image_color_rect'.
Wanted to know if anyone else have faced similar issue. Is this due to an issue with OpenCL or any other dependency not being present or is an issue solely on odroid.
Would highly appreciate your help regarding the same.

srikanths001, 2016-07-15:
Apologies for the opening a non issue. It worked after starting a roscore manually and then running kinect2_bridge node individually. I earlier thought that the kinect2_bridge launch file would automatically start a roscore. But strangely when it was launched using the launch file, there was no data being published to any of the data topics. But when I launched the node individually it was working fine.
Not sure why these two are different. 
Out of curiosity it would be interesting to know if any one else have written application for kinect using odroid. Wanted to know if it is viable solution. 
Closing the issue.

closed 2016-07-15 by srikanths001

wiedemeyer, 2016-07-15:
You are correct, if there is no roscore running, normally roslaunch is starting one automatically.