code-iai/iai_kinect2 #311: Getting big Extrinsics re-projection errors after calibration with kinect2_calibration

arik1089, 2016-07-16:
Hi,

I'm running ROS with the iai_kinect2 tool, on a Jetson TK1 board. 
Specifically, I'm trying to calibrate the RGB camera and the depth camera, to create aligned RGBD point clouds. 
I followed the tutorial here: [https://github.com/mintar/iai_kinect2/tree/master/kinect2_calibration](url), while using a large chess board 7x9x0.05 in size. 

All the steps of the calibration worked as expected, the camera recognized the chess pattern and I took many pictures using the record application from different distances and orientations of the board, while using a chair to hold the board, so the images were stable. 

I received low intrinsics re-projection errors: The RGB camera error was 0.212018, the IR camera error was 0.0957678. 
But the error I received for the sync calibration (extrinsics re-projection error) was: 17.431, which is very high. I tried this process multiple times with very similar results. 
It's also visible when running the kinect2_viewer app that the cameras aren't aligned smoothly. 

Have you ever encountered difficulties with the calibration process? I guess something in my calibration process is incorrect, and I would be very thankful to hear if you have any tips or ideas why I'm getting these results. 

Regards,
Arik.

wiedemeyer, 2016-07-22:
Hi,
the intrinsic parameters are looking fine. I don't know your exact setup, so I can only guess, but for the extrinsic parameters, each pair of images from both sensors (IR and RGB) has to be taken from the exact same position. It is very important that the device is not moving, therefore is has to be mounted on a tripod. Holding it in the hand, will not work.

arik1089, 2016-07-22:
Thanks for the comment!
I use a kind of book-stand to hold the Kinect, and I put the chessboard pattern on a chair in front of it, which I move to different angles in front of the Kinect. It means that for each synced image I take, both the camera and the pattern are static. 
I still get bad synchronized re-projection error. 

Is it possible there is a problem with the camera itself? 
Or maybe there are some angles that I should emphasize which I maybe missed?
I would love to hear what angles you used during your calibration that produced the best results. 

Thanks again,
Arik.

awhampton, 2016-07-22:
I am also running Jetson TK1s and so far haven't had issues with the calibration. Your process sounds better than mine :)

I will be calibrating sixteen kinects over the next few days, though, so I will post if I see anything similar.

I did have one kinect that was malfunctioning, for whatever it's worth.

arik1089, 2016-07-23:
Well it looks like the camera definitely was part of the problem. I tried the same process with a different Kinect, and received very similar intrinsics re-projection errors, but this time I received an extrinsics re-projection error of  0.44. 
The synchronized point cloud now looks a lot better than before, but it still looks a bit un-synchronized. For example - I record a chair, and a bit of it's color is on the background of the chair. 

I'm wondering what extrinsics re-projection error you receive in your calibration process, and if you manage to receive a good synchronization from it.

awhampton, 2016-07-25:
That is similar to the extrinsics re-projection error that I see. Here is a screenshot from one of the cameras after it has been calibrated. For this one, I used only about 50 images per calibration set.

[image]

arik1089, 2016-07-29:
Looks good! 

When you said you had a camera malfunction, what were the symptoms? 
I'm wondering if big extrinsics re-projection errors is a common symptom for a malfunctioning camera...
wiedemeyer - I would be glad to hear from your experience if you encountered such an error as well.

wiedemeyer, 2016-07-29:
I think a malfunction that prevents a proper calibration is very unlikely. Most probably something in the calibration process went wrong. There are some very important things:

- The pattern has to match exactly the dimensions that you set. Some printers are not as precise as others. Measure the printed pattern carefully. Length of each field, length over multiple fields, etc.
- Placing of the pattern is very important, which less images, or many images of a similar view to the pattern you can get a good reprojection error, but a bad calibration. I normally capture around 100 - 200 images by following these steps:
  1. place the pattern near to the camera
  2. position the pattern in the top left corner of the image
  3. move it along that row while taking pictures
  4. go down one row and do step iii. again until you end up in the right bottom corner
  5. rotate the pattern so that (1) the left border is further away from the camera than the right side, (2) the other way around, (3) the top border is further away than the bottom one, (4) the other way around, (5) left top corner is pointing up and the lower right corner is pointing down. For each of those rotations go back to step ii.
  6. place the pattern further away from the camera and go to step ii. Do this for 3 different distances, near (<= 1m), middle range (> 1m, < 2.5m) and far (>= 2.5m). (The possible distances depend on the chess board size.)
- Especially for the extrinsic calibration, it is important that the sensor and the pattern are not moving while taking the images.

awhampton, 2016-07-29:
My bad kinect was not reliably powering on. I spent a long time trying to figure it out!

I don't know enough about the hardware to have an idea if your problem might have a mechanical cause. It's strange that we're using all of the same devices, though, with different results. Sorry I can't be of more help with your issue. Good luck!

Tobias-Fischer, 2016-10-18:
Hi @arik1089, can you confirm that the camera was the issue? I am also facing the problem that the reprojection error is very high, after several trials.

closed 2016-12-13 by wiedemeyer