code-iai/iai_kinect2 #314: Kinectv2 Depth Map Poblem

shivamsardana, 2016-07-27:
0
down vote
favorite
I have installed libfreenect2 and iai_kinect2 for working on PCL and ROS on my Kinect. I have been working on how to get depth map data from Kinectv2.I have been able to see and check for rostopic list and I am not able to understand how to depth data in metres rather than integers.Please help

Now the problem is, which of  topics will give me the depth map information from the kinect (kinect v2). I have echoed all the topics (using $ rostopic echo ) and all the topics publish values in [0,255] range. I am not able to understand how do I get depth map, which (ideally) should be in floating point and not in discrete 0-255 range?

Also, what are those additional rosnodes, with points_xyzrgb_ appended to their name and how do I make use of them? Do I have to write some additional program to somehow extract depth map --- if so, how?

wiedemeyer, 2016-07-28:
All the data is stored in byte arrays, so it will always print 8 bit numbers. For the actual format you have to look at the encoding.

You could for example do `rostopic echo -n 1 /kinect2/sd/image_depth | head`. This ouputs the header of the message. You will see, that the depth is of type `16UC1` which is 16 bit.

Depth measurements itself are in mm, if you want them as float in meter you have to convert them yourself, like
```
cv::Mat depth16, depthFloat;
// Setting of mats
depth16.convertTo(depthFloat, CV_32F, 0.001);
```

The points_xyzrgb nodelets are creating point clouds from the depth and color images. They should start up with the launch file. The point clouds are published under `/kinect2/.../points`

shivamsardana, 2016-08-01:
I am a naive . I do not get how to use this mat function . Can you please explain step by step how to do it? I have tried and i really want point cloud to work on . Please help in procedure so just i can get metric depth data

wiedemeyer, 2016-08-02:
I think my answer was already more explanatory than usual. The best way to learn something is probably to teach it yourself. I would recommend to read ROS, PCL and OpenCV tutorials for the beginning, also web search engines might help a lot for most problems.

closed 2016-08-09 by wiedemeyer