code-iai/iai_kinect2 #317: Jetson Tx1 libfreenect2 iai_kinect2

Abbyls, 2016-08-28:
Hi, as the OPENCL can't support the jetson, the iai_kinect2 can't use with libfreenect2  to get the depth image! Same people have researching the CUDA code replace opencl. Is there any progress? Can i use this iai_kinect code to run on the jetson tx1?

wiedemeyer, 2016-08-29:
OpenCL is iai_kinect2 and libfreenect2 only optional. You can disable it with cmake.

Abbyls, 2016-08-29:
Sorry, i'm a newer in this field, can you tell me in detail? How to disable it with cmake?

wiedemeyer, 2016-09-02:
Either read the cmake documentation or use something like ccmake.

sjdrc, 2016-09-15:
I think the issue here is that while we can use CUDA as the depth method for libfreenect2, we have to use CPU as the depth registration method on the jetson. How difficult would it be to add CUDA as a depth registration method to iai_kinect2?

vishu2287, 2016-09-15:
with jetson boards, opengl method works perfect. cpu method works but delay will be huge.

sjdrc, 2016-09-15:
I'm getting really good performance with libfreenect2, but through ros it's unusable.

Here's what I've done:
* Fresh install with L4T 21.5 + CUDA + OpenCV4Tegra
* PCL 1.8 install from source
* ROS install from source to use unofficial OpenCV and PCL

Here's the output I get from '''roslaunch kinect2_bridge kinect2_bridge.launch publish_tf:=true''':

'''
[Info] [CudaDepthPacketProcessor] avg. time: 10.9905ms -> ~90.9878Hz
[Info] [TegraJpegRgbPacketProcessor] avg. time: 15.8092ms -> ~63.2542Hz
[ INFO] [1473920579.331584986]: [Kinect2Bridge::main] depth processing: ~473.006ms (~2.11414Hz) publishing rate: ~6.66475Hz
[ INFO] [1473920579.331784319]: [Kinect2Bridge::main] color processing: ~450.753ms (~2.21851Hz) publishing rate: ~1.66619Hz
'''

Running this maxes out two of my CPU cores, and I get less than 3fps in ros while I'm getting ~90fps/~65fps for depth and rgb from libfreenect2. Running Protonect maxes uses up about 1.5 cores on the CPU.

Here are my ideas on how to improve performance:

1. Modify iai_kinect2 to use more than two cores
2. Modify iai_kinect2 to use CUDA instead of OpenCL

Does anyone else have any ideas on how to improve performance?

wiedemeyer, 2016-09-16:
Hi,
what topics are you subscribing to? HD/QHD need more computational power than SD, especially without OpenCL.

To see, what is consuming so much CPU please add some time measuring to `processIrDepth`, `processColor`, `receiveIrDepth` and `receiveColor`. I am curious what causes it.

To your modifications:
1. iai_kinect2 is using multiple threads already. You can set the maximum with the parameter `worker_threads`.
2. The only OpenCL code is for computing the HD/ QHD depth image. If you stick to SD, the code is not used at all. But if you want, you can port it to CUDA. The SD color registration is done by code from libfreenect2, which only has a CPU implementation.
3. Point clouds are created by a nodelet from the `depth_image_proc` package. This is not part of `iai_kinect2`. Best performance is achieved (especially when transmitting over network) by subscribing to depth and color images and create the point cloud in your program directly, like it is done in the `kinect2_viewer`

sjdrc, 2016-09-18:
Hi Thiemo

I'll add some time logging to those functions later today or tomorrow and post a couple of logs

1. I changed the worker_threads parameter to 4, but it still only uses two of my CPU cores. Is worker_threads for OS threading or for multithreading?

2. I'm only using the SD topics. Am I forced to use CPU processing for processing these feeds?

3. Thanks, I'll give that a try

closed 2016-11-03 by wiedemeyer