code-iai/iai_kinect2 #318: The kinect2_bridge depth processing time to long?

Abbyls, 2016-08-31:
When i have the roslaunch kinect2_bridge kinect2_bridge.launch and rosrun kinect2_viewer kinect2_viewer kinect2 sd image I can get the image view, but the depth and rgb process time is too long to make the kinect2 applicating. I was used in the TX1.

[Info] [TurboJpegRgbPacketProcessor] avg. time: 45.6985ms -> ~21.8826Hz
[ INFO] [1472604278.047575423]: [Kinect2Bridge::main] depth processing: ~97.1099ms (~10.2976Hz) publishing rate: ~14.3159Hz
[ INFO] [1472604278.047706828]: [Kinect2Bridge::main] color processing: ~30.1297ms (~33.1898Hz) publishing rate: ~10.9866Hz
[Info] [DepthPacketStreamParser] 14 packets were lost
[Info] [DepthPacketStreamParser] 16 packets were lost
[Info] [CudaDepthPacketProcessor] avg. time: 5.45824ms -> ~183.209Hz
[ INFO] [1472604281.054030994]: [Kinect2Bridge::main] depth processing: ~92.5956ms (~10.7997Hz) publishing rate: ~14.6353Hz
[ INFO] [1472604281.054230055]: [Kinect2Bridge::main] color processing: ~31.247ms (~32.003Hz) publishing rate: ~11.6417Hz
You can see the depth processing freq is only 10 hz. 
and when i run  the ./Protonect i can see the four images, the debug as following:

[Info] [OpenGLDepthPacketProcessor] avg. time: 13.5973ms -> ~73.5438Hz
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Info] [TurboJpegRgbPacketProcessor] avg. time: 40.4644ms -> ~24.7131Hz

i make the kinect2 connected with the TX1's usb3.0  port.

wiedemeyer, 2016-08-31:
I guess the reason is that the CPU is under full load. Processing the SD RGB image is using much CPU. The image has to be registered, than it has to be flipped and at the end the color format has to be converted. The depth image has to be undistorted, flipped and converted too. This is probably to much for the TX1.

Abbyls, 2016-08-31:
@wiedemeyer 
Yes, i think you are right. As i used a PCI-E to connect the kinect2 usb, i can't get the image, and crashed! I think with the increase of the bandwidth the CPU load increasing! Is it?
Can i use the package 'kinect2_registraction' to registered?
Can i use the package 'kinect2_calibration' to left and right flipped?
and i wonder what do "color format has to be converted" mean?

wiedemeyer, 2016-09-06:
If you are having problems with the USB controller, than it is probably a driver issue and not related to iai_kinect2.
There are READMEs in each package folder. You can read those to understand what they are for.
Color format conversion: RGBA to BGR.

Abbyls, 2016-09-06:
@wiedemeyer 
sorry i mislead you, it's not the USB controller's problem, the problem is my CPU load are around 90%, and the frame rates is around 7 hz. as followed:
`[ INFO] [1473146239.715266524]: [Kinect2Bridge::main] depth processing: ~204.89ms (~4.88067Hz) publishing rate: ~14.977Hz
[ INFO] [1473146239.715418450]: [Kinect2Bridge::main] color processing: ~154.081ms (~6.49008Hz) publishing rate: ~5.65796Hz
[Info] [TegraJpegRgbPacketProcessor] avg. time: 4.99351ms -> ~200.26Hz
[Info] [CudaDepthPacketProcessor] avg. time: 5.99157ms -> ~166.901Hz
[ INFO] [1473146242.723445639]: [Kinect2Bridge::main] depth processing: ~226.668ms (~4.41174Hz) publishing rate: ~13.9619Hz
[ INFO] [1473146242.723579493]: [Kinect2Bridge::main] color processing: ~184.339ms (~5.42477Hz) publishing rate: ~4.65398Hz
[Info] [TegraJpegRgbPacketProcessor] avg. time: 4.87558ms -> ~205.104Hz
[Info] [DepthPacketStreamParser] 1 packets were lost
[Info] [CudaDepthPacketProcessor] avg. time: 5.88336ms -> ~169.971Hz`

As i know in recent days, is the opencl in registration can't support the jetson, results in the cpu as the registration method, so the cpu are overload. 
you suggested me to "The image has to be registered, than it has to be flipped and at the end the color format has to be converted. The depth image has to be undistorted, flipped and converted too. This is probably to much for the TX1." 
I wonder the image didn't registered? why i need registered? i remember there is a reg_method=cpu | opencl to get registration. what do you mean registered? 
and in the installLibfreenect2 i had the color format conversion.
So, i'm confused. what should i do to decrease the cpu's load and increase the frame rate, otherwise, i can't do my work further.

wiedemeyer, 2016-09-06:
if you are subscribing to the SD color image, than is needs to get registered to the depth image. This is done using the registration code provided by libfreenect2, there is no OpenCL/CUDA implementation for it. The kinect2_registration package and the `reg_method` parameter are only used for the HD/QHD depth to color registration, not for the color to depth registration used for the SD images.
The images need to be flipped because in the images from libfreenect2 left and right are flipped.
iai_kinect2 is not optimized to the lowest CPU load. It tries to reduce the work needed on the receiving nodes and to minimize the needed bandwidth, so that it can transfer up to 30 FPS of HD images over gigabit network.
If the frame rate is so important, simply use stronger hardware, or optimize the code to your needs.

Abbyls, 2016-09-06:
Thank for your petiently answer and your opensource code! i will have a try.

closed 2016-11-03 by wiedemeyer