code-iai/iai_kinect2 #319: questions about Kinect2 calibration tools

Severn-Vergil, 2016-08-31:
Hi, @wiedemeyer 

When I'm using Kinect2 calibration tools, I notice that sometimes I record "bad patterns" like this:
[bad pattern](http://images2015.cnblogs.com/blog/604597/201608/604597-20160831094955433-504340770.png)
I've read your instructions about proper calibration process, #311. I feel it hard to record perfectly as I have to select every pattern in its correct corner points,  and the complete calibration process usually cost 3-4 hours. 
I tried to use the [camera_calibration](http://wiki.ros.org/camera_calibration) package to choose pictures automatically and navoid the pattern checking problem, however I found the rosservice provided by Kinect2 bridge not support the camera_calibration package.
I would appreciate if you could provide me some suggestions about efficiently using kinect2_calibration, thank you in advance!

wiedemeyer, 2016-08-31:
Hi,
you can just delete the files where the detection was wrong and they will not be used for calibration. Just write down the frame number for miss detected frames while recording.
I think you can still use the camera calibration package. There is just no service that sets the parameters on the fly. Just convert the results into the yaml format that kinect2_bridge needs and store them in `kinect2_bridge\data\<serial_number>\`.
You are right, a proper calibration takes time. Suggestions you can find in the README or in many closed issues related to calibration.

Severn-Vergil, 2016-09-01:
Thank you! After I add '--no-service-check' parameter with camera_calibration, it works! And I just found that without given camera params, getColorCameraParams from libfreenect2 would be called in kinect2_bridge.cpp, as the 'raw' RGB-D stream can be called for calibration. Hope it would be useful for others.

closed 2016-09-01 by Severn-Vergil