code-iai/iai_kinect2 #321: Kinect2 distance

roliveiradoteu, 2016-09-03:
Trying to calibrate my kinect but found the squares are extremely difficult to find beyond say 2.5 meters, specially in the IR camera. 

I have tried the 9x6 25mm and custom size 5x7 42mm, using a sheet of glass. I am very confident on how flat it is and in the quality of the print :|

What are the distances you used? 

Cheers

//ro

roliveiradoteu, 2016-09-04:
Is it possible to redo just the sync step, considering the position will be different (I moved the camera)? 

What puzzles me is that the colour image is 16x9 and the ir is 4x3 meaning there is a substancial part of the colour image that is not covered by the chessboard, during this step i mean.

wiedemeyer, 2016-09-06:
I usually used distances between 0.6 and 2.5 meters, near, middle and far. To go further away, you need a larger board, but then you can not be as close.
The individual calibrations are looking fine. For the extrinsics, it is important that the sensor and the board **do not move** at all, because the images are not taken at the same time. Holding on of them in a hand is not working well.
Yes you can remove all `*_sync_*` files and record new ones.
The color image is much wider and the depth image is higher, but there is an overlapping part in the center area. In the images at https://github.com/code-iai/iai_kinect2#screenshots you should see where the overlap is. For the sync calibration you have to place the board so that it is detected on both images, so that both windows are showing the colored detection overlay.

roliveiradoteu, 2016-09-12:
Thanks Thiemo.
I redid the calibration using more extreme angles and got around 0.21 px, I wasn't tilting the calibration pattern by the required amount for good geometry to be detected. 

Regards,

roliveiradoteu, 2017-02-18:
@wiedemeyer 

The coloured depth image shows a black strip in the upper part, is this in indication of poor calibration ?

It is noticeable as well that the size of this black edge is greater in the sd image, but appears similar in both the qhd and hd images. 

[image]

[image]

wiedemeyer, 2017-02-28:
It could come from the rectification: https://github.com/code-iai/iai_kinect2#screenshots
There you can see a black pixels at the border of the color image. And because the depth image has a different aspect ratio there are valid depth pixels for those black pixels.

roliveiradoteu, 2017-03-08:
@wiedemeyer Given your statement in > https://github.com/code-iai/iai_kinect2/issues/349#issuecomment-270887609 does it make sense to use only the IR and RGD and pose calibration information? This would mean not using the calib_depth.yaml?

wiedemeyer, 2017-03-08:
The depth calibration has nothing to do with the black areas at top and bottom.

roliveiradoteu, 2017-03-08:
yes i appreciate that is due to the rectification. 

i am just wondering if I can use the default depth calibration and still use the intrinsic and extrinsic information from iai_kinect2.

wiedemeyer, 2017-03-08:
Yes you can. Just set the value in that file to 0.

closed 2017-04-23 by roliveiradoteu