code-iai/iai_kinect2 #326: Depth data from Kinect v2 in ROS

patrchri, 2016-10-19:
Hello,

I am using a Kinect v2 for colour detection and even though I have managed to detect an object using its colour from the color topics provided by your tools, I cannot get correct depth data from the depth image's topic. I would like to apologize in advance for a question not directly connected to your tools, but I cannot find an answer somewhere else about the Kinect v2. I have posted a similar question [here](http://answers.ros.org/question/245652/convert-raw-depth-data-from-depth-image-to-meters-kinect-v2/). I am generally new to vision and working with depth sensors, so please bare with me.

To be more specific, I have calibrated the sensor according to your instructions and when I am detecting an object whose centre of mass is at point (x,y) in the colour image, I am trying to use this point to go to the depth image and find the depth. I am using mainly the qhd resolution's topics.

I would like to ask the following;

1. Isn't the calibration a job that needs to be done from the factory, why is it essential to be done from us? I can understand the calibration of the intrinsic parameters of the rgb and ir camera (to be used for undistortion for example), but I cannot understand why I need to calibrate the depth shift for example. Isn't this something standard? When I am using the same sensor for gaming, why calibration is not essential?
2. After the calibration, I got a depth shift equal to 2.1052533694078949e+01. Is this normal?
3. When I put an object at a 60cm distance from the sensor and I detect its centre of mass at (x,y) at the colour image and I am printing the depth values in the depth image at the (x,y) point I get zero, no matter how far I am moving the object. The values change when I move my hand relatively close to the sensor in random directions but it's mostly zero. Isn't the depth image filled with 16 unsigned one channel integers? If so, how can I successfully extract this data?
4. How can I convert the raw data to millimetres?   

I can understand that you are constantly bombarded with questions and its not your job to resolve anything being asked, but I would greatly appreciate your help, because as an undergraduate student I don't have someone else to whom I can ask this kind of questions.

Thank you for your answers and for your time in advance,
Chris

wiedemeyer, 2016-10-27:
Hi Chris,
1. There is a factory calibration available. It is used for the SD color image registration. But for the HD registration extrinsic parameters are needed. Also a good hand made calibration can be better that the factory defaults. The depth correction is needed, because for manipulation objects you need precision, for  tracking people, like for games, it is not needed.
2. Around 2 cm is similar to the values I got.
3. Yes, it is `CV_16UC1`. Maybe the depth processing is not working. Please check with the `kinect2_viewer`, that your object is visible in the depth image / point cloud, if you can see the object there, than the depth at those points cannot be zero. You can read the values the same way than from any other image. Make sure to use the corresponding images, if you are using the QHD color image, then you also have to use QHD depth image to use the same x-y-coordinates, or you have to convert them.
4. The raw data is in millimeters.

patrchri, 2016-10-27:
Hello again,
firstly I would like to thank you for your answer because at least now I think I have a sense on what maybe is wrong. Some follow up questions occurred after your answer, I am marking as bold all my questions to help you for your answer:  

1, 2. During [calibration](https://github.com/code-iai/iai_kinect2/tree/master/kinect2_calibration#calibrating-the-kinect-one) at step 8, I got a re-projection error close to approximately 19.0, **would you advise me to attempt another calibration with an error between 0-1.0 or this doesn't affect much the qhd resolution's topics and the depth shift but only the HD extrinsic parameters ?** All the re-projection errors from the previous steps (steps 4,6) were arithmetically normal.
3. I am using the `"/kinect2/qhd/image_color_rect"` and `"/kinect2/qhd/image_depth_rect"` so a simple usage of the same x,y should work as you also suggest. However I followed your advice to use the `kinect2_viewer` and I got the following results:

Below is a color image to show you in what environment I am working and I am detecting a small red object in front of the sensor:
[image]

In this environment I `rosrun kinect2_viewer kinect2_viewer` and I got the following results:
```

[ INFO] [main] topic color: /kinect2/qhd/image_color_rect
[ INFO] [main] topic depth: /kinect2/qhd/image_depth_rect
[ INFO] [main] starting receiver...
```

and and the following image:

[image]

In the image the fps were not constant. From what I see, I assume I am getting wrong results because I can see the black depth points of the portraits on the wall of my house shifted in comparison to the rgb image and I can barely see my object. **Is the above assumption correct?** 

To show better what I mean, I move my hand in a random position and I see the same behaviour my hand in colour is shifted in comparison to my hand in black/depth:
[image]

wiedemeyer, 2016-10-28:
Yes, the depth image does not match the color image. The reprojection error for the extrinsics should be below 1. something below 0.2 is good. This is very important for the registration.
To get better frame rates, use the OpenCL or CUDA depth processor. And if you have an Intel GPU (IvyBridge (or maybe SandyBridge) or newer) use the VAAPI Jpeg decoder.

patrchri, 2016-10-31:
Hello again,

I apologize for delaying my answer, but I had to calibrate carefully the sensor this time according to your instructions and recommendations about the tripods used. 
I have managed to reduce the re-projection error of the extrinsics to 0.6 approximately, while keeping the re-projection errors of the intrinsics of the rgb camera and ir sensor below 0.2 . Possibly, I could achieve an even lower re-projection error in a next try, but I wanted to give it a try again to see if I will get the depth correctly. My depthShift this time in the calib_depth.yaml is: `-1.2437659317443790e+00`
This time I was working on the following environment tracking the purple object:

[image]

and I was still getting zero depth from the kinect while the object was at 50cm away from the sensor. The printing was done by:

`ROS_INFO("The object is in depth: %d",cv_ptr2->image.at<ushort>(Point(x,y)));`

This time, I `rosrun kinect2_viewer kinect2_viewer` and I got the following picture:

[image]

while the depth image was the following:

[image]

I am suddenly confused on what is wrong again. I haven't used yet the libraries you mentioned for better frame rates, I don't know if this is something that could affect that constant depth's zero I get.

In case you have an idea on what I should look for this time, I would again appreciate your help.

Thank you for your answer and time in advance once again,
Chris

wiedemeyer, 2016-11-03:
From the kinect2_viewer I can see that everything is fine. The depth image has 16 bit and most values are pretty low. Some image viewers cannot show them correctly, but trust me, most depth values are not zero.
From the kinect2_viewer you can see, that the bottle has no valid point in the middle (it's black and there are no 3D points). Try placing it somewhere else and check with the kinect2_viewer. Also it is more stable to look for the average depth in a region than just checking one depth value.

patrchri, 2016-11-03:
Thanks a lot for your help. I got depth finally correctly.
Before I close this issue, I would like to ask some confirmation questions:

1. Do you think that I should recalibrate the Kinect v2? Is my depth shift(`-1.2437659317443790e+00`) parameter correct? 
2. Does the light play a role in the depth data I get or the reflections of the light? For example a white desk on which I have put my object can affect the data in comparison to a black desk? I am asking because in the same environment as in my previous question, without changing anything in the printing I got depth this time of the point (x,y). `ROS_INFO("The object is in depth: %d",cv_ptr2->image.at<ushort>(Point(x,y)));`
I will however follow your advice of getting the mean depth in a region of the depth image instead of a point.

Thanks a lot for your tips and help once again.
Chris

wiedemeyer, 2016-11-07:
1. could be correct.
2. The depth values highly depend on the material and color. Black is bad, because it is not reflecting well.

closed 2016-11-07 by wiedemeyer

MohamedNaveed, 2017-05-19:
Hello Patrchri!
I'm also facing the same issue. The depth of the object I'm using is always zero. Can you tell me how you got it working ?

patrchri, 2017-05-19:
Hello Mohamed,

what I did was re-do the calibration of the sensor so I can align the depth data with the image data.
So, re-do the calibration carefully and with the correct equipment (tripod for chessboard and for kinect) and follow the instructions of Wiedermeyer correctly and you will be fine.

wuguangbin1230, 2017-08-05:
Hello, 
1. After the calibration, I got a depth shift equal to -1.9074631106413037e+01. Is this normal?
2. If I get the normal depth image (kinect v2), does the depth values in depth image are the real distance between objects and the sensor?

youngee529, 2017-08-10:
Hello,
where can I get the source for depth and rgb calibration?

tatag4u, 2019-04-27:
How to capture using kinectv2