code-iai/iai_kinect2 #327: Incorrect depth preception from Kinect2_bridge, but correct depth through ./Protonect

kevinlyu1, 2016-10-20:
I am having a problem with the kinect2_bridge. For some reason it marks a door incorrectly in the middle of the hallway when it is actually at the end of the hallway. I have calibrated the kinect2 multiple times using the kinect2_calibration tutorial. Below are the images that show the problem that I am getting through multiple angles within rviz and image_view of the depth image. The weird part is that when I run ./Protonect the depth image seems to be correct and it doesn't show the problem that the kinect2_bridge has. The red circle indicates the object in all of the pictures.

[image]

[image]

[image]

[image]

[image]

[image]

**Here are some real world pictures that were taken with a camera.**
[image]

[image]

In the end the problem is that the depth image is not correct when running kinect2_bridge and I can't seem to fix it. If anyone has had this problem or knows how to fix it that would be very helpful.

wiedemeyer, 2016-10-27:
It only uses the data that it gets from libfreenect2. If the door is seen, it must be seen by libfreenect2 too. The door seems to be very far away, much further than the kinect2 should see. Why it is happening I don't know. Maybe the distance is in a range, that the reflection from the light of a previous measurement is seen in the next measurement?
The easiest solution is to filter everything further away than a certain distance. Use the `max_depth` parameter for it.

kevinlyu1, 2016-10-27:
Could you explain more on the "reflection from the light of a previous measurement" I'm not quite sure what you mean?
Also the error does not show up in libfreenect2 when I run `./Protonect` shown in 6th image.
I have tried using the `max_depth` parameter, but the problem still shows up if the door is in view. The only way of removing the problem was to cover up the whole corridor so that the door wasn't seen at all.

wiedemeyer, 2016-10-28:
To get a depth image, the sensor takes multiple images shortly after each other. For this it sends out a IR flash. maybe the door is in a distance that the reflection of the flash from a previous image shows up in one of the following. But I am not sure, that's just some random guess, so don't care.
As you can see, Protonect is only showing results until the door on the left, therefore you cannot see it there, but it is probably also seen by Protonect. If you set `max_depth` to 4.5 meters it should not be seen by kinect2_bridge. Look at the terminal output from kinect2_bridge to see if you specified `max_depth` value is used.

kevinlyu1, 2016-10-29:
For the first part I'm not sure if that's what's happening because the problem is there at many different ranges. So if I move the robot close to the problem area it still shows up and even if i'm far away from it, the problem will stay in the same location.
Also I tried changing the `max_depth` and it still didn't fix the problem. Currently I haven't figured out a way to remove the problem unless I place something over the whole corridor so it doesn't see the door at all.