code-iai/iai_kinect2 #328: Calibration issues

natejgardner, 2016-10-22:
The calibration instructions don't seem to be working for me. I printed out the checkerboard pattern and ran `rosrun kinect2_calibration kinect2_calibration chess7x9x0.025 record color`. I hit the spacebar, and nothing happened. I hit s, and saved the image instead, since the spacebar did nothing. Results:

```
i3@flynn:~/kinect_cal_data$ rosrun kinect2_calibration kinect2_calibration chess7x9x0.025 record color
[ INFO] [main] Start settings:
       Mode: record
     Source: color
      Board: chess
 Dimensions: 7 x 9
 Field size: 0.025
Dist. model: 5 coefficients
Topic color: /kinect2/hd/image_mono
   Topic ir: /kinect2/sd/image_ir
Topic depth: /kinect2/sd/image_depth
       Path: ./

[ INFO] [main] starting recorder...
[ INFO] [Recorder::startRecord] Controls:
   [ESC, q] - Exit
 [SPACE, s] - Save current frame
        [l] - decrease min and max value for IR value range
        [h] - increase min and max value for IR value range
        [1] - decrease min value for IR value range
        [2] - increase min value for IR value range
        [3] - decrease max value for IR value range
        [4] - increase max value for IR value range
[ERROR] Tried to advertise a service that is already advertised in this node [/kinect2_calib_1477120806694659185/compressed/set_parameters]
[ERROR] Tried to advertise a service that is already advertised in this node [/kinect2_calib_1477120806694659185/compressed/set_parameters]
```

I then ran `rosrun kinect2_calibration kinect2_calibration chess7x9x0.025 calibrate color` and got:

```
i3@flynn:~/kinect_cal_data$ rosrun kinect2_calibration kinect2_calibration chess7x9x0.025 calibrate color
[ INFO] [main] Start settings:
       Mode: calibrate
     Source: color
      Board: chess
 Dimensions: 7 x 9
 Field size: 0.025
Dist. model: 5 coefficients
Topic color: /kinect2/hd/image_mono
   Topic ir: /kinect2/sd/image_ir
Topic depth: /kinect2/sd/image_depth
       Path: ./

[ INFO] [main] restoring files...
[ERROR] [CameraCalibration::restore] no files found!
[ INFO] [main] starting calibration...
[ERROR] [CameraCalibration::calibrateIntrinsics] no data for calibration provided!
```

Am I missing a step?

wiedemeyer, 2016-10-27:
When recording the window showing the image has to be active and the pattern has to be detected (colored overlay is shown on pattern).

natejgardner, 2016-10-28:
In that case, it appears the pattern is never detected. This is surprising, since the pattern is an appropriate distance, facing the kinect directly, and very distinct (the image has a completely featureless background). How can I debug to find out if it's attempting to detect the pattern?

wiedemeyer, 2016-10-28:
Please paste a screenshot of the kinect2_calibration window with the pattern in view.

natejgardner, 2016-11-01:
I believe I was accidentally using the wrong pattern. I was able to successfully calibrate!

wiedemeyer, 2016-11-02:
Thats good.

closed 2016-11-02 by wiedemeyer

hengguan, 2017-02-25:
press **spacebar** to record each image.

JiahengZhao, 2018-02-05:
I met the same error, would you please tell me how to solve it?

kaixindelele, 2018-03-20:
Maybe the sun caused this accident~I have encountered the same problem~