code-iai/iai_kinect2 #331: Invoking "cmake" failed with catkin_make -DCMAKE_BUILD_TYPE="Release"

crazysmart, 2016-11-03:
How to solve the error above? Thanks~

zhangte@zhangte-Lenovo-V3000:~/catkin_ws$ catkin_make -DCMAKE_BUILD_TYPE="Release"
Base path: /home/zhangte/catkin_ws
Source space: /home/zhangte/catkin_ws/src
Build space: /home/zhangte/catkin_ws/build
Devel space: /home/zhangte/catkin_ws/devel
Install space: /home/zhangte/catkin_ws/install
####
#### Running command: "cmake /home/zhangte/catkin_ws/src -DCMAKE_BUILD_TYPE=Release -DCATKIN_DEVEL_PREFIX=/home/zhangte/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/zhangte/catkin_ws/install -G Unix Makefiles" in "/home/zhangte/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/zhangte/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/indigo
-- This workspace overlays: /opt/ros/indigo
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/zhangte/catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.18
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 5 packages in topological order:
-- ~~  - iai_kinect2 (metapackage)
-- ~~  - kinect2_registration
-- ~~  - kinect2_bridge
-- ~~  - kinect2_calibration
-- ~~  - kinect2_viewer
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin metapackage: 'iai_kinect2'
-- ==> add_subdirectory(iai_kinect2/iai_kinect2)
-- +++ processing catkin package: 'kinect2_registration'
-- ==> add_subdirectory(iai_kinect2/kinect2_registration)
-- OpenCV ARCH: 
-- OpenCV RUNTIME: 
-- OpenCV STATIC: OFF
CMake Warning at /usr/local/opencv-2.4.13/cmake/OpenCVConfig.cmake:163 (message):
  Found OpenCV Windows Pack but it has not binaries compatible with your
  configuration.

  You should manually point CMake variable OpenCV_DIR to your build of OpenCV
  library.
Call Stack (most recent call first):
  iai_kinect2/kinect2_registration/CMakeLists.txt:29 (find_package)

CMake Error at iai_kinect2/kinect2_registration/CMakeLists.txt:29 (find_package):
  Found package configuration file:

    /usr/local/opencv-2.4.13/cmake/OpenCVConfig.cmake

  but it set OpenCV_FOUND to FALSE so package "OpenCV" is considered to be
  NOT FOUND.

-- Configuring incomplete, errors occurred!
See also "/home/zhangte/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/zhangte/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

wiedemeyer, 2016-11-03:
Hi,
that's not related to this package but to your OpenCV installation. Something seems to be wrong there. It is recommended to use the version that comes with ROS.

akpndt, 2016-11-17:
Hello, can anybody help me to figure out the these compilation errors: 
........partial...........
make[2]: *** [iai_kinect2/kinect2_registration/CMakeFiles/kinect2_registration.dir/src/depth_registration_cpu.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [iai_kinect2/kinect2_registration/CMakeFiles/kinect2_registration.dir/src/depth_registration_opencl.cpp.o] Error 1
make[2]: *** [iai_kinect2/kinect2_registration/CMakeFiles/kinect2_registration.dir/src/kinect2_registration.cpp.o] Error 1
make[1]: *** [iai_kinect2/kinect2_registration/CMakeFiles/kinect2_registration.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

wiedemeyer, 2016-11-25:
Well, please past the errors, and not the summary of the build.

closed 2016-12-13 by wiedemeyer

TrinhTUHH, 2017-07-28:
Hi, How do you fix this issue?