code-iai/iai_kinect2 #332: kinect2_bridge crashes (ROS newbie)

sa7ina, 2016-11-14:
Hi

Can please someone help me find out what the problem.
Installed step by step first libfreenect2 build\installation. (Protonect works flawlessly)
Then installed iai_kinect2 not working.
Tried all solutions here but nothing.
I am missing something.

Hoped someone here could help.
10X

Terminal:
`
roslaunch kinect2_bridge kinect2_bridge.launch 
... logging to /home/ei8ht/.ros/log/9152b312-aa0a-11e6-af63-342387a7febb/roslaunch-P570WM-4527.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://P570WM:37252/

SUMMARY
========

PARAMETERS
 * /kinect2_bridge/base_name: kinect2
 * /kinect2_bridge/base_name_tf: kinect2
 * /kinect2_bridge/bilateral_filter: True
 * /kinect2_bridge/calib_path: /home/ei8ht/catki...
 * /kinect2_bridge/depth_device: -1
 * /kinect2_bridge/depth_method: default
 * /kinect2_bridge/edge_aware_filter: True
 * /kinect2_bridge/fps_limit: -1.0
 * /kinect2_bridge/jpeg_quality: 90
 * /kinect2_bridge/max_depth: 12.0
 * /kinect2_bridge/min_depth: 0.1
 * /kinect2_bridge/png_level: 1
 * /kinect2_bridge/publish_tf: False
 * /kinect2_bridge/queue_size: 5
 * /kinect2_bridge/reg_device: -1
 * /kinect2_bridge/reg_method: default
 * /kinect2_bridge/sensor: 
 * /kinect2_bridge/use_png: False
 * /kinect2_bridge/worker_threads: 4
 * /kinect2_points_xyzrgb_hd/queue_size: 5
 * /kinect2_points_xyzrgb_qhd/queue_size: 5
 * /kinect2_points_xyzrgb_sd/queue_size: 5
 * /rosdistro: indigo
 * /rosversion: 1.11.20

NODES
  /
    kinect2 (nodelet/nodelet)
    kinect2_bridge (nodelet/nodelet)
    kinect2_points_xyzrgb_hd (nodelet/nodelet)
    kinect2_points_xyzrgb_qhd (nodelet/nodelet)
    kinect2_points_xyzrgb_sd (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [4539]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 9152b312-aa0a-11e6-af63-342387a7febb
process[rosout-1]: started with pid [4552]
started core service [/rosout]
process[kinect2-2]: started with pid [4568]
process[kinect2_bridge-3]: started with pid [4570]
process[kinect2_points_xyzrgb_sd-4]: started with pid [4571]
process[kinect2_points_xyzrgb_qhd-5]: started with pid [4572]
process[kinect2_points_xyzrgb_hd-6]: started with pid [4578]
[ INFO] [1479087305.170472759]: Loading nodelet /kinect2_bridge of type kinect2_bridge/kinect2_bridge_nodelet to manager kinect2 with the following remappings:
[ INFO] [1479087305.173463502]: waitForService: Service [/kinect2/load_nodelet] has not been advertised, waiting...
[ INFO] [1479087305.224721627]: Initializing nodelet with 12 worker threads.
[ INFO] [1479087305.235465102]: waitForService: Service [/kinect2/load_nodelet] is now available.
[ERROR] [1479087305.324431723]: Failed to load nodelet [/kinect2_bridge] of type [kinect2_bridge/kinect2_bridge_nodelet] even after refreshing the cache: Failed to load library /home/ei8ht/catkin_ws/install/lib//libkinect2_bridge_nodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libfreenect2.so.0.2: cannot open shared object file: No such file or directory)
[ERROR] [1479087305.324480681]: The error before refreshing the cache was: Failed to load library /home/ei8ht/catkin_ws/install/lib//libkinect2_bridge_nodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libfreenect2.so.0.2: cannot open shared object file: No such file or directory)
[FATAL] [1479087305.324662059]: Failed to load nodelet '/kinect2_bridge` of type `kinect2_bridge/kinect2_bridge_nodelet` to manager `kinect2'
[kinect2_bridge-3] process has died [pid 4570, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load kinect2_bridge/kinect2_bridge_nodelet kinect2 __name:=kinect2_bridge __log:=/home/ei8ht/.ros/log/9152b312-aa0a-11e6-af63-342387a7febb/kinect2_bridge-3.log].
log file: /home/ei8ht/.ros/log/9152b312-aa0a-11e6-af63-342387a7febb/kinect2_bridge-3*.log
[kinect2_bridge-3] restarting process
process[kinect2_bridge-3]: started with pid [4730]
`

[image]

sa7ina, 2016-11-14:
OK
Read some more and the problem was i had to source my workspace after compiling.
Now i have a new problem.

I run 'rosrun kinect2_bridge kinect2_bridge'.
Nothing is RED!...With is good.
The Kinect does lights on for a second than shouts down.

Terminal:
 'rosrun kinect2_bridge kinect2_bridge 
/home/ei8ht/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge: /usr/local/cuda/lib64/libOpenCL.so.1: no version information available (required by /home/ei8ht/catkin_ws/devel/lib/libkinect2_registration.so)
/home/ei8ht/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge: /usr/local/cuda/lib64/libOpenCL.so.1: no version information available (required by /home/ei8ht/catkin_ws/devel/lib/libkinect2_registration.so)
/home/ei8ht/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge: /usr/local/cuda/lib64/libOpenCL.so.1: no version information available (required by /home/ei8ht/freenect2/lib/libfreenect2.so.0.2)
/home/ei8ht/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge: /usr/local/cuda/lib64/libOpenCL.so.1: no version information available (required by /home/ei8ht/freenect2/lib/libfreenect2.so.0.2)
[ INFO] [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: default
        fps_limit: -1
       calib_path: /home/ei8ht/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: cuda
     depth_device: -1
       reg_method: opencl
       reg_device: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4
[Info] [CudaDepthPacketProcessorImpl] device 0: GeForce GTX 880M @ 993MHz Memory 8108MB
[Info] [CudaDepthPacketProcessorImpl] selected device 0
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 13 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:3 with serial 016514451147
[Info] [Freenect2Impl] found 1 devices
[ INFO] [Kinect2Bridge::initDevice] Kinect2 devices found: 
[ INFO] [Kinect2Bridge::initDevice]   0: 016514451147 (selected)
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[ INFO] [Kinect2Bridge::initDevice] starting kinect2
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
[ INFO] [Kinect2Bridge::initDevice] device serial: 016514451147
[ INFO] [Kinect2Bridge::initDevice] device firmware: 2.3.3913.0
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] canceling rgb transfers...
[Info] [Freenect2DeviceImpl] canceling depth transfers...
[Info] [Freenect2DeviceImpl] stopped
[ WARN] [Kinect2Bridge::initCalibration] using sensor defaults for color intrinsic parameters.
[ WARN] [Kinect2Bridge::initCalibration] using sensor defaults for ir intrinsic parameters.
[ WARN] [Kinect2Bridge::initCalibration] using defaults for rotation and translation.
[ WARN] [Kinect2Bridge::initCalibration] using defaults for depth shift.
[ INFO] [DepthRegistration::New] Using OpenCL registration method!
[ INFO] [DepthRegistration::New] Using OpenCL registration method!
Segmentation fault (core dumped)'

sa7ina, 2016-11-14:
I am answering myself here but it is a good thing for people with the same problem.

Ran it again this time with opencl registration method and depth method now it does work.
'rosrun kinect2_bridge kinect2_bridge _reg_method:=opencl'

BUT!
No image in RVIZ.

kinect2_bridge Terminal:
rosrun kinect2_bridge kinect2_bridge _reg_method:=opencl _depth_method:=opencl
/home/ei8ht/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge: /usr/local/cuda-8.0/targets/x86_64-linux/lib/libOpenCL.so.1: no version information available (required by /home/ei8ht/catkin_ws/devel/lib/libkinect2_registration.so)
/home/ei8ht/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge: /usr/local/cuda-8.0/targets/x86_64-linux/lib/libOpenCL.so.1: no version information available (required by /home/ei8ht/catkin_ws/devel/lib/libkinect2_registration.so)
/home/ei8ht/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge: /usr/local/cuda-8.0/targets/x86_64-linux/lib/libOpenCL.so.1: no version information available (required by /home/ei8ht/freenect2/lib/libfreenect2.so.0.2)
/home/ei8ht/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge: /usr/local/cuda-8.0/targets/x86_64-linux/lib/libOpenCL.so.1: no version information available (required by /home/ei8ht/freenect2/lib/libfreenect2.so.0.2)
[ INFO] [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: default
        fps_limit: -1
       calib_path: /home/ei8ht/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opencl
     depth_device: -1
       reg_method: opencl
       reg_device: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4
X server found. dri2 connection failed! 
/dev/dri/card0 not authenticated
/dev/dri/card1 not authenticated
X server found. dri2 connection failed! 
/dev/dri/card0 not authenticated
/dev/dri/card1 not authenticated
beignet-opencl-icd: no supported GPU found, this is probably the wrong opencl-icd package for this hardware
(If you have multiple ICDs installed and OpenCL works, you can ignore this message)
[Info] [OpenCLDepthPacketProcessorImpl]  devices:
[Info] [OpenCLDepthPacketProcessorImpl]   0: GeForce GTX 880M (GPU)[NVIDIA Corporation]
[Info] [OpenCLDepthPacketProcessorImpl]   1: GeForce GTX 880M (GPU)[NVIDIA Corporation]
[Info] [OpenCLDepthPacketProcessorImpl] selected device: GeForce GTX 880M (GPU)[NVIDIA Corporation]
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 13 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:3 with serial 016514451147
[Info] [Freenect2Impl] found 1 devices
[ INFO] [Kinect2Bridge::initDevice] Kinect2 devices found: 
[ INFO] [Kinect2Bridge::initDevice]   0: 016514451147 (selected)
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[ INFO] [Kinect2Bridge::initDevice] starting kinect2
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
[ INFO] [Kinect2Bridge::initDevice] device serial: 016514451147
[ INFO] [Kinect2Bridge::initDevice] device firmware: 2.3.3913.0
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] canceling rgb transfers...
[Info] [Freenect2DeviceImpl] canceling depth transfers...
[Info] [Freenect2DeviceImpl] stopped
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[... 81 lines cut ...]
[Info] [Freenect2DeviceImpl] started
[Info] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Info] [TurboJpegRgbPacketProcessor] avg. time: 15.56ms -> ~64.2675Hz
[ INFO] [Kinect2Bridge::main] color processing: ~11.6679ms (~85.7052Hz) publishing rate: ~26.3281Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 15.375ms -> ~65.0408Hz
[ INFO] [Kinect2Bridge::main] color processing: ~11.9951ms (~83.3673Hz) publishing rate: ~29.9013Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 15.8234ms -> ~63.1976Hz
[ INFO] [Kinect2Bridge::main] color processing: ~11.5071ms (~86.9026Hz) publishing rate: ~30.2332Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 15.7645ms -> ~63.4338Hz

rviz Terminal:
rosrun rviz rviz
[ INFO] [1479110685.744120053]: rviz version 1.11.15
[ INFO] [1479110685.744167801]: compiled against Qt version 4.8.6
[ INFO] [1479110685.744179668]: compiled against OGRE version 1.8.1 (Byatis)
[ INFO] [1479110686.500740054]: Stereo is NOT SUPPORTED
[ INFO] [1479110686.500865636]: OpenGl version: 4.5 (GLSL 4.5).
[ WARN] [1479110709.970204562]: MessageFilter [target=map ]: Dropped 100.00% of messages so far. Please turn the [ros.rviz.message_notifier] rosconsole logger to DEBUG for more information.
[ WARN] [1479110724.967653316]: MessageFilter [target=map ]: Dropped 100.00% of messages so far. Please turn the [ros.rviz.message_notifier] rosconsole logger to DEBUG for more information.
[... cut ...]
[ WARN] [1479110832.219174395]: MessageFilter [target=map ]: Dropped 100.00% of messages so far. Please turn the [ros.rviz.message_notifier] rosconsole logger to DEBUG for more information.
[ WARN] [1479110890.014020568]: MessageFilter [target=map ]: Dropped 100.00% of messages so far. Please turn the [ros.rviz.message_notifier] rosconsole logger to DEBUG for more information.
[ WARN] [1479110892.224836953]: MessageFilter [target=map ]: Dropped 100.00% of messages so far. Please turn the [ros.rviz.message_notifier] rosconsole logger to DEBUG for more information.

wiedemeyer, 2016-11-14:
Hi,
maybe CUDA is not working right. Try `./Protonect cuda` to check.

sa7ina, 2016-11-14:
Hi
10X a lot for trying to help.

Ran ./Protonect cuda and its running.

But i think something is wrong with CUDA...Look at the first 2 lines

Terminal:
./Protonect cuda
./Protonect: /usr/local/cuda/lib64/libOpenCL.so.1: no version information available (required by /home/ei8ht/libfreenect2/build/lib/libfreenect2.so.0.2)
./Protonect: /usr/local/cuda/lib64/libOpenCL.so.1: no version information available (required by /home/ei8ht/libfreenect2/build/lib/libfreenect2.so.0.2)
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./Protonect [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [CudaDepthPacketProcessorImpl] device 0: GeForce GTX 880M @ 993MHz Memory 8108MB
[Info] [CudaDepthPacketProcessorImpl] selected device 0
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 13 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:3 with serial 016514451147
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] status 0x090000: 9729
[Debug] [Freenect2DeviceImpl] status 0x090000: 9731
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 016514451147
device firmware: 2.3.3913.0
[Debug] [DepthPacketStreamParser] not all subsequences received 0
[Info] [CudaDepthPacketProcessor] avg. time: 1.96134ms -> ~509.855Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 15.7374ms -> ~63.5428Hz
[Info] [CudaDepthPacketProcessor] avg. time: 1.82992ms -> ~546.471Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 15.3938ms -> ~64.9611Hz
[Info] [CudaDepthPacketProcessor] avg. time: 1.88937ms -> ~529.278Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 15.6191ms -> ~64.0243Hz
[Info] [CudaDepthPacketProcessor] avg. time: 1.93619ms -> ~516.479Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 15.4839ms -> ~64.5833Hz
[Info] [CudaDepthPacketProcessor] avg. time: 1.94415ms -> ~514.365Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 14.8939ms -> ~67.1414Hz
[Info] [CudaDepthPacketProcessor] avg. time: 1.95872ms -> ~510.536Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 15.1663ms -> ~65.9357Hz
[Info] [CudaDepthPacketProcessor] avg. time: 1.94783ms -> ~513.393Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 14.9972ms -> ~66.6792Hz
[Info] [CudaDepthPacketProcessor] avg. time: 1.95501ms -> ~511.506Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 15.1108ms -> ~66.1779Hz
[Info] [CudaDepthPacketProcessor] avg. time: 1.94819ms -> ~513.296Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 15.907ms -> ~62.8654Hz

sa7ina, 2016-11-14:
Another thing is i have to source my workspace again every time i restart the computer.

wiedemeyer, 2016-11-14:
That is normal behavior, if you didn't put the sourcing command into `~/.bashrc`.
For finding out why it is not working with kinect2_bridge, try the followin steps from the [FAQ](https://github.com/code-iai/iai_kinect2#protonect-works-fine-but-kinect2_bridge-is-still-not-working--crashing):
```
cd <catkin_ws>
catkin_make -DCMAKE_BUILD_TYPE="Debug"
cd devel/lib/kinect2_bridge
gdb kinect2_bridge
// inside gdb: run until it crashes and do a backtrace
run
bt
quit
```
and paste the backtrace here.

sa7ina, 2016-11-14:
gdb kinect2_bridge
GNU gdb (Ubuntu 7.7.1-0ubuntu5~14.04.2) 7.7.1
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from kinect2_bridge...done.
(gdb) run
Starting program: /home/ei8ht/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge 
/home/ei8ht/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge: /usr/local/cuda-8.0/targets/x86_64-linux/lib/libOpenCL.so.1: no version information available (required by /home/ei8ht/catkin_ws/devel/lib/libkinect2_registration.so)
/home/ei8ht/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge: /usr/local/cuda-8.0/targets/x86_64-linux/lib/libOpenCL.so.1: no version information available (required by /home/ei8ht/catkin_ws/devel/lib/libkinect2_registration.so)
/home/ei8ht/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge: /usr/local/cuda-8.0/targets/x86_64-linux/lib/libOpenCL.so.1: no version information available (required by /home/ei8ht/freenect2/lib/libfreenect2.so.0.2)
/home/ei8ht/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge: /usr/local/cuda-8.0/targets/x86_64-linux/lib/libOpenCL.so.1: no version information available (required by /home/ei8ht/freenect2/lib/libfreenect2.so.0.2)
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffde6ce700 (LWP 3920)]
[New Thread 0x7fffdd5df700 (LWP 3922)]
[New Thread 0x7fffdcdde700 (LWP 3923)]
[New Thread 0x7fffd7fff700 (LWP 3924)]
[New Thread 0x7fffd77fe700 (LWP 3929)]
[New Thread 0x7fffd6ffd700 (LWP 3930)]
[ INFO] [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: default
        fps_limit: -1
       calib_path: /home/ei8ht/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: cuda
     depth_device: -1
       reg_method: opencl
       reg_device: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4
[New Thread 0x7fffce813700 (LWP 3950)]
[Info] [CudaDepthPacketProcessorImpl] device 0: GeForce GTX 880M @ 993MHz Memory 8108MB
[Info] [CudaDepthPacketProcessorImpl] selected device 0
[New Thread 0x7fffce012700 (LWP 3951)]
[New Thread 0x7fffcd811700 (LWP 3952)]
[New Thread 0x7fffb7fff700 (LWP 3953)]
[New Thread 0x7fffb77fe700 (LWP 3954)]
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 13 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:3 with serial 016514451147
[Info] [Freenect2Impl] found 1 devices
[ INFO] [Kinect2Bridge::initDevice] Kinect2 devices found: 
[ INFO] [Kinect2Bridge::initDevice]   0: 016514451147 (selected)
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[ INFO] [Kinect2Bridge::initDevice] starting kinect2
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[... 19 lines cut ...]
#1  0x00007fffb496f008 in ?? () from /usr/lib/x86_64-linux-gnu/libnvidia-opencl.so.1
#2  0x00007fffb4a23671 in ?? () from /usr/lib/x86_64-linux-gnu/libnvidia-opencl.so.1
#3  0x00007fffb4a237e5 in ?? () from /usr/lib/x86_64-linux-gnu/libnvidia-opencl.so.1
#4  0x00007fffb4974cb4 in ?? () from /usr/lib/x86_64-linux-gnu/libnvidia-opencl.so.1
#5  0x00007fffb49764e7 in ?? () from /usr/lib/x86_64-linux-gnu/libnvidia-opencl.so.1
#6  0x00007fffb4949c66 in ?? () from /usr/lib/x86_64-linux-gnu/libnvidia-opencl.so.1
#7  0x00007fffb4848f3d in ?? () from /usr/lib/x86_64-linux-gnu/libnvidia-opencl.so.1
#8  0x00007fffb4848ed8 in ?? () from /usr/lib/x86_64-linux-gnu/libnvidia-opencl.so.1
#9  0x00007fffee49a022 in ?? () from /usr/local/cuda-8.0/targets/x86_64-linux/lib/libOpenCL.so.1
#10 0x00007fffee49bd42 in ?? () from /usr/local/cuda-8.0/targets/x86_64-linux/lib/libOpenCL.so.1
#11 0x00007fffee49b4d0 in clGetPlatformIDs () from /usr/local/cuda-8.0/targets/x86_64-linux/lib/libOpenCL.so.1
#12 0x00007ffff5a22f63 in cl::Platform::get (platforms=0x7fffffffbca0) at /home/ei8ht/catkin_ws/src/iai_kinect2/kinect2_registration/include/internal/CL/cl.hpp:2333
#13 0x00007ffff5a16e2e in DepthRegistrationOpenCL::init (this=0xacd450, deviceId=-1) at /home/ei8ht/catkin_ws/src/iai_kinect2/kinect2_registration/src/depth_registration_opencl.cpp:180
#14 0x00007ffff5a12c33 in DepthRegistration::init (this=0xacd450, cameraMatrixRegistered=..., sizeRegistered=..., cameraMatrixDepth=..., sizeDepth=..., distortionDepth=..., rotation=..., translation=..., 
    zNear=0.5, zFar=12, deviceId=-1) at /home/ei8ht/catkin_ws/src/iai_kinect2/kinect2_registration/src/kinect2_registration.cpp:52
[... cut ...]
#16 0x000000000044eaf5 in Kinect2Bridge::initialize (this=0x7fffffffcbf0) at /home/ei8ht/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:314
#17 0x000000000044c5f3 in Kinect2Bridge::start (this=0x7fffffffcbf0) at /home/ei8ht/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:152
#18 0x0000000000448168 in main (argc=1, argv=0x7fffffffdb78) at /home/ei8ht/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:1603
(gdb) quit
A debugging session is active.

	Inferior 1 [process 3915] will be killed.

Quit anyway? (y or n) y

sa7ina, 2016-11-15:
Works!
Rebuilt it resourced it this time properly.
Run great in opencl:
rosrun kinect2_bridge kinect2_bridge _reg_method:=opencl _depth_method:=opencl _publish_tf:=true

10X a lot for your help my friend.

closed 2016-11-25 by wiedemeyer

smarthhy, 2018-07-29:
How did you solve this problem? thanks