code-iai/iai_kinect2 #333: kinect2_bridge crash on hq topic

Raffa87, 2016-11-14:
Hi!
When I try to subscribe to hq or qhq topics, I get the following error:

drm_intel_gem_bo_context_exec() failed: Invalid argument
[ERROR] [1479138227.574739485]: [DepthRegistrationOpenCL::registerDepth] [depth_registration_opencl.cpp](283) data->queue.enqueueNDRangeKernel(data->kernelSetZero, cl::NullRange, range, cl::NullRange, NULL, &eventZero[1]) failed: -5
[Info] [DepthPacketStreamParser] -16359 packets were lost
drm_intel_gem_bo_context_exec() failed: Invalid argument
[ERROR] [1479138227.602298204]: [DepthRegistrationOpenCL::registerDepth] [depth_registration_opencl.cpp](283) data->queue.enqueueNDRangeKernel(data->kernelSetZero, cl::NullRange, range, cl::NullRange, NULL, &eventZero[1]) failed: -5
drm_intel_gem_bo_context_exec() failed: Invalid argument
[ERROR] [1479138227.639239737]: [DepthRegistrationOpenCL::registerDepth] [depth_registration_opencl.cpp](283) data->queue.enqueueNDRangeKernel(data->kernelSetZero, cl::NullRange, range, cl::NullRange, NULL, &eventZero[1]) failed: -5
drm_intel_gem_bo_context_exec() failed: Invalid argument
[ERROR] [1479138227.747751320]: [DepthRegistrationOpenCL::registerDepth] [depth_registration_opencl.cpp](283) data->queue.enqueueNDRangeKernel(data->kernelSetZero, cl::NullRange, range, cl::NullRange, NULL, &eventZero[1]) failed: -5
OpenCV Error: Assertion failed (0 <= _rowRange.start && _rowRange.start <= _rowRange.end && _rowRange.end <= m.rows) in Mat, file /build/buildd/opencv-2.4.8+dfsg1/modules/core/src/matrix.cpp, line 284
terminate called after throwing an instance of 'cv::Exception'
  what():  /build/buildd/opencv-2.4.8+dfsg1/modules/core/src/matrix.cpp:284: error: (-215) 0 <= _rowRange.start && _rowRange.start <= _rowRange.end && _rowRange.end <= m.rows in function Mat

[kinect2-1] process has died [pid 1052, exit code -6, cmd /opt/ros/indigo/lib/nodelet/nodelet manager __name:=kinect2 __log:=/home/nuc1/.ros/log/b4e37662-a81c-11e6-9362-c03fd562e167/kinect2-1.log].
log file: /home/nuc1/.ros/log/b4e37662-a81c-11e6-9362-c03fd562e167/kinect2-1*.log

wiedemeyer, 2016-11-25:
There seems to be something wrong with your OpenCL. Which GPU and OpenCL version are you using? Does `Protonect cl` work? You could also use the CPU registration.

Raffa87, 2016-12-01:
Hi! Thank you for your support
Actually, it seems to be a problem with OpenCL, since also Protonect cl is not working. How can i get the GPU and OpenCL version used? It is an Intel NUC
Moreover, I tried to use CPU method as registration method and CPU/OpenGL as depth method, but I get data only on sd topics

wiedemeyer, 2016-12-12:
You may have to install a newer Beignet version that supports your CPU/GPU.

Raffa87, 2016-12-14:
Hi! Thank you for your support!
I have installed Beignet, but  Protonect cl is still not working

wiedemeyer, 2016-12-14:
Make sure the version you installed supports your CPU. You can probably check it on the Beignet website.

Raffa87, 2016-12-16:
Hi!
I installed Beignet following the instructions described here: https://gist.github.com/spiralray/cae0bc235509e495fec1

However, Protonect cl still doesn't work

This is the result of clinfo:

Failed to release test userptr object! (9) i915 kernel driver may not be sane!
Failed to release test userptr object! (9) i915 kernel driver may not be sane!
Failed to release test userptr object! (9) i915 kernel driver may not be sane!
Failed to release test userptr object! (9) i915 kernel driver may not be sane!
Number of platforms:				 2
  Platform Profile:				 FULL_PROFILE
  Platform Version:				 OpenCL 1.2 beignet 1.0.3
  Platform Name:				 Intel Gen OCL Driver
  Platform Vendor:				 Intel
  Platform Extensions:				 cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_byte_addressable_store cl_khr_icd
  Platform Profile:				 FULL_PROFILE
  Platform Version:				 OpenCL 1.2 beignet 1.0.2
  Platform Name:				 Intel Gen OCL Driver
  Platform Vendor:				 Intel
  Platform Extensions:				 cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_byte_addressable_store cl_khr_icd

  Platform Name:				 Intel Gen OCL Driver
Number of devices:				 1
  Device Type:					 CL_DEVICE_TYPE_GPU
  Device ID:					 2582
  Max compute units:				 20
  Max work items dimensions:			 3
    Max work items[0]:				 512
    Max work items[1]:				 512
    Max work items[2]:				 512
  Max work group size:				 512
  Preferred vector width char:			 16
  Preferred vector width short:			 8
  Preferred vector width int:			 4
  Preferred vector width long:			 2
  Preferred vector width float:			 4
  Preferred vector width double:		 0
  Native vector width char:			 8
  Native vector width short:			 8
  Native vector width int:			 4
  Native vector width long:			 2
  Native vector width float:			 4
  Native vector width double:			 2
  Max clock frequency:				 1000Mhz
  Address bits:					 32
  Max memory allocation:			 1073741824
  Image support:				 Yes
  Max number of images read arguments:		 128
  Max number of images write arguments:		 8
  Max image 2D width:				 8192
  Max image 2D height:				 8192
  Max image 3D width:				 8192
  Max image 3D height:				 8192
  Max image 3D depth:				 2048
  Max samplers within kernel:			 16
  Max size of kernel argument:			 1024
  Alignment (bits) of base address:		 1024
  Minimum alignment (bytes) for any datatype:	 128
  Single precision floating point capability
    Denorms:					 No
    Quiet NaNs:					 Yes
    Round to nearest even:			 Yes
    Round to zero:				 No
    Round to +ve and infinity:			 No
    IEEE754-2008 fused multiply-add:		 No
  Cache type:					 Read/Write
  Cache line size:				 64
  Cache size:					 8192
[... 75 lines cut ...]
  Max number of constant args:			 8
  Local memory type:				 Global
  Local memory size:				 65536
  Error correction support:			 0
  Unified memory for Host and Device:		 1
  Profiling timer resolution:			 80
  Device endianess:				 Little
  Available:					 Yes
  Compiler available:				 Yes
  Execution capabilities:				 
    Execute OpenCL kernels:			 Yes
    Execute native function:			 Yes
  Queue properties:				 
    Out-of-Order:				 No
    Profiling :					 Yes
  Platform ID:					 0x7f9d601e45c0
  Name:						 Intel(R) HD Graphics Haswell Ultrabook GT2 Mobile
  Vendor:					 Intel
  Device OpenCL C version:			 OpenCL C 1.2 beignet 1.0.2
  Driver version:				 1.0.2
  Profile:					 FULL_PROFILE
  Version:					 OpenCL 1.2 beignet 1.0.2
  Extensions:					 cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_byte_addressable_store cl_khr_icd

Thanks!!!!

wiedemeyer, 2017-01-06:
beignet 1.0.2 is really old. Instructions on how to install beignet 1.1.1 can be found in the libfreenect README: https://github.com/OpenKinect/libfreenect2/blob/master/README.md#linux

closed 2017-02-28 by wiedemeyer