code-iai/iai_kinect2 #334: Kinect depth data inconsistent with Aruco marker transforms

miguelriemoliveira, 2016-11-18:
Hello,

I am using aruco markers (https://www.uco.es/investiga/grupos/ava/node/26) for calibrating extrinsics, i.e., camera to marker transformations.

The problem is that the results I get with the aruco markers are inconsistent with the point cloud given by the kinect. I read in some issue that the "depth registration is a preliminary piece of code", but still I am not even sure which one is giving the correct transformation, aruco or kinect.

Here is an example:

[image]

[image]

The depth measurements are off by 2 or 3 cm (along the camera's optical axis), when compared to the aruco markers.

BTW, I calibrated the kinect using the documented procedures and the results seem fine, i.e., typical values.

Do you have any information concerning the accuracy of the kinect depth measurements?

Thanks in advance,

Miguel

wiedemeyer, 2016-11-18:
Hi,
the depth registration should work fine. The depth calibration is preliminary code. But did you do the depth calibration? And what is the computed depth shift? If you didn't do that step maybe it helps, or if you had done it, check if the values made sense.
For our Kinects we had some more or less static offset of 1 to 3 cm, depending on the device.

miguelriemoliveira, 2016-11-18:
Hi Thiemo,

Depth registration, i.e., how well depth values map to the egb camera works great.

I think I did the depth calibration, its this right?

```bash
rosrun kinect2_calibration kinect2_calibration chess5x7x0.03 calibrate depth
```

Results are in calib_depth.yaml, rightbl?

```yaml
%YAML:1.0
depthShift: -2.3241240937897850e+01
```

Is this milimeters? 2.3 centimeters? does it make sense?

The driver automatically compensates for this shift right?

Miguel

wiedemeyer, 2016-11-18:
Yes, yes, and yes.
Try to set the depthShift to 0 and check if it solves your problem.

miguelriemoliveira, 2016-11-18:
Hello again,

Thank you, thank you and thank you :).

I did as you suggested and now it seems perfect. Take a look:

[image]

[image]

What could be the explanation for this? Is the depth calibration not working properly?

Regards,

Miguel

closed 2016-11-23 by miguelriemoliveira