code-iai/iai_kinect2 #335: Where are the image width and height values in the calib files?

miguelriemoliveira, 2016-11-18:
I was searching for the width and height values, possibly for both the RGB as well as the IR images, but could not find them in any of the four calib files generated after the calibration procedures.

Shouldn't they be somewhere? I mean, the calibrations, at least the intrinsic calibration for sure but possibly all others, are only valid for a particular image size.

Do you just assume its always the calibration for FullHD size?

Thanks,

Miguel

wiedemeyer, 2016-11-25:
The intrinsic parameters are only valid for a certain sensor and the resolution of the sensor is fixed. So there is no need to store them.
The intrinsic parameter for the color sensor are for FullHD, because it is the sensors resolution. The intrinsic parameters for the IR for for its native resolution.

miguelriemoliveira, 2016-11-25:
Hi,

OK, thanks for the response.

closed 2016-11-25 by miguelriemoliveira