code-iai/iai_kinect2 #338: Kinect2 image resize methods

Severn-Vergil, 2016-11-24:
Hi everyone,
    If I want to reshape HD or QHD messages from Kinect 2 to the size of 640*480 like Kinect 1, how can I realize this? Or the parameters of image_height and image_width are provided to tune the size of image?
Thank you in advance!
Severn

wiedemeyer, 2016-11-25:
You cannot change the image size with iai_kinect2. But you can either find a ROS node that does it for you, or implement one yourself.

YuQingL, 2016-11-26:
@Severn-Vergil 
Hi Severn-Vergil 
Sorry to bother. I want to store  color images and depth images with 640*480 resolution. Do you achieve it?  Can you help me?
Thank you in advance!

YuQingL, 2016-11-26:
@wiedemeyer 
Hi wiedemeyer .
I found a strange phenomenon.
”rosrun kinect2_viewer kinect2_viewer sd cloud“，the results of the preservation of the color picture to carry a lot of black noise; 
“rosrun kinect2_viewer kinect2_viewer qhd cloud ”and “rosrun kinect2_viewer kinect2_viewer hd cloud”， color images are normal. 
Why does this happen?Can you help me?
Thanks a lot~

Severn-Vergil, 2016-11-28:
@YuQingL 
This package may works: https://github.com/artivis/ros_imresize

YuQingL, 2016-11-29:
@Severn-Vergil 
Thank you for the kindness!

wiedemeyer, 2016-11-30:
@YuQingL That is normal. It can only calculate the SD color pixel for a valid depth pixel. If there is no depth, then there is no color.

closed 2016-12-13 by wiedemeyer