code-iai/iai_kinect2 #339: kinect2_viewer segfaults

bnjff, 2016-12-01:
I've been battling with an error when running kinect2_viewer. It happens only one machine with Ubuntu 14.04 and ROS Indigo when it tries to render the pointcloud.
Kinect2_bridge, Protonect, cloud rendering using PCL (i.e. pcl_visualizer) all work fine. Does anyone have an idea what could be the problem here? Thanks!

`
(gdb) run
Starting program: /home/user/src/catkin_ws/devel/lib/kinect2_viewer/kinect2_viewer 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
X server found. dri2 connection failed! 
DRM_IOCTL_I915_GEM_APERTURE failed: Invalid argument
Assuming 131072kB available aperture size.
May lead to reduced performance or incorrect rendering.
get chip id failed: -1 [22]
param: 4, val: 0
[New Thread 0x7fffcb55f700 (LWP 22610)]
[ INFO] [main] topic color: /kinect2/qhd/image_color_rect
[ INFO] [main] topic depth: /kinect2/qhd/image_depth_rect
[New Thread 0x7fffca471700 (LWP 22612)]
[New Thread 0x7fffc9c70700 (LWP 22613)]
[New Thread 0x7fffc946f700 (LWP 22614)]
[New Thread 0x7fffc8c6e700 (LWP 22619)]
[ INFO] [main] starting receiver...
[New Thread 0x7fffc3bd3700 (LWP 22646)]
[New Thread 0x7fffc33d2700 (LWP 22647)]
[New Thread 0x7fffc2bd1700 (LWP 22648)]
[New Thread 0x7fffc23d0700 (LWP 22649)]
[New Thread 0x7fffc1bcf700 (LWP 22650)]
[New Thread 0x7fffc13ce700 (LWP 22651)]
[New Thread 0x7fffc0bcd700 (LWP 22652)]
[New Thread 0x7fff97fff700 (LWP 22653)]
ERROR: In /build/buildd/vtk-5.8.0/Rendering/vtkXOpenGLRenderWindow.cxx, line 404
vtkXOpenGLRenderWindow (0x97d5d0): Could not find a decent visual

ERROR: In /build/buildd/vtk-5.8.0/Rendering/vtkXOpenGLRenderWindow.cxx, line 404
vtkXOpenGLRenderWindow (0x97d5d0): Could not find a decent visual

ERROR: In /build/buildd/vtk-5.8.0/Rendering/vtkXOpenGLRenderWindow.cxx, line 404
vtkXOpenGLRenderWindow (0x97d5d0): Could not find a decent visual

ERROR: In /build/buildd/vtk-5.8.0/Rendering/vtkXOpenGLRenderWindow.cxx, line 631
vtkXOpenGLRenderWindow (0x97d5d0): Cannot create GLX context.  Aborting.

Program received signal SIGABRT, Aborted.
0x00007ffff2f60c37 in __GI_raise (sig=sig@entry=6) at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
56	../nptl/sysdeps/unix/sysv/linux/raise.c: No such file or directory.

(gdb) bt
#0  0x00007ffff2f60c37 in __GI_raise (sig=sig@entry=6) at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
#1  0x00007ffff2f64028 in __GI_abort () at abort.c:89
#2  0x00007ffff4b6fd85 in vtkXOpenGLRenderWindow::CreateAWindow() () from /usr/lib/libvtkRendering.so.5.8
#3  0x00007ffff4b6c3b2 in vtkXOpenGLRenderWindow::WindowInitialize() () from /usr/lib/libvtkRendering.so.5.8
#4  0x00007ffff4b6c24d in vtkXOpenGLRenderWindow::Start() () from /usr/lib/libvtkRendering.so.5.8
#5  0x00007ffff4ae5927 in vtkXRenderWindowInteractor::Initialize() () from /usr/lib/libvtkRendering.so.5.8
#6  0x00007ffff51ed22a in pcl::visualization::PCLVisualizer::createInteractor (this=this@entry=0x7bf0d0)
    at /home/user/lib/pcl/visualization/src/pcl_visualizer.cpp:303
#7  0x00007ffff51eff4b in pcl::visualization::PCLVisualizer::PCLVisualizer (this=0x7bf0d0, name="Cloud Viewer", 
    create_interactor=<optimized out>) at /home/user/lib/pcl/visualization/src/pcl_visualizer.cpp:177
#8  0x0000000000438f2a in Receiver::cloudViewer() ()
#9  0x0000000000450e31 in Receiver::start(Receiver::Mode) ()
#10 0x0000000000426761 in main ()

`

wiedemeyer, 2016-12-01:
Maybe it is related to the custom PCL version that you are using. Try to use the one, that comes with the official repositories.

bnjff, 2016-12-06:
Thanks for suggestion! The error remains with the default PCL as well.
```
(gdb) run
Starting program: /home/user/src/catkin_ws/devel/lib/kinect2_viewer/kinect2_viewer 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
X server found. dri2 connection failed! 
DRM_IOCTL_I915_GEM_APERTURE failed: Invalid argument
Assuming 131072kB available aperture size.
May lead to reduced performance or incorrect rendering.
get chip id failed: -1 [22]
param: 4, val: 0
[New Thread 0x7fffcb4de700 (LWP 8859)]
[ INFO] [main] topic color: /kinect2/qhd/image_color_rect
[ INFO] [main] topic depth: /kinect2/qhd/image_depth_rect
[New Thread 0x7fffca3f0700 (LWP 8861)]
[New Thread 0x7fffc9bef700 (LWP 8862)]
[New Thread 0x7fffc93ee700 (LWP 8863)]
[New Thread 0x7fffc8bed700 (LWP 8868)]
[ INFO] [main] starting receiver...
[New Thread 0x7fffc39b9700 (LWP 8895)]
[New Thread 0x7fffc31b8700 (LWP 8896)]
[New Thread 0x7fffc29b7700 (LWP 8897)]
[New Thread 0x7fffc21b6700 (LWP 8898)]
[New Thread 0x7fffc19b5700 (LWP 8899)]
[New Thread 0x7fffc11b4700 (LWP 8900)]
[New Thread 0x7fffc09b3700 (LWP 8901)]
[New Thread 0x7fff9ffff700 (LWP 8902)]
ERROR: In /build/buildd/vtk-5.8.0/Rendering/vtkXOpenGLRenderWindow.cxx, line 404
vtkXOpenGLRenderWindow (0x97db10): Could not find a decent visual

ERROR: In /build/buildd/vtk-5.8.0/Rendering/vtkXOpenGLRenderWindow.cxx, line 404
vtkXOpenGLRenderWindow (0x97db10): Could not find a decent visual

ERROR: In /build/buildd/vtk-5.8.0/Rendering/vtkXOpenGLRenderWindow.cxx, line 404
vtkXOpenGLRenderWindow (0x97db10): Could not find a decent visual

ERROR: In /build/buildd/vtk-5.8.0/Rendering/vtkXOpenGLRenderWindow.cxx, line 631
vtkXOpenGLRenderWindow (0x97db10): Cannot create GLX context.  Aborting.

Program received signal SIGABRT, Aborted.
0x00007ffff2ed0c37 in __GI_raise (sig=sig@entry=6) at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
56	../nptl/sysdeps/unix/sysv/linux/raise.c: No such file or directory.

(gdb) bt
#0  0x00007ffff2ed0c37 in __GI_raise (sig=sig@entry=6) at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
#1  0x00007ffff2ed4028 in __GI_abort () at abort.c:89
#2  0x00007ffff4adfd85 in vtkXOpenGLRenderWindow::CreateAWindow() () from /usr/lib/libvtkRendering.so.5.8
#3  0x00007ffff4adc3b2 in vtkXOpenGLRenderWindow::WindowInitialize() () from /usr/lib/libvtkRendering.so.5.8
#4  0x00007ffff4adc24d in vtkXOpenGLRenderWindow::Start() () from /usr/lib/libvtkRendering.so.5.8
#5  0x00007ffff4a55927 in vtkXRenderWindowInteractor::Initialize() () from /usr/lib/libvtkRendering.so.5.8
#6  0x00007ffff5152ada in pcl::visualization::PCLVisualizer::createInteractor() ()
   from /usr/lib/libpcl_visualization.so.1.7
#7  0x00007ffff5155aef in pcl::visualization::PCLVisualizer::PCLVisualizer(std::string const&, bool) ()
   from /usr/lib/libpcl_visualization.so.1.7
#8  0x000000000043953a in Receiver::cloudViewer() ()
#9  0x0000000000450f5d in Receiver::start(Receiver::Mode) ()
#10 0x0000000000426b81 in main ()
```

wiedemeyer, 2016-12-12:
That seems to be an issue of your system/configuration. Make sure you have an x server running and opengl is supported. You could check if `glxgears` works.

closed 2017-01-06 by wiedemeyer

QinZiwen, 2018-03-11:
@wiedemeyer glxgears can works:
```
$ glxgears 
Running synchronized to the vertical refresh.  The framerate should be
approximately the same as the monitor refresh rate.
301 frames in 5.0 seconds = 60.172 FPS
301 frames in 5.0 seconds = 60.020 FPS
XIO:  fatal IO error 11 (Resource temporarily unavailable) on X server ":0"
      after 54 requests (54 known processed) with 0 events remaining.
```
But this problem still exists:
```
ERROR: In /build/buildd/vtk-5.8.0/Rendering/vtkXOpenGLRenderWindow.cxx, line 404
vtkXOpenGLRenderWindow (0x24b6ae0): Could not find a decent visual
ERROR: In /build/buildd/vtk-5.8.0/Rendering/vtkXOpenGLRenderWindow.cxx, line 404
vtkXOpenGLRenderWindow (0x24b6ae0): Could not find a decent visual
ERROR: In /build/buildd/vtk-5.8.0/Rendering/vtkXOpenGLRenderWindow.cxx, line 631
vtkXOpenGLRenderWindow (0x24b6ae0): Cannot create GLX context.  Aborting.
```