code-iai/iai_kinect2 #340: catkin_make error

epiception, 2016-12-02:
Note: my makefu is a little weak, bear with me.

I got the following while trying to make my catkin_package:

 No rule to make target '/usr/local/lib/libpcl_apps.so', needed by '/home/epiception/catkin_ws/devel/lib/kinect2_viewer/kinect2_viewer'.  Stop.
CMakeFiles/Makefile2:5652: recipe for target 'iai_kinect2/kinect2_viewer/CMakeFiles/kinect2_viewer.dir/all' failed

I understand that libpcl_apps.so was not available at the location. Is there some way to fix this?

wiedemeyer, 2016-12-12:
Try to install pcl.

epiception, 2016-12-12:
My system currently has pcl installed. Also PCLVisualizer class works fine with the Kinect. I'm only facing the problem for kinect2_viewer.

wiedemeyer, 2016-12-12:
normally pcl is installed in `/usr/lib` but in your error it said `/usr/local/lib/` which means you have a custom version of pcl installed. please use the one from the official repositories instead.

closed 2017-01-06 by wiedemeyer

epiception, 2017-02-03:
To help anyone who finds a similar traceback while catkin_make compilation, the trick is to recompile PCL library with cmake and enable the BUILD_apps rule while cmake. I guess it wasn't enabled by default. It was easier to do the same with cmake-gui (ccmake ..)