code-iai/iai_kinect2 #342: "image_color_rect"  zero values where depth is zero

ralpguler, 2016-12-08:
Hi,

Is it possible to obtain "image_color_rect" values that are non-zero in locations where no depth information is available?

I have tried calling the  libfreenect2/registration->apply function with enable_filter as 'False' but this didn't do it.

Thank you.

wiedemeyer, 2016-12-12:
Hi,
there is a simple solution. Create a mask from the depth image and and set all color image pixel to black where the mask is zero.

ralpguler, 2016-12-12:
@wiedemeyer, thank you for your reply.

Perhaps I was misunderstood. By default, the "image_color_rect"(SD) values are zero. I provide an example below(taken from https://github.com/code-iai/iai_kinect2/issues/173) 

What I am trying to do is to obtain non-zero values.

[image]

wiedemeyer, 2016-12-12:
It needs depth information to know which SD color pixel belongs to which HD color pixel. Without depth, no mapping. You can only use the HD/QHD color and depth images if black pixels is a problem in your application.

closed 2016-12-13 by wiedemeyer