code-iai/iai_kinect2 #345: The problem of  "Calibrate the depth measurements"

YuQingL, 2016-12-15:
Hi,
I carried kinect2 calibration according to this site "https://github.com/code-iai/iai_kinect2/tree/master/kinect2_calibration". 
Now I have generated these files：calib_color.yaml   calib_ir.yaml   calib_pose.yaml
Next: "rosrun kinect2_calibration kinect2_calibration chess5x7x0.03 calibrate depth"
Then the following error occurred: 
[ERROR] [DepthCalibration::compareDists] no distance sample data!

Can please someone help me find out what the problem.
My OS is Ubuntu 14.04 with ROS Indigo.

srinimd2005, 2016-12-16:
Check this out 

https://github.com/code-iai/iai_kinect2/issues/243

YuQingL, 2016-12-19:
@srinimd2005 
Thank you for your suggestion!
According to #243,I solved my problem.  
At my first try I just recorded and calibrated 80  color pictures.
After I recorded 150 pictures more for each color, IR and sync images,and then it worked.

closed 2017-01-06 by wiedemeyer

Isabelp, 2017-03-10:
Hi, 

I am having the same problem that AlexxxBeck #243 and YuQingL when I try to calibrate depth (step 9 of the calibration process). Is the depth calibration still a draft? 

I do not need to work with colours but only with the xyz point clouds, so I think I really need to perform this depth calibration, am I right?

I am working on Ubuntu 14.04 with ROS Indigo too.

I would appreciate any guidance.