code-iai/iai_kinect2 #347: Invoking "   catkin_make -DCMAKE_BUILD_TYPE="Release"   "failed

melodybinbin, 2016-12-24:
The terminal dispays as following: 

yuanlibin@yuanlibin:~/catkin_ws/src/iai_kinect2$ cd ~/catkin_ws
yuanlibin@yuanlibin:~/catkin_ws$ catkin_make -DCMAKE_BUILD_TYPE="Release"
Base path: /home/yuanlibin/catkin_ws
Source space: /home/yuanlibin/catkin_ws/src
Build space: /home/yuanlibin/catkin_ws/build
Devel space: /home/yuanlibin/catkin_ws/devel
Install space: /home/yuanlibin/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/yuanlibin/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/yuanlibin/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/indigo
-- This workspace overlays: /opt/ros/indigo
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/yuanlibin/catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.18
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 5 packages in topological order:
-- ~~  - iai_kinect2 (metapackage)
-- ~~  - kinect2_registration
-- ~~  - kinect2_bridge
-- ~~  - kinect2_calibration
-- ~~  - kinect2_viewer
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin metapackage: 'iai_kinect2'
-- ==> add_subdirectory(iai_kinect2/iai_kinect2)
-- +++ processing catkin package: 'kinect2_registration'
-- ==> add_subdirectory(iai_kinect2/kinect2_registration)
CMake Warning at iai_kinect2/kinect2_registration/CMakeLists.txt:31 (find_package):
  By not providing "FindEigen3.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Eigen3", but
  CMake did not find one.

  Could not find a package configuration file provided by "Eigen3" with any
  of the following names:

    Eigen3Config.cmake
    eigen3-config.cmake

  Add the installation prefix of "Eigen3" to CMAKE_PREFIX_PATH or set
  "Eigen3_DIR" to a directory containing one of the above files.  If "Eigen3"
  provides a separate development package or SDK, be sure it has been
  installed.

-- CPU based depth registration disabled
-- OpenCL based depth registration enabled
CMake Warning at iai_kinect2/kinect2_registration/CMakeLists.txt:60 (message):
  Your libOpenCL.so is incompatible with CL/cl.h.  Install ocl-icd-opencl-dev
  to update libOpenCL.so?

-- +++ processing catkin package: 'kinect2_bridge'
-- ==> add_subdirectory(iai_kinect2/kinect2_bridge)
-- Using these message generators: gencpp;genlisp;genpy
-- +++ processing catkin package: 'kinect2_calibration'
-- ==> add_subdirectory(iai_kinect2/kinect2_calibration)
-- +++ processing catkin package: 'kinect2_viewer'
-- ==> add_subdirectory(iai_kinect2/kinect2_viewer)
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   thread
--   date_time
--   iostreams
--   serialization
--   chrono
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
-- checking for module 'openni2-dev'
--   package 'openni2-dev' not found
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
-- checking for module 'openni2-dev'
--   package 'openni2-dev' not found
-- looking for PCL_COMMON
-- looking for PCL_KDTREE
-- looking for PCL_OCTREE
-- looking for PCL_SEARCH
-- looking for PCL_SAMPLE_CONSENSUS
-- looking for PCL_FILTERS
-- looking for PCL_FEATURES
-- looking for PCL_IO
-- looking for PCL_GEOMETRY
-- looking for PCL_VISUALIZATION
-- looking for PCL_OUTOFCORE
-- looking for PCL_REGISTRATION
-- looking for PCL_KEYPOINTS
-- looking for PCL_SURFACE
-- looking for PCL_SEGMENTATION
-- looking for PCL_RECOGNITION
-- looking for PCL_TRACKING
-- looking for PCL_PEOPLE
-- Configuring done
-- Generating done
-- Build files have been written to: /home/yuanlibin/catkin_ws/build
####
#### Running command: "make -j4 -l4" in "/home/yuanlibin/catkin_ws/build"
####
Scanning dependencies of target kinect2_registration
[ 33%] [ 33%] Building CXX object iai_kinect2/kinect2_registration/CMakeFiles/kinect2_registration.dir/src/kinect2_registration.cpp.o
Building CXX object iai_kinect2/kinect2_registration/CMakeFiles/kinect2_registration.dir/src/depth_registration_opencl.cpp.o
Linking CXX shared library /home/yuanlibin/catkin_ws/devel/lib/libkinect2_registration.so
[ 33%] Built target kinect2_registration
Scanning dependencies of target kinect2_calibration
Scanning dependencies of target kinect2_bridge_nodelet
Scanning dependencies of target kinect2_bridge
[ 50%] Scanning dependencies of target kinect2_viewer
Building CXX object iai_kinect2/kinect2_calibration/CMakeFiles/kinect2_calibration.dir/src/kinect2_calibration.cpp.o
[ 66%] [ 83%] Building CXX object iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge.dir/src/kinect2_bridge.cpp.o
Building CXX object iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge_nodelet.dir/src/kinect2_bridge.cpp.o
[100%] Building CXX object iai_kinect2/kinect2_viewer/CMakeFiles/kinect2_viewer.dir/src/viewer.cpp.o
Linking CXX executable /home/yuanlibin/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge
Linking CXX shared library /home/yuanlibin/catkin_ws/devel/lib/libkinect2_bridge_nodelet.so
Linking CXX executable /home/yuanlibin/catkin_ws/devel/lib/kinect2_calibration/kinect2_calibration
[100%] Built target kinect2_bridge_nodelet
/home/yuanlibin/catkin_ws/devel/lib/libkinect2_registration.so: undefined reference to `clRetainDevice'
/home/yuanlibin/catkin_ws/devel/lib/libkinect2_registration.so: undefined reference to `clReleaseDevice'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/yuanlibin/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge] Error 1
make[1]: *** [iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/home/yuanlibin/catkin_ws/devel/lib/libkinect2_registration.so: undefined reference to `clRetainDevice'
/home/yuanlibin/catkin_ws/devel/lib/libkinect2_registration.so: undefined reference to `clReleaseDevice'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/yuanlibin/catkin_ws/devel/lib/kinect2_calibration/kinect2_calibration] Error 1
make[1]: *** [iai_kinect2/kinect2_calibration/CMakeFiles/kinect2_calibration.dir/all] Error 2
Linking CXX executable /home/yuanlibin/catkin_ws/devel/lib/kinect2_viewer/kinect2_viewer
/home/yuanlibin/catkin_ws/devel/lib/libkinect2_registration.so: undefined reference to `clReleaseDevice'
/home/yuanlibin/catkin_ws/devel/lib/libkinect2_registration.so: undefined reference to `clRetainDevice'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/yuanlibin/catkin_ws/devel/lib/kinect2_viewer/kinect2_viewer] Error 1
make[1]: *** [iai_kinect2/kinect2_viewer/CMakeFiles/kinect2_viewer.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
yuanlibin@yuanlibin:~/catkin_ws$

I use the NVIDIA GTX1060,and installed cuda 7.5.
How to solve this problem?

srinimd2005, 2017-01-02:
Have you built and installed libfreenect2? Is protonect is working? Have you built the libfreenect with CUDA?

closed 2017-02-28 by wiedemeyer

melodybinbin, 2017-03-05:
Thanks for your help @srinimd2005 @wiedemeyer

melodybinbin, 2017-03-05:
Thanks for your help @srinimd2005 @wiedemeyer.I used ROS indigo and ubuntu 14.04.

When I built libfreenect2 before,and protonect is working,however terminal showed as following:
……
-- Performing Test OpenCL_CXX_WORKS
-- Performing Test OpenCL_CXX_WORKS - Failed
CMake Warning at CMakeLists.txt:309 (MESSAGE):
  Your libOpenCL.so is incompatible with CL/cl.h.  Install ocl-icd-opencl-dev
  to update libOpenCL.so?
……
-- Feature list:
--   CUDA    yes
--   CXX11    disabled
--   Examples    yes
--   OpenCL    yes but buggy
--   OpenGL    yes
--   OpenNI2    yes
--   TegraJPEG    no
--   Threading    tinythread
--   TurboJPEG    yes
--   VAAPI    no
--   VideoToolbox    no (Apple only)
-- Configuring done
……
So I uninstall the cuda-8.0,and install again(including openCL).Then I rebuild libfreenect2.The terminal shows as following:
……
-- Performing Test OpenCL_CXX_WORKS
-- Performing Test OpenCL_CXX_WORKS - Success
……
-- Feature list:
--   CUDA    yes
--   CXX11    disabled
--   Examples    yes
--   OpenCL    yes　　　　　　　　　　　　　　　
--   OpenGL    yes
--   OpenNI2    yes
--   TegraJPEG    no
--   Threading    tinythread
--   TurboJPEG    yes
--   VAAPI    no
--   VideoToolbox    no (Apple only)
-- Configuring done
……

Finally when I execute the following instructions,it is successful as following:
yuanlibin@yuanlibin:~/catkin_ws$ catkin_make -DCMAKE_BUILD_TYPE="Release"
……
[100%] Building CXX object iai_kinect2/kinect2_viewer/CMakeFiles/kinect2_viewer.dir/src/viewer.cpp.o
Linking CXX shared library /home/yuanlibin/catkin_ws/devel/lib/libkinect2_bridge_nodelet.so
Linking CXX executable /home/yuanlibin/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge
Linking CXX executable /home/yuanlibin/catkin_ws/devel/lib/kinect2_calibration/kinect2_calibration
[100%] Built target kinect2_bridge_nodelet
[100%] Built target kinect2_bridge
[100%] Built target kinect2_calibration
Linking CXX executable /home/yuanlibin/catkin_ws/devel/lib/kinect2_viewer/kinect2_viewer
[100%] Built target kinect2_viewer
yuanlibin@yuanlibin:~/catkin_ws$