code-iai/iai_kinect2 #348: kinect2_calibration: Duration of Calibration Process

cgsch, 2016-12-28:
Hi folks,

I very much hope that I didn't overlook any familiar topic so far, but I have a question referring the Kinect-One calibration using the iai_kinect2 package in combination with libfreenect2.
After I spent several hours in thinking about how to properly position the chessboard pattern in the field of view of the camera and googling around, I already passed four calibration sets without receiving any satisfying result in the way of a proper RGB-D image (following the README.md primarily). So far the calibration sets contained somewhat from 40 to 140 frames per set until I came across @wiedemeyer's post in issue #311 where he explains his method of calibration (one of the last posts).
After trying this out and getting up to 440 frames I had to actually confine my work on just taking the images and calibrate later on due to time issues (because I noticed in the past that calibrating the datas on my MacBook Pro which runs Ubuntu 16.04 LTS can easily take half an hour if there are somewhat around 100 frames to process in each step (color, ir and sync)). So I started the color images' calibration process on the next day.

**Unfortunatly I had to notice that even 24 hours later the calibration process haven't been terminated.**

Has anybody of you guys noticed the same kind of problem with your calibration method?
In fact I had to calibrate 2600 images in total (sum of color, ir and sync images) which (in addition of the 1300 .txt-files containing the detected points) makes approximately 900 MB of data to process while calibrating. 

I know that I use Ubuntu 16.04 LTS with ROS Kinetic which should be flawlessly working with libfreenect2 but is not mentioned as compatible with iai_kinect2. Does somebody know if it is not mentioned because it is in fact not working with 16.04 and Kinetic or just because it isn't tested to be so? I'm struggling with asking this question because I made everything to be installed (ROS Kinetic, libfreenect2 and iai_kinect2) without any warning or error occurring. Si far this seemed to work fine for me until I now got the above mentioned problem.

Can anybody help me or has any further questions which can lead to a solution?

Best regards.

srinimd2005, 2017-01-02:
I had the similar problem but indeed I had only 250 images per calibration set. I too noticed that it took so long to finish the calibration with 250 images per set (intrinsic calibration  - appr. 2.5 Hrs, extrensic calibration - more than 12 Hrs). So I guess you need more than 24 Hrs for calibrating the extrensic parameter especially sync calibration with 440 image per set.

cgsch, 2017-01-02:
Is this the way to do it, surely? With approximately 120 frames, my laptop took about half an hour for computing intrinsics as well as extrinsics (for each step, meaning color, ir and sync (therefore about 1,5h in total)). This let's me assume that there is no linear relation between the number of input-images and the time of termination. If my laptop really needs 24 hours for calculating 250 frames, I rather don't have a guess at the termination time with 400 frames... 
Did you, @srinimd2005 , noticed a cpu usage of not more than 40%, too? Have you achieved a more cpu-effective way to use the calibration algorithm? Regards.

srinimd2005, 2017-01-02:
I too assume the same that there is no linear relation between number of images and the calibration process as it is increasing exponentially. And w.r.t  to my CPU usage, I am using a Intel Core i7 processor 3 gen and I found that one of my CPU core (out of 4) was consuming 50% all the time during the process. I have not explored other methods of calibration as the result what I achieved was satisfactory. I guess the processing time could be reduced if we could somehow use multiple threads for this process because from the CPU usage only one of the CPU core was computing while the others are idle.

wiedemeyer, 2017-01-06:
The calibration is done by OpenCV and is single threaded. There are termination criteria parameters set in the code of kinect2_calibration and you could lower them if needed. Also too many images may not increase calibration results much but on the other hand they increase computation time a lot. I think around 100 to 150 images per set should be sufficient.

closed 2017-02-28 by wiedemeyer

leoujz, 2017-03-08:
can you guys detect checkerboard on IR image successfully?

JimTheKiwi, 2017-03-08:
@leoujz  If you are not seeing the checkerboard on IR at all, then maybe see https://github.com/code-iai/iai_kinect2/issues/80#issuecomment-94461306 "some inks might be transparent to IR. Are you using an ink printer? Could you try it on a laser printer?"

leoujz, 2017-03-17:
> checkerboard on IR at all, then maybe see #80 (comment)

Thank you very much, a ink printer is used, now I have tried laser printer
and pattern can be detected. Thank you again.