code-iai/iai_kinect2 #350: "roslaunch kinect2_bridge kinect2_bridge.launch"process has died

jasoninsh, 2017-01-04:
better-nan@better-nan:~/catkin_ws/src$ roslaunch kinect2_bridge kinect2_bridge.launch
... logging to /home/better-nan/.ros/log/00fb7ee6-d185-11e6-8a8f-507b9d0d6caa/roslaunch-better-nan-28551.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://better-nan:44823/

SUMMARY
========

PARAMETERS
 * /kinect2_bridge/base_name: kinect2
 * /kinect2_bridge/base_name_tf: kinect2
 * /kinect2_bridge/bilateral_filter: True
 * /kinect2_bridge/calib_path: /home/better-nan/...
 * /kinect2_bridge/depth_device: -1
 * /kinect2_bridge/depth_method: default
 * /kinect2_bridge/edge_aware_filter: True
 * /kinect2_bridge/fps_limit: -1.0
 * /kinect2_bridge/jpeg_quality: 90
 * /kinect2_bridge/max_depth: 12.0
 * /kinect2_bridge/min_depth: 0.1
 * /kinect2_bridge/png_level: 1
 * /kinect2_bridge/publish_tf: False
 * /kinect2_bridge/queue_size: 5
 * /kinect2_bridge/reg_device: -1
 * /kinect2_bridge/reg_method: default
 * /kinect2_bridge/sensor: 
 * /kinect2_bridge/use_png: False
 * /kinect2_bridge/worker_threads: 4
 * /kinect2_points_xyzrgb_hd/queue_size: 5
 * /kinect2_points_xyzrgb_qhd/queue_size: 5
 * /kinect2_points_xyzrgb_sd/queue_size: 5
 * /rosdistro: jade
 * /rosversion: 1.11.20

NODES
  /
    kinect2 (nodelet/nodelet)
    kinect2_bridge (nodelet/nodelet)
    kinect2_points_xyzrgb_hd (nodelet/nodelet)
    kinect2_points_xyzrgb_qhd (nodelet/nodelet)
    kinect2_points_xyzrgb_sd (nodelet/nodelet)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[kinect2-1]: started with pid [28569]
process[kinect2_bridge-2]: started with pid [28570]
process[kinect2_points_xyzrgb_sd-3]: started with pid [28571]
process[kinect2_points_xyzrgb_qhd-4]: started with pid [28572]
process[kinect2_points_xyzrgb_hd-5]: started with pid [28574]
[ INFO] [1483428178.156375545]: Loading nodelet /kinect2_bridge of type kinect2_bridge/kinect2_bridge_nodelet to manager kinect2 with the following remappings:
[ INFO] [1483428178.163859929]: waitForService: Service [/kinect2/load_nodelet] has not been advertised, waiting...
[ INFO] [1483428178.167715275]: Initializing nodelet with 4 worker threads.
[ INFO] [1483428178.185237977]: waitForService: Service [/kinect2/load_nodelet] is now available.
[ INFO] [1483428178.308906772]: [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: default
        fps_limit: -1
       calib_path: /home/better-nan/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: default
     depth_device: -1
       reg_method: default
       reg_device: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 5
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 12 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:9 with serial 041823341247
[Info] [Freenect2Impl] found 1 devices
[ INFO] [1483428178.461390028]: [Kinect2Bridge::initDevice] Kinect2 devices found: 
[ INFO] [1483428178.461451286]: [Kinect2Bridge::initDevice]   0: 041823341247 (selected)
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[ INFO] [1483428178.562527358]: [Kinect2Bridge::initDevice] starting kinect2
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
[ INFO] [1483428179.193046295]: [Kinect2Bridge::initDevice] device serial: 041823341247
[ INFO] [1483428179.193111052]: [Kinect2Bridge::initDevice] device firmware: 4.0.3911.0
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] canceling rgb transfers...
[Info] [Freenect2DeviceImpl] canceling depth transfers...
[Info] [Freenect2DeviceImpl] stopped
[FATAL] [1483428183.405699393]: Failed to load nodelet '/kinect2_bridge` of type `kinect2_bridge/kinect2_bridge_nodelet` to manager `kinect2'
[FATAL] [1483428183.405699792]: Failed to load nodelet '/kinect2_points_xyzrgb_qhd` of type `depth_image_proc/point_cloud_xyzrgb` to manager `kinect2'
[FATAL] [1483428183.406130693]: Failed to load nodelet '/kinect2_points_xyzrgb_sd` of type `depth_image_proc/point_cloud_xyzrgb` to manager `kinect2'
[kinect2-1] process has died [pid 28569, exit code -11, cmd /opt/ros/jade/lib/nodelet/nodelet manager __name:=kinect2 __log:=/home/better-nan/.ros/log/00fb7ee6-d185-11e6-8a8f-507b9d0d6caa/kinect2-1.log].
log file: /home/better-nan/.ros/log/00fb7ee6-d185-11e6-8a8f-507b9d0d6caa/kinect2-1*.log
[kinect2_bridge-2] process has died [pid 28570, exit code 255, cmd /opt/ros/jade/lib/nodelet/nodelet load kinect2_bridge/kinect2_bridge_nodelet kinect2 __name:=kinect2_bridge __log:=/home/better-nan/.ros/log/00fb7ee6-d185-11e6-8a8f-507b9d0d6caa/kinect2_bridge-2.log].
log file: /home/better-nan/.ros/log/00fb7ee6-d185-11e6-8a8f-507b9d0d6caa/kinect2_bridge-2*.log
[kinect2_points_xyzrgb_sd-3] process has died [pid 28571, exit code 255, cmd /opt/ros/jade/lib/nodelet/nodelet load depth_image_proc/point_cloud_xyzrgb kinect2 rgb/camera_info:=kinect2/sd/camera_info rgb/image_rect_color:=kinect2/sd/image_color_rect depth_registered/image_rect:=kinect2/sd/image_depth_rect depth_registered/points:=kinect2/sd/points __name:=kinect2_points_xyzrgb_sd __log:=/home/better-nan/.ros/log/00fb7ee6-d185-11e6-8a8f-507b9d0d6caa/kinect2_points_xyzrgb_sd-3.log].
log file: /home/better-nan/.ros/log/00fb7ee6-d185-11e6-8a8f-507b9d0d6caa/kinect2_points_xyzrgb_sd-3*.log
[kinect2_points_xyzrgb_qhd-4] process has died [pid 28572, exit code 255, cmd /opt/ros/jade/lib/nodelet/nodelet load depth_image_proc/point_cloud_xyzrgb kinect2 rgb/camera_info:=kinect2/qhd/camera_info rgb/image_rect_color:=kinect2/qhd/image_color_rect depth_registered/image_rect:=kinect2/qhd/image_depth_rect depth_registered/points:=kinect2/qhd/points __name:=kinect2_points_xyzrgb_qhd __log:=/home/better-nan/.ros/log/00fb7ee6-d185-11e6-8a8f-507b9d0d6caa/kinect2_points_xyzrgb_qhd-4.log].
log file: /home/better-nan/.ros/log/00fb7ee6-d185-11e6-8a8f-507b9d0d6caa/kinect2_points_xyzrgb_qhd-4*.log
[kinect2_bridge-2] restarting process
process[kinect2_bridge-2]: started with pid [28735]
[kinect2_points_xyzrgb_sd-3] restarting process
process[kinect2_points_xyzrgb_sd-3]: started with pid [28736]
[kinect2_points_xyzrgb_qhd-4] restarting process
process[kinect2_points_xyzrgb_qhd-4]: started with pid [28737]
[ INFO] [1483428183.692018921]: Loading nodelet /kinect2_bridge of type kinect2_bridge/kinect2_bridge_nodelet to manager kinect2 with the following remappings:
[ INFO] [1483428183.702122555]: waitForService: Service [/kinect2/load_nodelet] could not connect to host [better-nan:56913], waiting...
[kinect2_points_xyzrgb_hd-5] process has finished cleanly
log file: /home/better-nan/.ros/log/00fb7ee6-d185-11e6-8a8f-507b9d0d6caa/kinect2_points_xyzrgb_hd-5*.log
[kinect2_points_xyzrgb_hd-5] restarting process
process[kinect2_points_xyzrgb_hd-5]: started with pid [29185]
 
I have cuda and add CUDA to my  environment.
Thanks for any help!

wiedemeyer, 2017-01-06:
try to run the kinect2_bridge node directly with rosrun. If that doesn't give you a meaningful error message, compile it in debug and run it in gdb.

yalan2017, 2017-02-10:
Hi 
I can run libfreenect2 without any problem, but for for iai_kinect2 part when i would to run "roslaunch kinect2_bridge kinect2_bridge.launch." i face with this error: 
Thank you for your great job

process[kinect2_points_xyzrgb_qhd-5]: started with pid [23020]
[kinect2_points_xyzrgb_hd-6] restarting process
process[kinect2_points_xyzrgb_hd-6]: started with pid [23021]
[ INFO] [1486694772.768844716]: Loading nodelet /kinect2_bridge of type kinect2_bridge/kinect2_bridge_nodelet to manager kinect2 with the following remappings:
[ INFO] [1486694772.775450046]: waitForService: Service [/kinect2/load_nodelet] could not connect to host [Cyber:44516], waiting...

yalan2017, 2017-02-10:
hi 
i updated my nvidia driver now i have such error! any problem with opencl? how should i fix it
thank you

X server found. dri2 connection failed! 
DRM_IOCTL_I915_GEM_APERTURE failed: Invalid argument
Assuming 131072kB available aperture size.
May lead to reduced performance or incorrect rendering.
get chip id failed: -1 [22]
param: 4, val: 0
beignet-opencl-icd: no supported GPU found, this is probably the wrong opencl-icd package for this hardware
(If you have multiple ICDs installed and OpenCL works, you can ignore this message)
[ INFO] [1486730543.585667517]: [DepthRegistrationOpenCL::init] devices:
[ INFO] [1486730543.585887325]: [DepthRegistrationOpenCL::init]   0: GeForce GTX 750 Ti
[ INFO] [1486730543.586001202]: [DepthRegistrationOpenCL::init] selected device: GeForce GTX 750 Ti
X server found. dri2 connection failed! 
DRM_IOCTL_I915_GEM_APERTURE failed: Invalid argument
Assuming 131072kB available aperture size.
May lead to reduced performance or incorrect rendering.
get chip id failed: -1 [22]
param: 4, val: 0
beignet-opencl-icd: no supported GPU found, this is probably the wrong opencl-icd package for this hardware
(If you have multiple ICDs installed and OpenCL works, you can ignore this message)
[ INFO] [1486730543.669790701]: [DepthRegistrationOpenCL::init] devices:
[ INFO] [1486730543.669894752]: [DepthRegistrationOpenCL::init]   0: GeForce GTX 750 Ti
[ INFO] [1486730543.669996153]: [DepthRegistrationOpenCL::init] selected device: GeForce GTX 750 Ti
[ INFO] [1486730543.773662791]: [Kinect2Bridge::main] waiting for clients to connect

wiedemeyer, 2017-02-28:
I don't know, but you could try to uninstall beignet if you are unsing nvidia.

luffy1996, 2017-03-05:
Hie @wiedemeyer , The problem exists with rosrun as well . Protonet is working well and fine.

wiedemeyer, 2017-03-08:
@luffy1996 which problem? the one from yalan2017 or jasoninsh?
You can also select the registration and depth processing method by a parameter. Try out the cpu methods.

luffy1996, 2017-03-09:
@wiedemeyer the problem is it gets stuck here. I can not find out why this is getting stuck here.  The Protonect works out fine and there was 100% building in catkin_make. I will be grateful to you for your help.

####################################################################
luffy@raftel:~/catkin_ws$ roslaunch kinect2_bridge kinect2_bridge.launch
... logging to /home/luffy/.ros/log/e4ae0360-0462-11e7-8033-d0bf9c9c8ded/roslaunch-raftel-28036.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://raftel:37508/

SUMMARY
========

PARAMETERS
 * /kinect2_bridge/base_name: kinect2
 * /kinect2_bridge/base_name_tf: kinect2
 * /kinect2_bridge/bilateral_filter: True
 * /kinect2_bridge/calib_path: /home/luffy/catki...
 * /kinect2_bridge/depth_device: -1
 * /kinect2_bridge/depth_method: default
 * /kinect2_bridge/edge_aware_filter: True
 * /kinect2_bridge/fps_limit: -1.0
 * /kinect2_bridge/jpeg_quality: 90
 * /kinect2_bridge/max_depth: 12.0
 * /kinect2_bridge/min_depth: 0.1
 * /kinect2_bridge/png_level: 1
 * /kinect2_bridge/publish_tf: False
 * /kinect2_bridge/queue_size: 5
 * /kinect2_bridge/reg_device: -1
 * /kinect2_bridge/reg_method: default
 * /kinect2_bridge/sensor: 
 * /kinect2_bridge/use_png: False
 * /kinect2_bridge/worker_threads: 4
 * /kinect2_points_xyzrgb_hd/queue_size: 5
 * /kinect2_points_xyzrgb_qhd/queue_size: 5
 * /kinect2_points_xyzrgb_sd/queue_size: 5
 * /rosdistro: indigo
 * /rosversion: 1.11.20

NODES
  /
    kinect2 (nodelet/nodelet)
    kinect2_bridge (nodelet/nodelet)
    kinect2_points_xyzrgb_hd (nodelet/nodelet)
    kinect2_points_xyzrgb_qhd (nodelet/nodelet)
    kinect2_points_xyzrgb_sd (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [28048]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to e4ae0360-0462-11e7-8033-d0bf9c9c8ded
process[rosout-1]: started with pid [28061]
started core service [/rosout]
process[kinect2-2]: started with pid [28074]
process[kinect2_bridge-3]: started with pid [28079]
process[kinect2_points_xyzrgb_sd-4]: started with pid [28080]
process[kinect2_points_xyzrgb_qhd-5]: started with pid [28086]
[ INFO] [1489020845.331584020]: Loading nodelet /kinect2_bridge of type kinect2_bridge/kinect2_bridge_nodelet to manager kinect2 with the following remappings:
[ INFO] [1489020845.340072227]: waitForService: Service [/kinect2/load_nodelet] has not been advertised, waiting...
process[kinect2_points_xyzrgb_hd-6]: started with pid [28100]
[ INFO] [1489020845.383206742]: Initializing nodelet with 4 worker threads.
[ INFO] [1489020845.390147274]: waitForService: Service [/kinect2/load_nodelet] is now available.
[ INFO] [1489020845.520706658]: [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: default
        fps_limit: -1
       calib_path: /home/luffy/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: default
     depth_device: -1
       reg_method: default
       reg_device: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 5
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4
[Info] [CudaDepthPacketProcessorImpl] device 0: GeForce 940M @ 1176MHz Memory 2047MB
[Info] [CudaDepthPacketProcessorImpl] selected device 0
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 10 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:5 with serial 027237744647
[Info] [Freenect2Impl] found 1 devices
[ INFO] [1489020845.847186692]: [Kinect2Bridge::initDevice] Kinect2 devices found: 
[ INFO] [1489020845.847305140]: [Kinect2Bridge::initDevice]   0: 027237744647 (selected)
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[ INFO] [1489020845.944943374]: [Kinect2Bridge::initDevice] starting kinect2
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
[ INFO] [1489020846.578533018]: [Kinect2Bridge::initDevice] device serial: 027237744647
[ INFO] [1489020846.578573819]: [Kinect2Bridge::initDevice] device firmware: 4.0.3911.0
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] canceling rgb transfers...
[Info] [Freenect2DeviceImpl] canceling depth transfers...
[Info] [Freenect2DeviceImpl] stopped
[ WARN] [1489020846.862482658]: [Kinect2Bridge::initCalibration] using sensor defaults for color intrinsic parameters.
[ WARN] [1489020846.862532271]: [Kinect2Bridge::initCalibration] using sensor defaults for ir intrinsic parameters.
[ WARN] [1489020846.862548159]: [Kinect2Bridge::initCalibration] using defaults for rotation and translation.
[ WARN] [1489020846.862563228]: [Kinect2Bridge::initCalibration] using defaults for depth shift.
[ INFO] [1489020846.934800168]: [DepthRegistration::New] Using CPU registration method!
[ INFO] [1489020846.934929107]: [DepthRegistration::New] Using CPU registration method!
[ INFO] [1489020847.021904940]: [Kinect2Bridge::main] waiting for clients to connect

wiedemeyer, 2017-03-09:
It is not stuck. I don't know how often I answered to this before, but a search in the issues should come up with many posts. `waiting for clients to connect` means that everything is fine, it just waits for topic subscriptions.

closed 2017-03-09 by wiedemeyer

yalan2017, 2017-03-10:
hi i just reinstall my ros and its working now , thank you for your great job

nabeelyousfi, 2018-01-17:
when i am trying to run this command infinte loop is running 
roslaunch kinect2_bridge kinect2_bridge.launch  _depth_method:=cpu _reg_method:=cpu

it shows some errors like 

process[kinect2_bridge-2]: started with pid [10616]
[ INFO] [1516188394.849046764]: Loading nodelet /kinect2_bridge of type kinect2_bridge/kinect2_bridge_nodelet to manager kinect2 with the following remappings:
[ INFO] [1516188394.941377152]: [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: default
        fps_limit: -1
       calib_path: /home/nabeel/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: default
     depth_device: -1
       reg_method: default
       reg_device: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 5
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4
beignet-opencl-icd: no supported GPU found, this is probably the wrong opencl-icd package for this hardware
(If you have multiple ICDs installed and OpenCL works, you can ignore this message)
[Info] [OpenCLDepthPacketProcessorImpl]  devices:
[Error] [OpenCLDepthPacketProcessorImpl] could not find any suitable device
[Error] [OpenCLAllocator] b->buffer = cl::Buffer(context, CL_MEM_WRITE_ONLY | CL_MEM_ALLOC_HOST_PTR, size, NULL, &err): -34
[Error] [OpenCLAllocator] b->buffer = cl::Buffer(context, CL_MEM_WRITE_ONLY | CL_MEM_ALLOC_HOST_PTR, size, NULL, &err): -34
libva info: VA-API version 0.39.0
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_0_39
libva info: va_openDriver() returns 0
[Info] [VaapiRgbPacketProcessorImpl] driver: Intel i965 driver for Intel(R) Sandybridge Mobile - 1.7.0
[Error] [VaapiRgbPacketProcessorImpl] vaQueryConfigEntrypoints(display, VAProfileJPEGBaseline, entrypoints, &num_entrypoints): the requested VAProfile is not supported
[Error] [OpenCLAllocator] b->buffer = cl::Buffer(context, CL_MEM_READ_ONLY | CL_MEM_ALLOC_HOST_PTR, size, NULL, &err): -34
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 8 usb devices connected
[Info] [Freenect2Impl] found 0 devices
[ERROR] [1516188394.945173788]: [Kinect2Bridge::initDevice] no Kinect2 devices found!
[Error] [OpenCLAllocator] queue.enqueueUnmapMemObject(b->buffer, b->data, NULL, &event): -36
[Error] [OpenCLAllocator] queue.enqueueUnmapMemObject(b->buffer, b->data, NULL, &event): -36
[Error] [OpenCLAllocator] queue.enqueueUnmapMemObject(b->buffer, b->data, NULL, &event): -36
[ERROR] [1516188394.945611821]: [Kinect2Bridge::start] Initialization failed!
[FATAL] [1516188394.967567244]: Failed to load nodelet '/kinect2_bridge` of type `kinect2_bridge/kinect2_bridge_nodelet` to manager `kinect2'
[kinect2_bridge-2] process has died [pid 10616, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load kinect2_bridge/kinect2_bridge_nodelet kinect2 __name:=kinect2_bridge __log:=/home/nabeel/.ros/log/4008d6aa-fb79-11e7-878c-60d819231e93/kinect2_bridge-2.log].
log file: /home/nabeel/.ros/log/4008d6aa-fb79-11e7-878c-60d819231e93/kinect2_bridge-2*.log