code-iai/iai_kinect2 #353: Where can I learn the algorithm

XeMinZa, 2017-01-09:
well,I got the parameters of my kinect2 and the result from visualization is ok! But I don't want to use ROS and how can I realiza it with C++. 
## Only I want to do is using the parameter to transform my image. 
Where can I learn the algorithm in this part, the source code is too complex for me to read.

wiedemeyer, 2017-01-09:
The registration method from libfreenect2 does not need those parameters, you can simply use it. And if you really want to understand the algorithm, then you need read the source code.

XeMinZa, 2017-01-09:
@wiedemeyer thanks. I have completed my code. The registration is really simply to use.

closed 2017-01-09 by XeMinZa

SunnZhang, 2017-03-14:
Hi, there! I got some issues about the registration method too, I wonder if you can help me.

XeMinZa, 2017-03-14:
What is problem? I didn't use the iai to registrate the ir image with rgb image. What I used is the freenect2 API. That's really simple, and in my situation, it perform good enough for me.