code-iai/iai_kinect2 #354: kinect2_viewer catkin_make error

dinghal, 2017-01-12:
when i catkin_make "kinect2_viewer: appear the following error:
/usr/include/pcl-1.7/pcl/visualization/pcl_visualizer.h:283:88: error: macro "PCL_DEPRECATED" requires 2 arguments, but only 1 given
 "addCoordinateSystem (scale, id, viewport) with id a unique string identifier.")
/usr/include/pcl-1.7/pcl/visualization/pcl_visualizer.h:283:88: error: macro "PCL_DEPRECATED" requires 2 arguments, but only 1 given
"addCoordinateSystem (scale, id, viewport) with id a unique string identifier.")

how i to solve this problem?

wiedemeyer, 2017-02-28:
Maybe you are using a wrong PCL version? What OS, libs, etc. are you using?

la9881275, 2017-05-10:
@wiedemeyer Running “rosrun kinect2_viewer kinect2_viewer” yields "Segmentation fault (core dumped)". I can't fix it after days of efforts. My computer is equipped with Ubuntu14.04 and PCL1.7*. Can you help me？ Thank you！