code-iai/iai_kinect2 #355: can the iR value be zero if the x,y,z depth value are valid

mpatalberta, 2017-01-13:
I am using the intensity value to help filter data out. I am noticing that many of the valid x,y,z coordinates have a corresponding zero value for intensity. How is this possible?

mpatalberta, 2017-01-13:
Below is the code where it appears the IR information is remapped.
Since both the depth camera and ir camera are one and the same why do we need to do this remap?
    // IR
    if(status[IR_SD] || status[IR_SD_RECT])
    {
      cv::flip(images[IR_SD], images[IR_SD], 1);
    }
    if(status[IR_SD_RECT])
    {
      cv::remap(images[IR_SD], images[IR_SD_RECT], map1Ir, map2Ir, cv::INTER_AREA);
    }

mpatalberta, 2017-01-14:
I am unclear why IR_SD and IR_RECT are not the same. I am also not clear why any type of remap is required for either the IR_SD and DEPTH_SD. Do we not map the rgb pixel to the depth map?

wiedemeyer, 2017-01-15:
The remap is for rectification. The only difference is, that for ir the values are interpolated (see cv::INTER_AREA) and for depth nearest neighbor is used. If you want to check for valid pixels either use the depth image and not the ir or change the interpolation method for ir in the code to nearest neighbor. Or use the non rectified ir and depth topics, these are more or less the raw images from the kinect2.

mpatalberta, 2017-01-15:
Why do you need to do a INTER_AREA to get the IR_RECT?

mpatalberta, 2017-01-16:
I have tracked down that my kinectv2 has 2 row that are dead( return niether depth or intensity).
I checked two of my cameras : one was 848 no intensity and the other is 130 intensity values that are dead.  In the two cameras the dead location are on the edge.

An option in the interepolation of the SD_RECT needs to be added for None or near.

closed 2017-01-16 by mpatalberta