code-iai/iai_kinect2 #356: Program received signal SIGSEGV, Segmentation fault.

jeasinema, 2017-01-15:
I've compiled `iai_freenect`, and I've install essential dependencies(`libfreenect2`, `image-pipeline`, etc).But when I try to launch `kinect_bridge`, It still **crashed**. The same when I using `rosrun`. Then I run it directly using `gdb`, and here are what I got:

```
(gdb) r
Starting program: /home/mxj/catkin_wb/devel/lib/kinect2_bridge/kinect2_bridge 
/home/mxj/catkin_wb/devel/lib/kinect2_bridge/kinect2_bridge: /usr/local/cuda/lib64/libOpenCL.so.1: no version information available (required by /home/mxj/catkin_wb/devel/lib/libkinect2_registration.so)
/home/mxj/catkin_wb/devel/lib/kinect2_bridge/kinect2_bridge: /usr/local/cuda/lib64/libOpenCL.so.1: no version information available (required by /home/mxj/catkin_wb/devel/lib/libkinect2_registration.so)
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffe0476700 (LWP 32634)]
[New Thread 0x7fffdef3c700 (LWP 32637)]
[New Thread 0x7fffde73b700 (LWP 32638)]
[New Thread 0x7fffddf3a700 (LWP 32639)]
[New Thread 0x7fffdd739700 (LWP 32644)]
[New Thread 0x7fffcffff700 (LWP 32645)]
[ INFO] [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: default
        fps_limit: -1
       calib_path: /home/mxj/catkin_wb/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opengl
     depth_device: -1
       reg_method: opencl
       reg_device: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4
[New Thread 0x7fffc6b49700 (LWP 32665)]
[New Thread 0x7fffc6348700 (LWP 32666)]
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 11 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:15 with serial 075027740847
[Info] [Freenect2Impl] found 1 devices
[ INFO] [Kinect2Bridge::initDevice] Kinect2 devices found: 
[ INFO] [Kinect2Bridge::initDevice]   0: 075027740847 (selected)
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[ INFO] [Kinect2Bridge::initDevice] starting kinect2
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
[ INFO] [Kinect2Bridge::initDevice] device serial: 075027740847
[ INFO] [Kinect2Bridge::initDevice] device firmware: 4.0.3911.0
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] canceling rgb transfers...
[Info] [Freenect2DeviceImpl] canceling depth transfers...
[Info] [Freenect2DeviceImpl] stopped
[ WARN] [Kinect2Bridge::initCalibration] using sensor defaults for color intrinsic parameters.
[ WARN] [Kinect2Bridge::initCalibration] using sensor defaults for ir intrinsic parameters.
[ WARN] [Kinect2Bridge::initCalibration] using defaults for rotation and translation.
[ WARN] [Kinect2Bridge::initCalibration] using defaults for depth shift.
[ INFO] [DepthRegistration::New] Using OpenCL registration method!
[ INFO] [DepthRegistration::New] Using OpenCL registration method!
X server found. dri2 connection failed! 
DRM_IOCTL_I915_GEM_APERTURE failed: Invalid argument
Assuming 131072kB available aperture size.
May lead to reduced performance or incorrect rendering.
get chip id failed: -1 [22]
param: 4, val: 0

Program received signal SIGSEGV, Segmentation fault.
0x00007fffb971d4e0 in drm_intel_bufmgr_gem_enable_reuse ()
   from /usr/lib/x86_64-linux-gnu/libdrm_intel.so.1
(gdb) bt
#0  0x00007fffb971d4e0 in drm_intel_bufmgr_gem_enable_reuse ()
   from /usr/lib/x86_64-linux-gnu/libdrm_intel.so.1
#1  0x00007fffbae16b58 in ?? () from /usr/lib/beignet/libcl.so
#2  0x00007fffbae16d07 in ?? () from /usr/lib/beignet/libcl.so
#3  0x00007fffbae16e21 in ?? () from /usr/lib/beignet/libcl.so
#4  0x00007fffbae16f08 in ?? () from /usr/lib/beignet/libcl.so
#5  0x00007fffbae1245d in ?? () from /usr/lib/beignet/libcl.so
#6  0x00007fffbae12521 in ?? () from /usr/lib/beignet/libcl.so
#7  0x00007ffff58ca027 in cl::Platform::getDevices(unsigned long, std::vector<cl::Device, std::allocator<cl::Device> >*) const ()
   from /home/mxj/catkin_wb/devel/lib/libkinect2_registration.so
#8  0x00007ffff58c2a04 in getDevices(std::vector<cl::Platform, std::allocator<cl::Platform> > const&, std::vector<cl::Device, std::allocator<cl::Device> >&) ()
   from /home/mxj/catkin_wb/devel/lib/libkinect2_registration.so
#9  0x00007ffff58c2eac in DepthRegistrationOpenCL::init(int) ()
   from /home/mxj/catkin_wb/devel/lib/libkinect2_registration.so
#10 0x00007ffff58bd74f in DepthRegistration::init(cv::Mat const&, cv::Size_<int> const&, cv::Mat const&, cv::Size_<int> const&, cv::Mat const&, cv::Mat const&, cv::Mat const&, float, float, int) ()
   from /home/mxj/catkin_wb/devel/lib/libkinect2_registration.so
#11 0x00000000004131d5 in Kinect2Bridge::initRegistration(std::string const&, int, double) ()
#12 0x0000000000429341 in Kinect2Bridge::initialize() ()
---Type <return> to continue, or q <return> to quit---
#13 0x000000000042a415 in Kinect2Bridge::start() ()
#14 0x000000000040e7da in main ()
(gdb) 
(gdb) frame 1
#1  0x00007fffbae16b58 in ?? () from /usr/lib/beignet/libcl.so
(gdb) frame 0
#0  0x00007fffb971d4e0 in drm_intel_bufmgr_gem_enable_reuse ()
   from /usr/lib/x86_64-linux-gnu/libdrm_intel.so.1
(gdb) info register
rax            0x0	0
rbx            0x18a5430	25842736
rcx            0x0	0
rdx            0x7fffb9934820	140736306825248
rsi            0x0	0
rdi            0x0	0
rbp            0x1	0x1
rsp            0x7fffffffbae8	0x7fffffffbae8
r8             0x0	0
r9             0xffffe6da	4294960858
r10            0x7ffff5150be0	140737305185248
r11            0x0	0
r12            0x7fffffffbb70	140737488337776
r13            0x18a5430	25842736
r14            0x7fffffffbc60	140737488338016
r15            0x7fffffffd048	140737488343112
rip            0x7fffb971d4e0	0x7fffb971d4e0 <drm_intel_bufmgr_gem_enable_reuse>
eflags         0x10206	[ PF IF RF ]
cs             0x33	51
ss             0x2b	43
ds             0x0	0
es             0x0	0
```

mpatalberta, 2017-01-16:
Try the cpu only version does that work?

wiedemeyer, 2017-01-19:
Seems to be an issue with your Beignet installation. First of all try the CPU version as @mpatalberta suggested with the parameter `reg_method:=cpu`. Also you could try to use the Beignet version that the libfreenect README points at.

closed 2017-02-28 by wiedemeyer