code-iai/iai_kinect2 #357: FPS higher than 30

JimmyDaSilva, 2017-01-17:
Hello @wiedemeyer, 
I see that with my machine the depth processing can go up to 60 FPS thanks to the cuda "mode" of the library.
Could it be possible to get the ros image topics published at higher rates ?

For instance, I see that whith the QVGA mode in openni2_launch, you can choose 60FPS if you're willing to get half the resolution.

wiedemeyer, 2017-01-19:
Hi @JimmyDaSilva,
as fas as I know, the Kinect2 is only sending ir and color images with 30 FPS, with bad lighting the color image frame rate even goes down to 15 FPS. I don't know of any way to increase the output FPS of the device.
The processing speed measure only says how many frames it could theoretically process in a second given the time it took to process a single frame. For example, on a 980 Ti the depth processing takes 1 ms, so it could process 1000 frames per second.

JimmyDaSilva, 2017-01-19:
Thanks for the reply @wiedemeyer .
Too bad...

closed 2017-01-19 by JimmyDaSilva

s880367, 2017-04-06:
Hi @JimmyDaSilva 
Can I ask you which version of cuda did you install on your machine ?
I'm using Ubuntu 14.04 and Nvidia970 but both cuda 7.5 and 6.5 were failed in catkin_make.
Thanks.

JimmyDaSilva, 2017-04-06:
CUDA 8.0 installation worked great for me on different machines (laptop, desktop) and different OS (ubuntu 16.04 and 14.04)
https://developer.nvidia.com/cuda-downloads