code-iai/iai_kinect2 #358: qt5 Error while cmake step

pallavbakshi, 2017-01-18:
```
/usr/bin/ld: cannot find -lQt5::Core
/usr/bin/ld: cannot find -lQt5::Gui
/usr/bin/ld: cannot find -lQt5::Widgets
/usr/bin/ld: cannot find -lQt5::Test
/usr/bin/ld: cannot find -lQt5::Concurrent
/usr/bin/ld: cannot find -lQt5::OpenGL
collect2: error: ld returned 1 exit status
make[2]: *** [/home/pb/catkin_ws/devel/lib/libkinect2_registration.so] Error 1
make[1]: *** [iai_kinect2/kinect2_registration/CMakeFiles/kinect2_registration.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
```
I get the above error when trying the cmake step. I have already installed all the dependencies. I have also installed qt5 but in the home directory. If the solution is to link it to the library then kindly tell me the steps to do so. I have already searched google and stackoverlfow for the issue. There are some similar cases but they simply tell me to add link to the qt5 library while making the process BUT THAT DOESN'T WORK. 

Thanks!! :dancing_men:

wiedemeyer, 2017-01-19:
Hi,
I there seems to be something wrong with your environment. This package does not use / need QT at all and I don't know why it is trying to link it. Which OS and library versions are you using? Maybe it's a dependency from OpenCV or PCL?

pallavbakshi, 2017-01-19:
Hey,
I am using Ubuntu 14.04, Indigo ros, master iaikinect, OpenCV 2.4.10, PCL 1.7.2

wiedemeyer, 2017-01-19:
Then try to use the PCL version that comes with the ubuntu repositories.

pallavbakshi, 2017-01-19:
okay.. that will PCL 1.8

wiedemeyer, 2017-01-19:
If I remember correctly, it was using 1.7.1.

pallavbakshi, 2017-01-19:
Okay then I am downloading these versions:
Opencv 2.4.8 - http://packages.ubuntu.com/source/trusty/opencv
PCL version unknown but will follow these instructions - http://askubuntu.com/questions/625854/pcl-ubuntu-installation

pallavbakshi, 2017-01-19:
Hey, 

I solved that problem somehow. I didn't reinstall OpenCV or PCL. I will update my solution on this page once I can confirm if it is working properly. But now I am stuck at

`[kinect2_bridge.launch] is neither a launch file in package [kinect2_bridge] nor is [kinect2_bridge] a launch file name
The traceback for the exception was written to the log file
`

Now, there is another person who raised the same [issue] () but the solution you suggested over there doesn't work. I have already tried sourced the workspace and the ros setup.

pallavbakshi, 2017-01-20:
Hi,

It turns out it didn't install. I had to reinstall the OpenCV and PCL but still getting the same error. Any other thing I can do. This time I downloaded PCL from repository and OpenCV 2.9 because the one on repository is 3.xx

pallavbakshi, 2017-01-23:
Hey,

I know you have no obligation to reply to the issues raised. Can you still kindly help me out with this? I already did what you asked me for - uninstalling OpenCV, PCL and reinstalling them. But none of them worked out. 

Now I have fresh installed Ubuntu 14.04 and downloaded ROS Indigo again. I have downloaded libfreenect2. However, I haven't downloaded OpenCV yet. Can you PLEASE precisely point out the OpenCV version you tested your code with last time. I can download that so as to make it work this time. It will be even better if you can point me out to PCL version you have used as well. However, I think PCL gets installed automatically with ROS. 

Please help me out! I can help you with documentation in return once I complete my project (some pro bono work).

Thanks,
Pallav

wiedemeyer, 2017-01-24:
I meant you should use the versions that come with Ubuntu/ROS. So simply do an `apt-get install ...`.
And the previous error appears if the ROS workspace is not sourced in the current terminal.

pallavbakshi, 2017-01-26:
thanks! I did a fresh install and then didn't download opencv or pcl. Both of these libraries come along with ROS so you don't need to install it separately. If anyone gets a similar error in future then tweet me a message.

closed 2017-01-26 by pallavbakshi

dhiegomaga, 2017-10-28:
I had the exact same issue, Ubuntu 16.04. 
You sure it doesn't depend on Qt ? 
I had installed the .deb package of PCL from[ this website](https://larrylisky.com/2016/11/03/point-cloud-library-on-ubuntu-16-04-lts/). So instead, I installed the Qt dependency: 

` sudo apt-get install libvtk5.10-qt4 libvtk5.10 libvtk5-dev`

After this the error disappeared. But another one started: 

```
Could not find a package configuration file provided by "cv_bridge" with
  any of the following names:

    cv_bridgeConfig.cmake
    cv_bridge-config.cmake
```
I had OpenCV 3.2.0 previously installed on the machine (`python; import cv2; cv2.__version__` gave `opencv 3.2.0`). So what I did was download, compile and install opencv 2.4.13 at **/usr/local**. Maybe it messed with that, but I don't know.

mptangtang, 2018-11-28:
Have you ever solved this problem? I have the exact problem as you have described.

mptangtang, 2018-11-28:
My error is exactly the same with you.

> ```
> /usr/bin/ld: cannot find -lQt5::Core
> /usr/bin/ld: cannot find -lQt5::Gui
> /usr/bin/ld: cannot find -lQt5::Widgets
> /usr/bin/ld: cannot find -lQt5::Test
> /usr/bin/ld: cannot find -lQt5::Concurrent [...]