code-iai/iai_kinect2 #360: Waiting for clients to connect

adilbonzi, 2017-02-01:
adil@HP-Pavilion-g6-Notebook-PC:~/catkin_ws$ catkin_make -DCMAKE_BUILD_TYPE="Debug"
Base path: /home/adil/catkin_ws
Source space: /home/adil/catkin_ws/src
Build space: /home/adil/catkin_ws/build
Devel space: /home/adil/catkin_ws/devel
Install space: /home/adil/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/adil/catkin_ws/build"
####
####
#### Running command: "make -j4 -l4" in "/home/adil/catkin_ws/build"
####
[  4%] Built target openni_camera_gencfg
[ 59%] Built target openni_driver
[ 72%] Built target kinect2_registration
[ 77%] Built target openni_node
[ 81%] [ 86%] Built target openni_nodelet
Built target kinect2_bridge_nodelet
[ 90%] Built target kinect2_bridge
[ 95%] Built target kinect2_calibration
[100%] Built target kinect2_viewer
adil@HP-Pavilion-g6-Notebook-PC:~/catkin_ws$ cd devel/lib/kinect2_bridge
adil@HP-Pavilion-g6-Notebook-PC:~/catkin_ws/devel/lib/kinect2_bridge$ gdb kinect2_bridge
GNU gdb (Ubuntu 7.7.1-0ubuntu5~14.04.2) 7.7.1
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from kinect2_bridge...done.
(gdb) // inside gdb: run until it crashes and do a backtrace
Undefined command: "".  Try "help".
(gdb) run
Starting program: /home/adil/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffddc2d700 (LWP 12389)]
[New Thread 0x7fffdcb3e700 (LWP 12391)]
[New Thread 0x7fffd7fff700 (LWP 12392)]
[New Thread 0x7fffd77fe700 (LWP 12393)]
[New Thread 0x7fffd6ffd700 (LWP 12398)]
[New Thread 0x7fffd6012700 (LWP 12399)]
[ INFO] [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: default
        fps_limit: -1
       calib_path: /home/adil/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opencl
     depth_device: -1
       reg_method: opencl
       reg_device: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4
[Info] [OpenCLDepthPacketProcessorImpl]  devices:
[Info] [OpenCLDepthPacketProcessorImpl]   0: Intel(R) HD Graphics IvyBridge M GT2 (GPU)[Intel]
[Info] [OpenCLDepthPacketProcessorImpl] selected device: Intel(R) HD Graphics IvyBridge M GT2 (GPU)[Intel]
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[New Thread 0x7fffbffff700 (LWP 12419)]
[New Thread 0x7fffbf7fe700 (LWP 12420)]
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 12 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:10 with serial 031845735147
[Info] [Freenect2Impl] found 1 devices
[ INFO] [Kinect2Bridge::initDevice] Kinect2 devices found: 
[ INFO] [Kinect2Bridge::initDevice]   0: 031845735147 (selected)
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[ INFO] [Kinect2Bridge::initDevice] starting kinect2
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
[ INFO] [Kinect2Bridge::initDevice] device serial: 031845735147
[ INFO] [Kinect2Bridge::initDevice] device firmware: 4.0.3916.0
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] canceling rgb transfers...
[Info] [Freenect2DeviceImpl] canceling depth transfers...
[Info] [Freenect2DeviceImpl] stopped
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[ WARN] [Kinect2Bridge::initCalibration] using sensor defaults for color intrinsic parameters.
[ WARN] [Kinect2Bridge::initCalibration] using sensor defaults for ir intrinsic parameters.
[ WARN] [Kinect2Bridge::initCalibration] using defaults for rotation and translation.
[ WARN] [Kinect2Bridge::initCalibration] using defaults for depth shift.
[ INFO] [DepthRegistration::New] Using OpenCL registration method!
[ INFO] [DepthRegistration::New] Using OpenCL registration method!
[ INFO] [DepthRegistrationOpenCL::init] devices:
[ INFO] [DepthRegistrationOpenCL::init]   0: Intel(R) HD Graphics IvyBridge M GT2
[ INFO] [DepthRegistrationOpenCL::init] selected device: Intel(R) HD Graphics IvyBridge M GT2
[ INFO] [DepthRegistrationOpenCL::init] devices:
[ INFO] [DepthRegistrationOpenCL::init]   0: Intel(R) HD Graphics IvyBridge M GT2
[ INFO] [DepthRegistrationOpenCL::init] selected device: Intel(R) HD Graphics IvyBridge M GT2
[New Thread 0x7fffb7815700 (LWP 12458)]
[New Thread 0x7fffaf014700 (LWP 12459)]
[New Thread 0x7fffb7014700 (LWP 12460)]
[New Thread 0x7fffb6813700 (LWP 12461)]
[New Thread 0x7fffb6012700 (LWP 12462)]
[ INFO] [Kinect2Bridge::main] waiting for clients to connect

I followed the FAQ page. I access my libfreenect2 folder and go into build folder and access protonect from bin folder. From there the protonect is working perfectly by showing the IR, depth, thermal and real imaging. But when I follow the FAQ's, and access the examples, I cannot access any protonect files and it gives me this error,

"adil@HP-Pavilion-g6-Notebook-PC:~/libfreenect2/examples$ ./Protonect gl
bash: ./Protonect: No such file or directory." 

Im using Intel® Core™ i5-3210M CPU @ 2.50GHz × 4 and my graphics is Intel® Ivybridge Mobile with 64 bit OS. Kindly advise me on this matter. As i have been trying to run kinect for my project for 3D mapping.

Tobias-Fischer, 2017-02-01:
Seems like all is working fine - just go ahead and connect a client ;-). For example, open rviz and add the published point cloud data (don't forget to change the fixed frame if necessary). Once there is a node requesting data, the node will start publishing them. Alternatively, just do a `rostopic echo /kinect2/X` (e.g. `rostopic echo /kinect2/hd/camera_info`) and this will achieve the same.

closed 2017-02-03 by adilbonzi

HalleyJiang, 2017-08-13:
ERROR: Unable to communicate with master!

RRRaymond, 2020-09-06:
> Seems like all is working fine - just go ahead and connect a client ;-). For example, open rviz and add the published point cloud data (don't forget to change the fixed frame if necessary). Once [...]

thanks a lot!