code-iai/iai_kinect2 #362: Segfault when running kinect2_bridge

dwhit, 2017-02-01:
Hi I'm getting a weird error. I had iai_kinect2 installed and running correctly, but yesterday after rebuilding my catkin_ws, running kinect2_bridge fails. I recompiled in debug mode and ran bridge in gdb, as has been suggested in other issue responses, and recieved the following output (backtrace at the bottom). It seems to be an opencv issue, which is strange, since it's using the opencv ROS provides.

(gdb) r
Starting program: /home/dwhit/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge 
/home/dwhit/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge: /usr/local/cuda/lib64/libOpenCL.so.1: no version information available (required by /home/dwhit/catkin_ws/devel/lib/libkinect2_registration.so)
/home/dwhit/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge: /usr/local/cuda/lib64/libOpenCL.so.1: no version information available (required by /home/dwhit/catkin_ws/devel/lib/libkinect2_registration.so)
/home/dwhit/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge: /usr/local/cuda/lib64/libOpenCL.so.1: no version information available (required by /home/dwhit/freenect2/lib/libfreenect2.so.0.2)
/home/dwhit/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge: /usr/local/cuda/lib64/libOpenCL.so.1: no version information available (required by /home/dwhit/freenect2/lib/libfreenect2.so.0.2)
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffe697b700 (LWP 18374)]
[New Thread 0x7fffe617a700 (LWP 18375)]
[New Thread 0x7fffe5979700 (LWP 18376)]
[New Thread 0x7fffe5178700 (LWP 18381)]
[New Thread 0x7fffd7fff700 (LWP 18382)]
[New Thread 0x7fffd77fe700 (LWP 18383)]
[ INFO] [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: default
        fps_limit: -1
       calib_path: /home/dwhit/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: cuda
     depth_device: -1
       reg_method: opencl
       reg_device: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4
[New Thread 0x7fffd4972700 (LWP 18404)]
[Thread 0x7fffd4972700 (LWP 18404) exited]
[New Thread 0x7fffd4972700 (LWP 18405)]
[New Thread 0x7fffc10f8700 (LWP 18408)]
[Info] [CudaDepthPacketProcessorImpl] device 0: GeForce GTX 1070 @ 1683MHz Memory 8110MB
[Info] [CudaDepthPacketProcessorImpl] selected device 0
[New Thread 0x7fffc08f7700 (LWP 18409)]
[New Thread 0x7fffbbfff700 (LWP 18410)]
libva info: VA-API version 0.39.0
libva info: va_getDriverName() returns -1
libva error: va_getDriverName() failed with unknown libva error,driver_name=(null)
[Error] [VaapiRgbPacketProcessorImpl] vaInitialize(display, &major_ver, &minor_ver): unknown libva error
[New Thread 0x7fffbb7fe700 (LWP 18411)]
[New Thread 0x7fffbaffd700 (LWP 18412)]
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 9 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:5 with serial 500450742442
[Info] [Freenect2Impl] found 1 devices
[ INFO] [Kinect2Bridge::initDevice] Kinect2 devices found: 
[ INFO] [Kinect2Bridge::initDevice]   0: 500450742442 (selected)
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[ INFO] [Kinect2Bridge::initDevice] starting kinect2
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
[ INFO] [Kinect2Bridge::initDevice] device serial: 500450742442
[ INFO] [Kinect2Bridge::initDevice] device firmware: 4.0.3916.0
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] canceling rgb transfers...
[Info] [Freenect2DeviceImpl] canceling depth transfers...
[Info] [Freenect2DeviceImpl] stopped

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff714b642 in cv::Mat::copyTo(cv::_OutputArray const&) const ()
   from /usr/lib/x86_64-linux-gnu/libopencv_core.so.2.4
(gdb) bt
#0  0x00007ffff714b642 in cv::Mat::copyTo(cv::_OutputArray const&) const ()
   from /usr/lib/x86_64-linux-gnu/libopencv_core.so.2.4
#1  0x000000000044acdf in cv::Mat::clone (this=0x7fffffffc9e8)
    at /opt/ros/indigo/include/opencv-3.1.0-dev/opencv2/core/mat.inl.hpp:643
#2  0x0000000000454779 in Kinect2Bridge::initDevice (this=0x7fffffffc830, sensor="500450742442")
    at /home/dwhit/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:621
#3  0x000000000044f92f in Kinect2Bridge::initialize (this=0x7fffffffc830)
    at /home/dwhit/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:305
#4  0x000000000044d4e1 in Kinect2Bridge::start (this=0x7fffffffc830)
    at /home/dwhit/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:152
#5  0x0000000000448a46 in main (argc=1, argv=0x7fffffffd7b8)
    at /home/dwhit/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:1623
(gdb) Quit

wiedemeyer, 2017-02-02:
seems like the error is related to VA-API:
```
libva info: VA-API version 0.39.0
libva info: va_getDriverName() returns -1
libva error: va_getDriverName() failed with unknown libva error,driver_name=(null)
[Error] [VaapiRgbPacketProcessorImpl] vaInitialize(display, &major_ver, &minor_ver): unknown libva error
```
You can try to build libfreenect2 without vaapi support or fix the issue with it.

dwhit, 2017-02-02:
I was getting that error when it was working as well, so I don't think that is what caused it. I ended up fixing the issue by creating a new catkin_ws and reinstalling packages one at a time. One of them depends on opencv3 and so I would end up with opencv3 in /opt/ros/indigo/share. I manually removed the opencv3 folder from there and rebuilt iai_kinect2, and it works, for both cuda and opencl. VA-API definitely doesn't work, but TurboJPEG seems fine for now, I'm still getting 30hz for both color and depth.

closed 2017-02-02 by dwhit