code-iai/iai_kinect2 #365: iai_kinect2 /dev/dri/card0 not authenticated

rastaxe, 2017-02-03:
I have a problem with iai_kinect2 package (Indigo). I can successfully use Kinect V2 on my Intel NUC. I am not able to use it when the computer does not have the monitor plugged (through ssh). When I launch `roslaunch kinect2_bridge kinect2_bridge.launch`

I got this error:

> /dev/dri/card0 not authenticated

Is it possible to run kinect2_bridge without a screen monitor?

mpatalberta, 2017-02-05:
YES you can run the kinect system without a screen.
You have 2 approaches. The first is to try to change udev and allow the video driver be part of your group. The problem is /dev/dri/card0 is owned by root and video. Put yourself in the video group this may help.

The second me this is to write a system service (which has initially root privilege ) and the inherit some of the user environment. This is what I have done and this works.

rastaxe, 2017-02-06:
Thanks for your help. Adding the user to video group was not enough. 
Following [this](https://github.com/code-iai/iai_kinect2/issues/63), I tried enabling the drm render nodes by editing /etc/default/grub so that
`GRUB_CMDLINE_LINUX_DEFAULT="quiet splash drm.rnodes=1"`
and after `update-grub` it worked!

closed 2017-02-28 by wiedemeyer