code-iai/iai_kinect2 #369: Cannot compile with OpenCL enabled

samarth-robo, 2017-02-19:
Hi, I get the following error in `kinect2_bridge` when I try to compile iai_kinect2 with OpenCL enabled:
`
/home/samarth/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp: In member function ‘bool Kinect2Bridge::initPipeline(const string&, int32_t)’:
/home/samarth/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:427:28: error: expected type-specifier
       packetPipeline = new libfreenect2::OpenCLKdePacketPipeline(device);
                            ^
/home/samarth/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp: In member function ‘bool Kinect2Bridge::initPipeline(const string&, int32_t)’:
/home/samarth/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:427:28: error: expected type-specifier
       packetPipeline = new libfreenect2::OpenCLKdePacketPipeline(device);
`
Does someone know what is going wrong?

I am using ROS Kinectic, Ubuntu 16.04. I successfully compiled libfreenect2 v0.2.0. Here's the CMake output from that:
`
-- using stdlib as threading library
-- Linking with these libraries: 
 /usr/lib/x86_64-linux-gnu/libusb-1.0.so
 va-drm
 va
 /usr/lib/x86_64-linux-gnu/libjpeg.so
 /usr/lib/x86_64-linux-gnu/libturbojpeg.so.0
 /usr/lib/x86_64-linux-gnu/libglfw.so
 /usr/lib/x86_64-linux-gnu/libGL.so
 /opt/intel/opencl/libOpenCL.so
-- Could NOT find Doxygen (missing:  DOXYGEN_EXECUTABLE) 
-- Configurating examples
-- Using in-tree freenect2 target
-- Feature list:
--   CUDA    disabled
--   CXX11    yes
--   Examples    yes
--   OpenCL    yes
--   OpenGL    yes
--   OpenNI2    yes
--   TegraJPEG    disabled
--   Threading    std::thread
--   TurboJPEG    yes
--   VAAPI    yes
--   VideoToolbox    no (Apple only)
-- Configuring done
-- Generating done
-- Build files have been written to: /home/samarth/libraries/libfreenect2/build
`

I have installed OpenCL directly from Intel's website, here's the output of `clinfo`:
`
Number of platforms                               1
  Platform Name                                   Intel(R) OpenCL
  Platform Vendor                                 Intel(R) Corporation
  Platform Version                                OpenCL 2.0 
  Platform Profile                                FULL_PROFILE
  Platform Extensions                             cl_khr_3d_image_writes cl_khr_byte_addressable_store cl_khr_depth_images cl_khr_fp64 cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_icd cl_khr_image2d_from_buffer cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_spir
  Platform Extensions function suffix             INTEL

  Platform Name                                   Intel(R) OpenCL
Number of devices                                 2
  Device Name                                     Intel(R) HD Graphics
  Device Vendor                                   Intel(R) Corporation
  Device Vendor ID                                0x8086
  Device Version                                  OpenCL 2.0 
  Driver Version                                  r4.0.59481
  Device OpenCL C Version                         OpenCL C 2.0 
  Device Type                                     GPU
  Device Profile                                  FULL_PROFILE
  Max compute units                               23
  Max clock frequency                             1000MHz
  Device Partition                                (core)
    Max number of sub-devices                     0
    Supported partition types                     by <unknown> (0x7F5E00000000)
  Max work item dimensions                        3
  Max work item sizes                             256x256x256
  Max work group size                             256
  Preferred work group size multiple              32
  Preferred / native vector sizes                 
    char                                                16 / 16      
    short                                                8 / 8       
    int                                                  4 / 4       
    long                                                 1 / 1       
    half                                                 8 / 8        (cl_khr_fp16)
    float                                                1 / 1       
    double                                               1 / 1        (cl_khr_fp64)
`
(plus a lot more, I can share it if you want)

iai_kinect2 seems to recognize OpenCL OK. Here's the CMake output from when it was compiling `kinect2_registration`:
`
-- Using CATKIN_DEVEL_PREFIX: /home/samarth/catkin_ws/devel_isolated
-- Using CMAKE_PREFIX_PATH: /home/samarth/catkin_ws/devel_isolated;/opt/ros/kinetic
-- This workspace overlays: /home/samarth/catkin_ws/devel_isolated;/opt/ros/kinetic
-- Using PYTHON_EXECUTABLE: /home/samarth/libraries/anaconda2/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/samarth/catkin_ws/build_isolated/kinect2_registration/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.4
-- Looking for CL_VERSION_2_0
-- Looking for CL_VERSION_2_0 - found
-- Found OpenCL: /opt/intel/opencl/libOpenCL.so (found version "2.0") 
-- CPU based depth registration enabled
-- OpenCL based depth registration enabled
-- Performing Test OpenCL_C_WORKS
-- Performing Test OpenCL_C_WORKS - Success
-- Performing Test OpenCL_CXX_WORKS
-- Performing Test OpenCL_CXX_WORKS - Success
-- Configuring done
-- Generating done
-- Build files have been written to: /home/samarth/catkin_ws/build_isolated/kinect2_registration
`
But it also gave warnings:
`
In file included from /home/samarth/catkin_ws/src/iai_kinect2/kinect2_registration/src/depth_registration_opencl.cpp:33:0:
/opt/intel/opencl/include/CL/cl.hpp: In constructor ‘cl::Image1D::Image1D(const cl::Context&, cl_mem_flags, cl::ImageFormat, size_t, void*, cl_int*)’:
/opt/intel/opencl/include/CL/cl.hpp:3582:9: warning: missing braces around initializer for ‘_cl_image_desc::<anonymous union>’ [-Wmissing-braces]
         };
         ^
/opt/intel/opencl/include/CL/cl.hpp: In constructor ‘cl::Image1DBuffer::Image1DBuffer(const cl::Context&, cl_mem_flags, cl::ImageFormat, size_t, const cl::Buffer&, cl_int*)’:
/opt/intel/opencl/include/CL/cl.hpp:3656:9: warning: missing braces around initializer for ‘_cl_image_desc::<anonymous union>’ [-Wmissing-braces]
         };
         ^
/opt/intel/opencl/include/CL/cl.hpp: In constructor ‘cl::Image1DArray::Image1DArray(const cl::Context&, cl_mem_flags, cl::ImageFormat, size_t, size_t, size_t, void*, cl_int*)’:
/opt/intel/opencl/include/CL/cl.hpp:3717:9: warning: missing braces around initializer for ‘_cl_image_desc::<anonymous union>’ [-Wmissing-braces]
         };
         ^
/opt/intel/opencl/include/CL/cl.hpp: In constructor ‘cl::Image2D::Image2D(const cl::Context&, cl_mem_flags, cl::ImageFormat, size_t, size_t, size_t, void*, cl_int*)’:
/opt/intel/opencl/include/CL/cl.hpp:3804:13: warning: missing braces around initializer for ‘_cl_image_desc::<anonymous union>’ [-Wmissing-braces]
             };
             ^
/opt/intel/opencl/include/CL/cl.hpp: In constructor ‘cl::Image2DArray::Image2DArray(const cl::Context&, cl_mem_flags, cl::ImageFormat, size_t, size_t, size_t, size_t, size_t, void*, cl_int*)’:
/opt/intel/opencl/include/CL/cl.hpp:3983:9: warning: missing braces around initializer for ‘_cl_image_desc::<anonymous union>’ [-Wmissing-braces]
         };
         ^
/opt/intel/opencl/include/CL/cl.hpp: In constructor ‘cl::Image3D::Image3D(const cl::Context&, cl_mem_flags, cl::ImageFormat, size_t, size_t, size_t, size_t, size_t, void*, cl_int*)’:
/opt/intel/opencl/include/CL/cl.hpp:4073:13: warning: missing braces around initializer for ‘_cl_image_desc::<anonymous union>’ [-Wmissing-braces]
             };
             ^
`

Thanks a lot for creating this library!

samarth-robo, 2017-02-20:
Solved: `kinect2_bridge` is looking for `OpenCL**Kde**PacketPipeline`, which is only available in the master branch of `libfreenect2`, not the v0.2.0 tag.

Switching to master branch and recompiling `libfreenect2` solved the problem!

closed 2017-02-20 by samarth-robo

arnabsinha, 2018-08-23:
I have changed the branch as suggested by samarth-robot, but still having the same problem