code-iai/iai_kinect2 #370: running under ssh

alexge233, 2017-02-21:
Hi,

I have gotten the package to compile, calibrated the camera and works fine but only when run under X11.
Running from ssh (even after I enable X11 `ssh -X`) gives libGL and GLFW errors:

```
libGL error: No matching fbConfigs or visuals found
libGl error: failed to load driver: swrast
```

But I only want to run the bridge to acquire depth data and point cloud data, I won't be using X11 or any kind of desktop environment. 
How can I run the bridge for this scenario?

Many thanks,
Alex

wiedemeyer, 2017-02-28:
I think there are already some issues with this. You can also try to use OpenCL/Cuda instead of OpenGL. Or have a look at VirtualGL.

alexge233, 2017-03-02:
Hi,

I opened a [similar issue on libfreenect2](https://github.com/OpenKinect/libfreenect2/issues/800) and their response was to simply rebuild `libfreenect2` without GL.
Seems to be working, without even using SSH -Y -X.

closed 2017-03-02 by alexge233