code-iai/iai_kinect2 #371: kinect2_viewer [DepthRegistrationOpenCL::init] could not find any suitable device

BHAASKARR, 2017-02-24:
Hi All,

When i try to install ROS Indigo with libfreenect2 and iai_kinect2 in my computer with
_Operating System: Ubuntu 14.04.5 LTS
Kernel: Linux 4.4.0-64-generic_
I followed all steps mentioned in
    http://wiki.ros.org/indigo/Installation/Ubuntu
    http://wiki.ros.org/catkin/Tutorials/create_a_workspace
    https://github.com/openkinect/libfreenect2
    (Note: I haven't installed optional softwares
    Example : **Install OpenCL (optional) which is not installed**)
    **_Protonect is working and showing color, depth and ir images as I got output when i
    Run the test program: ./bin/Protonect_**
    https://github.com/code-iai/iai_kinect2
    I am not getting any output when i run
    _roslaunch kinect2_bridge kinect2_bridge.launch_
    but i got following output.

[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Info] [Freenect2DeviceImpl] canceling depth transfers...
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Info] [Freenect2DeviceImpl] stopped
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Error] [OpenCLDepthPacketProcessorImpl] program = cl::Program(context, source, &err): -34
[ WARN] [1487965098.877783656]: [Kinect2Bridge::initCalibration] using sensor defaults for color intrinsic parameters.
[ WARN] [1487965098.877814869]: [Kinect2Bridge::initCalibration] using sensor defaults for ir intrinsic parameters.
[ WARN] [1487965098.877830038]: [Kinect2Bridge::initCalibration] using defaults for rotation and translation.
[ WARN] [1487965098.877847897]: [Kinect2Bridge::initCalibration] using defaults for depth shift.
[ INFO] [1487965098.955753303]: [DepthRegistration::New] Using OpenCL registration method!
[ INFO] [1487965098.955800060]: [DepthRegistration::New] Using OpenCL registration method!
beignet-opencl-icd: no supported GPU found, this is probably the wrong opencl-icd package for this hardware
(If you have multiple ICDs installed and OpenCL works, you can ignore this message)
[ INFO] [1487965098.967884971]: [DepthRegistrationOpenCL::init] devices:
[ERROR] [1487965098.967923702]: [DepthRegistrationOpenCL::init] could not find any suitable device
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] releasing usb interfaces...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed
[ERROR] [1487965098.968220168]: [Kinect2Bridge::start] Initialization failed!
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] already closed, doing nothing
[Error] [OpenCLAllocator] queue.enqueueUnmapMemObject(b->buffer, b->data, NULL, &event): -36
[Error] [OpenCLAllocator] queue.enqueueUnmapMemObject(b->buffer, b->data, NULL, &event): -36
[Error] [OpenCLAllocator] queue.enqueueUnmapMemObject(b->buffer, b->data, NULL, &event): -36
[FATAL] [1487965099.069387343]: Failed to load nodelet '/kinect2_bridgeof typekinect2_bridge/kinect2_bridge_nodeletto managerkinect2'
^C[kinect2_points_xyzrgb_hd-6] killing on exit
[kinect2_points_xyzrgb_qhd-5] killing on exit
[kinect2_points_xyzrgb_sd-4] killing on exit
[kinect2_bridge-3] killing on exit
[kinect2-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

Your help much appreciated.

Thanks,
Bhaskar.

BHAASKARR, 2017-02-24:
@Pincello @wiedemeyer 
Please help me to solve this

wiedemeyer, 2017-02-28:
If you have OpenCL installed it will try to use it. Just set it using the arguments of the launch file. Set `depth_method` and `reg_method``to cpu.

BHAASKARR, 2017-02-28:
@wiedemeyer  I haven't installed OpenCL as it is optional, so how to solve the problem without installing OpenCL.

BHAASKARR, 2017-03-01:
@wiedemeyer  :  also I am getting error as below when i am trying to launch 

bhaskar@D125-40806Linux:~/catkin_ws$ _roslaunch kinect2_bridge kinect2_bridge.launch_
[kinect2_bridge.launch] is neither a launch file in package [kinect2_bridge] nor is [kinect2_bridge] a launch file name
The traceback for the exception was written to the log file

BHAASKARR, 2017-03-01:
Thank you @wiedemeyer 
I am able to see the output without OpenCL
by running the below commands in 3 different command prompts
Linux:~$ roscore 
Linux:~/catkin_ws$ rosrun kinect2_bridge kinect2_bridge _depth_method:=opengl _reg_method:=cpu
Linux:~$ rosrun kinect2_viewer kinect2_viewer

wiedemeyer, 2017-03-02:
Good to know that it works for you now.

closed 2017-03-02 by wiedemeyer

leviresende, 2019-05-08:
I had the same issue, so I edited the parameters to:
<arg name="depth_method"      default="opengl"/> 
<arg name="reg_method"        default="cpu"/>

After that, it is working well.

leviresende, 2019-05-08:
I had the same issue, so I edited the parameters to:

**depth_method to opengl
reg_method to cpu**

After that, it is working well.