code-iai/iai_kinect2 #374: Calibration issues about cannot saving color and depth images

Allen-jl, 2017-02-28:
Hi guys, ima doing the calibration of Kinect v2, however, when i follow the detailed steps, i found i could just save ir images.
Every time I run: rosrun kinect2_calibration kinect2_calibration chess5x7x0.03 record color, and after activating the window, it just can't save images after i pressing space
But it works when I run: rosrun kinect2_calibration kinect2_calibration chess5x7x0.03 record ir, which means I can save images when i press space...

hoping you guys could help me solve this problem, thx

wiedemeyer, 2017-02-28:
The pattern needs to be detected in order to save images. If it is detected a colored overlay is shown in the window.

Allen-jl, 2017-03-01:
@wiedemeyer Hi, thx for your reply
The problem is in my color/depth mode, there is no colored overlay. But in ir mode, the calibration board could be detected and a colored overlay is shown in the window.
I wonder why.

Allen-jl, 2017-03-02:
@wiedemeyer Finally the pattern could be detected... (maybe I just put the pattern a little far from the camera...) anyway, thx man!

closed 2017-03-02 by Allen-jl