code-iai/iai_kinect2 #375: Can not start second device when running two Kinects

easypeet, 2017-03-01:
Hello folks,
I have been using iai_kinect2 to run one Kinect v2 for some time now and yesterday I tried adding a second one. What I did was to run two instances of the kinect2_bridge.launch file with different launch parameters. I did the following in two separate terminals:

Terminal 1:
roslaunch kinect2_bridge kinect2_bridge.launch sensor:=033012150245 base_name:=kinectNew publish_tf:=true

Terminal 2:
roslaunch kinect2_bridge kinect2_bridge.launch sensor:=007098623642 base_name:=kinectOld publish_tf:=true

rosnode list gives me:

/kinectNew
/kinectNew_bridge
/kinectNew_points_xyzrgb_hd
/kinectNew_points_xyzrgb_qhd
/kinectNew_points_xyzrgb_sd
/kinectOld
/kinectOld_bridge
/kinectOld_points_xyzrgb_hd
/kinectOld_points_xyzrgb_qhd
/kinectOld_points_xyzrgb_sd

and rostopic list contains all the expected topics in the correct domains

The problem starts when I try to subscribe to a topic from each Kinect.
rostopic echo /kinectNew/sd/points publishes the pointcloud from the first Kinect.

Then I do: rostopic echo /kinectOld/sd/points in another terminal and I get the following error message:
[Kinect2Bridge::callbackStatus] could not start device!

The whole message:

Node: /kinectOld
Time: 08:40:54.612239331 (2017-03-01)
Severity: Error
Published Topics: /kinectOld/bond, /kinectOld/hd/camera_info, /kinectOld/hd/image_color, /kinectOld/hd/image_color/compressed, /kinectOld/hd/image_color_rect, /kinectOld/hd/image_color_rect/compressed, /kinectOld/hd/image_depth_rect, /kinectOld/hd/image_depth_rect/compressed, /kinectOld/hd/image_mono, /kinectOld/hd/image_mono/compressed, /kinectOld/hd/image_mono_rect, /kinectOld/hd/image_mono_rect/compressed, /kinectOld/hd/points, /kinectOld/qhd/camera_info, /kinectOld/qhd/image_color, /kinectOld/qhd/image_color/compressed, /kinectOld/qhd/image_color_rect, /kinectOld/qhd/image_color_rect/compressed, /kinectOld/qhd/image_depth_rect, /kinectOld/qhd/image_depth_rect/compressed, /kinectOld/qhd/image_mono, /kinectOld/qhd/image_mono/compressed, /kinectOld/qhd/image_mono_rect, /kinectOld/qhd/image_mono_rect/compressed, /kinectOld/qhd/points, /kinectOld/sd/camera_info, /kinectOld/sd/image_color_rect, /kinectOld/sd/image_color_rect/compressed, /kinectOld/sd/image_depth, /kinectOld/sd/image_depth/compressed, /kinectOld/sd/image_depth_rect, /kinectOld/sd/image_depth_rect/compressed, /kinectOld/sd/image_ir, /kinectOld/sd/image_ir/compressed, /kinectOld/sd/image_ir_rect, /kinectOld/sd/image_ir_rect/compressed, /kinectOld/sd/points, /rosout, /tf

[31m[Kinect2Bridge::callbackStatus] [31mcould not start device![0m

Location:
~/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:callbackStatus:790

----------------------------------------------------------------------------------------------------

I can do the same in the opposite order, but then the other Kinect wont start. 

Does anyone know what is going on here? Are there some parameters I have forgotten to change?

Thanks in advance.

wiedemeyer, 2017-03-02:
Hi,
the 2 sensors have to be on different USB controllers, because 1 controller gives up to 5 gbit/s bandwith and one Kinect2 reserves 3 gbit/s. So you can not use 2 Kinects on the same controller. This problem is described in several issues on the libfreenect2 GitHub page. Please look there if you need more information.
If that doesn't solve it, provide the full output of the 2 kinect2_bridge instances.

easypeet, 2017-03-10:
Thanks for the reply. I was aware of this and I was using two different USB controllers, but that did not make any difference. What I did to make it work for me was to buy an external USB 3 controller for the PCI slot.

closed 2017-03-10 by easypeet