code-iai/iai_kinect2 #377: Unable to compile anymore

soulslicer, 2017-03-03:
```
/usr/local/include/opencv2/flann/saving.h:113:63: error: exception handling disabled, use -fexceptions to enable
         throw FLANNException("Invalid index file, cannot read");
```

I am getting the following error when compiling

soulslicer, 2017-03-03:
This happens in the kinect registration package. However, when I added the following line to the kinect_registration CMakeLists.txt, it works

`add_definitions( -fexceptions )`

EXing, 2017-03-08:
@soulslicer it helps!!!!!

wiedemeyer, 2017-03-08:
Never seen this error. Could it be that you are not using the version that comes with Ubuntu?

EXing, 2017-03-09:
I don't know! I'm in ubuntu 16.04. before it, the project has already been built. then, it happens come with some other errors.

VasLem, 2017-03-09:
I confirm there was an installation error and that @soulslicer solution fixed it. I was forced to reinstall the whole iai_kinect2 after the latest ROS Kinetic update, as it rendered my previous installation buggy. The folowing error appeared that forced me to reinstall:
```
[ERROR] [1489029105.365911194]: Failed to load nodelet [/kinect2_bridge] of type [kinect2_bridge/kinect2_bridge_nodelet] even after refreshing the cache: Could not find library corresponding to plugin kinect2_bridge/kinect2_bridge_nodelet. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
```

soulslicer, 2017-03-09:
This is an issue with the new OpenCV update 3.2.0

zipeilu, 2017-03-28:
Same issue occurred to me after the the latest ROS update. Every node of iai_kinect2 didn't work and failed to locate the libopencv_improc3.so. I tried to recompile iai_kinect2 then this issue came out. After all, @soulslicer 's solution works really well! Thanks a lot!

TheTesla, 2017-03-31:
please look here: #397 
This fixed the problem for me

alansrobotlab, 2017-04-09:
I can confirm.  Saw the same problem with nvidia cuda + opencl install on 16.04 kinetic.

Inserted "add_definitions( -fexceptions )" and allowed catkin_make to complete:
```
if(OpenCL_FOUND)
  message(STATUS "OpenCL based depth registration enabled")
  set(EXPORTED_DEPENDENCIES OpenCL)
  add_definitions( -fexceptions )
```

Error seen during catkin_make without -fexceptions:
```
In file included from /opt/ros/kinetic/include/opencv-3.2.0-dev/opencv2/flann/flann_base.hpp:41:0,
                 from /opt/ros/kinetic/include/opencv-3.2.0-dev/opencv2/flann.hpp:48,
                 from /opt/ros/kinetic/include/opencv-3.2.0-dev/opencv2/opencv.hpp:62,
                 from /home/grey/inmoov_ros/src/iai_kinect2/kinect2_registration/include/kinect2_registration/kinect2_registration.h:24,
                 from /home/grey/inmoov_ros/src/iai_kinect2/kinect2_registration/src/kinect2_registration.cpp:18:
/opt/ros/kinetic/include/opencv-3.2.0-dev/opencv2/flann/saving.h: In function ‘cvflann::IndexHeader cvflann::load_header(FILE*)’:
/opt/ros/kinetic/include/opencv-3.2.0-dev/opencv2/flann/saving.h:113:63: error: exception handling disabled, use -fexceptions to enable
         throw FLANNException("Invalid index file, cannot read");
```

yoneken, 2017-09-18:
This issue is still exist on 16.04 kinetic.. #397 is also useful for me.

aadityacr7, 2017-09-20:
Its working. Thank you so much @soulslicer . 

Works on Ubuntu 16.04, ROS-Kinetic, Kinect V2

aaronsnoswell, 2018-01-22:
I saw the same issue on my machine - Ubuntu 16.04, ROS Kinetic. Making the changes suggested by @alansrobotlab fixed it for me. I suggest pulling #397.

christian-rauch, 2018-01-22:
Disabling the exception checking is enforced at
https://github.com/code-iai/iai_kinect2/blob/4d41e24c4728d917af928b8a4a6cf236849ba8d4/kinect2_registration/CMakeLists.txt#L66

Removing (or commenting) this line also solves the problem.