code-iai/iai_kinect2 #378: during the calibration, can the color and ir images  take at the same time?

EXing, 2017-03-03:
during the calibration, can the color and ir images  take at the same time?
for the every set, taking 100 images seems hard without tripod!

leoujz, 2017-03-08:
Can you detect checkerboard successfully on IR image?

EXing, 2017-03-08:
yes!

EXing, 2017-03-08:
@leoujz yes!!

wiedemeyer, 2017-03-08:
No, they can not be taken at the exact same time. You can get good results without a tripod too, but it may be harder, or less good.

closed 2017-03-11 by EXing

JimmyDaSilva, 2017-03-13:
@wiedemeyer This I don't understand. As we take both depth and rgb during the sync calibration step, couldn't we :
- grab rgb and depth images at the same time
- calibrate rgb
- calibrate depth
- rectify previous rgb and depth
- calibrate sync 

I am just curious.

wiedemeyer, 2017-03-13:
The problem is, that ir and color have different sizes. Ir has more height, while color is much wider. For a good calibration you also want samples from the non overlapping areas of images as well, that is why it is split up in 3 parts.

JimmyDaSilva, 2017-03-13:
Oh I see. We need to get to the corners of the RGB image at some point.
Thank you! It makes sense to me now.