code-iai/iai_kinect2 #382: cmake errors

jackj012, 2017-03-08:
I follow the installation step by step, and the libfreenect2 runs  correctly, but some errors happen when running cmake command. It seems to be something with opencv. my system is ubuntu 16.04 and i just install the system recently. so what should i do? thanks

`~/catkin_ws$ catkin_make -DCMAKE_BUILD_TYPE="Release"
Base path: /home/jjalex/catkin_ws
Source space: /home/jjalex/catkin_ws/src
Build space: /home/jjalex/catkin_ws/build
Devel space: /home/jjalex/catkin_ws/devel
Install space: /home/jjalex/catkin_ws/install
####
#### Running command: "cmake /home/jjalex/catkin_ws/src -DCMAKE_BUILD_TYPE=Release -DCATKIN_DEVEL_PREFIX=/home/jjalex/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/jjalex/catkin_ws/install -G Unix Makefiles" in "/home/jjalex/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/jjalex/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/jjalex/catkin_ws/devel;/opt/ros/kinetic
-- This workspace overlays: /home/jjalex/catkin_ws/devel;/opt/ros/kinetic
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/jjalex/catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.6
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 5 packages in topological order:
-- ~~  - iai_kinect2 (metapackage)
-- ~~  - kinect2_registration
-- ~~  - kinect2_bridge
-- ~~  - kinect2_calibration
-- ~~  - kinect2_viewer
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin metapackage: 'iai_kinect2'
-- ==> add_subdirectory(iai_kinect2/iai_kinect2)
-- +++ processing catkin package: 'kinect2_registration'
-- ==> add_subdirectory(iai_kinect2/kinect2_registration)
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Found OpenCV: /opt/ros/kinetic (found version "3.2.0") 
-- Try OpenMP C flag = [-fopenmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Success
-- Try OpenMP CXX flag = [-fopenmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Success
-- Found OpenMP: -fopenmp  
-- Looking for CL_VERSION_2_0
-- Looking for CL_VERSION_2_0 - found
-- Found OpenCL: /usr/lib/x86_64-linux-gnu/libOpenCL.so (found version "2.0") 
-- CPU based depth registration enabled
-- OpenCL based depth registration enabled
-- Performing Test OpenCL_C_WORKS
-- Performing Test OpenCL_C_WORKS - Success
-- Performing Test OpenCL_CXX_WORKS
-- Performing Test OpenCL_CXX_WORKS - Failed
CMake Warning at iai_kinect2/kinect2_registration/CMakeLists.txt:60 (message):
  Your libOpenCL.so is incompatible with CL/cl.h.  Install ocl-icd-opencl-dev
  to update libOpenCL.so?

-- +++ processing catkin package: 'kinect2_bridge'
-- ==> add_subdirectory(iai_kinect2/kinect2_bridge)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- +++ processing catkin package: 'kinect2_calibration'
-- ==> add_subdirectory(iai_kinect2/kinect2_calibration)
-- +++ processing catkin package: 'kinect2_viewer'
-- ==> add_subdirectory(iai_kinect2/kinect2_viewer)
-- Checking for module 'eigen3'
--   Found eigen3, version 3.2.92
[... 140 lines cut ...]
             };
             ^
In file included from /opt/ros/kinetic/include/opencv-3.2.0/opencv2/flann/flann_base.hpp:41:0,
                 from /opt/ros/kinetic/include/opencv-3.2.0/opencv2/flann.hpp:48,
                 from /opt/ros/kinetic/include/opencv-3.2.0/opencv2/opencv.hpp:62,
                 from /home/jjalex/catkin_ws/src/iai_kinect2/kinect2_registration/include/kinect2_registration/kinect2_registration.h:24,
                 from /home/jjalex/catkin_ws/src/iai_kinect2/kinect2_registration/src/depth_registration_opencl.h:22,
                 from /home/jjalex/catkin_ws/src/iai_kinect2/kinect2_registration/src/depth_registration_opencl.cpp:39:
/opt/ros/kinetic/include/opencv-3.2.0/opencv2/flann/saving.h: In function ‘cvflann::IndexHeader cvflann::load_header(FILE*)’:
/opt/ros/kinetic/include/opencv-3.2.0/opencv2/flann/saving.h:113:63: error: exception handling disabled, use -fexceptions to enable
         throw FLANNException("Invalid index file, cannot read");
                                                               ^
iai_kinect2/kinect2_registration/CMakeFiles/kinect2_registration.dir/build.make:86: recipe for target 'iai_kinect2/kinect2_registration/CMakeFiles/kinect2_registration.dir/src/depth_registration_cpu.cpp.o' failed
make[2]: *** [iai_kinect2/kinect2_registration/CMakeFiles/kinect2_registration.dir/src/depth_registration_cpu.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
iai_kinect2/kinect2_registration/CMakeFiles/kinect2_registration.dir/build.make:62: recipe for target 'iai_kinect2/kinect2_registration/CMakeFiles/kinect2_registration.dir/src/kinect2_registration.cpp.o' failed
make[2]: *** [iai_kinect2/kinect2_registration/CMakeFiles/kinect2_registration.dir/src/kinect2_registration.cpp.o] Error 1
iai_kinect2/kinect2_registration/CMakeFiles/kinect2_registration.dir/build.make:110: recipe for target 'iai_kinect2/kinect2_registration/CMakeFiles/kinect2_registration.dir/src/depth_registration_opencl.cpp.o' failed
make[2]: *** [iai_kinect2/kinect2_registration/CMakeFiles/kinect2_registration.dir/src/depth_registration_opencl.cpp.o] Error 1
CMakeFiles/Makefile2:906: recipe for target 'iai_kinect2/kinect2_registration/CMakeFiles/kinect2_registration.dir/all' failed
make[1]: *** [iai_kinect2/kinect2_registration/CMakeFiles/kinect2_registration.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
`

wiedemeyer, 2017-03-08:
Duplicated issue #377.
There is a solution in that thread.

closed 2017-03-08 by wiedemeyer

cshreyastech, 2019-06-07:
i have the issue.