code-iai/iai_kinect2 #385: why the output of cameraMatrixColor and cameraMatrixDepth in kinect_viewer is the same, and different from calibration data!!

EXing, 2017-03-12:
the calibration data is:
color:
cameraMatrix: !!opencv-matrix
   rows: 3
   cols: 3
   dt: d
   data: [ 1.0616386063746841e+03, 0., 9.5682520158547027e+02, 0.,
       1.0624871535307041e+03, 5.5898767450407252e+02, 0., 0., 1. ]
distortionCoefficients: !!opencv-matrix
   rows: 1
   cols: 5
   dt: d
   data: [ 5.0524359263624842e-02, -5.2138435623346392e-02,
       5.5681620754204010e-04, -6.1261672711284106e-05,
       5.9451195497605240e-03 ]
rotation: !!opencv-matrix
   rows: 3
   cols: 3
   dt: d
   data: [ 1., 0., 0., 0., 1., 0., 0., 0., 1. ]
projection: !!opencv-matrix
   rows: 4
   cols: 4
   dt: d
   data: [ 1.0616386063746841e+03, 0., 9.5682520158547027e+02, 0., 0.,
       1.0624871535307041e+03, 5.5898767450407252e+02, 0., 0., 0., 1.,
       0., 0., 0., 0., 1. ]

ir:
cameraMatrix: !!opencv-matrix
   rows: 3
   cols: 3
   dt: d
   data: [ 3.5965302518592574e+02, 0., 2.5270250727573617e+02, 0.,
       3.5928980195092879e+02, 2.0621558235538006e+02, 0., 0., 1. ]
distortionCoefficients: !!opencv-matrix
   rows: 1
   cols: 5
   dt: d
   data: [ 9.2681154376030916e-02, -2.8056650772859665e-01,
       -6.4386717052318152e-04, 3.4854477208117249e-04,
       1.1082825112036084e-01 ]
rotation: !!opencv-matrix
   rows: 3
   cols: 3
   dt: d
   data: [ 1., 0., 0., 0., 1., 0., 0., 0., 1. ]
projection: !!opencv-matrix
   rows: 4
   cols: 4
   dt: d
   data: [ 3.5965302518592574e+02, 0., 2.5270250727573617e+02, 0., 0.,
       3.5928980195092879e+02, 2.0621558235538006e+02, 0., 0., 0., 1.,
       0., 0., 0., 0., 1. ]

and the output of kinect2_viewer is:
depth = 
 [530.819303187342, 0, 478.4126007927351;
 0, 531.243576765352, 279.4938372520363;
 0, 0, 1]

color = 
 [530.819303187342, 0, 478.4126007927351;
 0, 531.243576765352, 279.4938372520363;
 0, 0, 1]

EXing, 2017-03-12:
I'm using orbslam2, which should choose?and is it affect Camera.bf(the IR projector baseline times fx)? because orbslam2 use them as parameters!

EXing, 2017-03-13:
@wiedemeyer please!

lvxingzhe1991, 2017-06-23:
Cly@ly-HP-Pavilion-Notebook:~/catkin_ws$ rosrun kinect2_calibration kinect2_calration chess5x7x0.03 record color
[ INFO] [main] Start settings:
       Mode: record
     Source: color
      Board: chess
 Dimensions: 5 x 7
 Field size: 0.03
Dist. model: 5 coefficients
Topic color: /kinect2/hd/image_mono
   Topic ir: /kinect2/sd/image_ir
Topic depth: /kinect2/sd/image_depth
       Path: ./

[ INFO] [main] starting recorder...
[ INFO] [Recorder::startRecord] Controls:
   [ESC, q] - Exit
 [SPACE, s] - Save current frame
        [l] - decrease min and max value for IR value range
        [h] - increase min and max value for IR value range
        [1] - decrease min value for IR value range
        [2] - increase min value for IR value range
        [3] - decrease max value for IR value range
        [4] - increase max value for IR value range
[ERROR] Tried to advertise a service that is already advertised in this node [/kinect2_calib_1498205553081554212/compressed/set_parameters]
[ERROR] Tried to advertise a service that is already advertised in this node [/kinect2_calib_1498205553081554212/compressed/set_parameters]
can you help me?