code-iai/iai_kinect2 #386: Can't compile Kinect2_Bridge

JoshuaSiy, 2017-03-14:
Hello I can't compile using catkin_make , using "catkin_make -DCMAKE_BUILD_TYPE="Release -Dfreenect2_DIR=path_to_freenect2/lib/cmake/freenect2". 

New to ROS, linux and posting issues at github. I hope i don't make any disturbances or making wrong formats.
 
Things i did: installed rtabmap and rtabmap_ros, 

Details that might be helpful:
standalone rtabmap works using libfreenect2 driver.  Freenect2's ./Protonect cl gl cpu and cuda works
I modified cmakelists.txt in rtabmap,rtabmapros, and cvsba  around  find package (OPENCV REQUIRED) to find PACKAGE (OPENCV 2.4.11 REQUIRED)
libfreenect2 cmake used [ cmake .. -Dfreenect2_DIR=$HOME/freenect2/lib/cmake/freenect2 -DENABLE_CXX11=ON ]

Hardware: Intel i7 Nvidia GeForce 1050 
Installed: Cuda 

Error Produced:  (catkin_make at 100% but fails )
/home/nerga/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp: In member function ‘bool Kinect2Bridge::initPipeline(const string&, int32_t)’:
/home/nerga/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:427:28: error: expected type-specifier
       packetPipeline = new libfreenect2::OpenCLKdePacketPipeline(device);
                            ^
/home/nerga/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:436:28: error: expected type-specifier
       packetPipeline = new libfreenect2::CudaKdePacketPipeline(device);
                            ^
/home/nerga/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp: In member function ‘bool Kinect2Bridge::initPipeline(const string&, int32_t)’:
/home/nerga/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:427:28: error: expected type-specifier
       packetPipeline = new libfreenect2::OpenCLKdePacketPipeline(device);
                            ^
/home/nerga/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:436:28: error: expected type-specifier
       packetPipeline = new libfreenect2::CudaKdePacketPipeline(device);
                            ^
iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge_nodelet.dir/build.make:62: recipe for target 'iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge_nodelet.dir/src/kinect2_bridge.cpp.o' failed
make[2]: *** [iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge_nodelet.dir/src/kinect2_bridge.cpp.o] Error 1
CMakeFiles/Makefile2:2296: recipe for target 'iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge_nodelet.dir/all' failed
make[1]: *** [iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge_nodelet.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....

JoshuaSiy, 2017-03-16:
Okay i managed to just install a new machine and performed trials with replicated errors it seems like my freenect2 library can't be detected. Tentatively, seems like the only difference this time i i used catkin_make -Dfreenect2_DIR=path_to_freenect2/lib/cmake/freenect2 instead of catkin_make -DCMAKE_BUILD_TYPE="Release"  -Dfreenect2_DIR=path_to_freenect2/lib/cmake/freenect2. I'm sorry if i failed to understand the problem or managed to give any proper solutions. iai_kinect2 now managed to compile completely on my side. Just new problems with rtabmap_ros.

closed 2017-03-16 by JoshuaSiy

aaramirezd, 2017-03-16:
@JoshuaSiy I have the same error, but I can't do ubuntu reinstallation, you have any other idea for fix this?
I did ROS upgrade and kinect2_bridge didn't work more.