code-iai/iai_kinect2 #387: problem withkinect2_calibration!!!

Domenico1992, 2017-03-14:
after capturing a budget of about 120 frames for both the IR calibration, color calibration and sync and respecting all instructions on the Web page, the output are these and when i launch the viewer after the calibration the points in the video are not aligned with the image, but are shifted by a long

for the color calibration:

%YAML:1.0

cameraMatrix: !!opencv-matrix
   
rows: 3
   
cols: 3
   
dt: d
   
data: [ 1.1004687793237374e+03, 0., 1.0069369709567793e+03, 0.,
       
1.0978066255792717e+03, 4.5910703830681450e+02, 0., 0., 1. ]

distortionCoefficients: !!opencv-matrix
   
rows: 1
   
cols: 5
   
dt: d
   
data: [ 5.1273018211442180e-02, -4.1309038575363652e-02,
       
-1.5403380499678277e-02, 1.3607453419991436e-02,
       
1.2777603635683029e-02 ]

rotation: !!opencv-matrix
   
rows: 3
   
cols: 3
   
dt: d
   
data: [ 1., 0., 0., 0., 1., 0., 0., 0., 1. ]

projection: !!opencv-matrix
   
rows: 4
   
cols: 4
   
dt: d
   
data: [ 1.1004687793237374e+03, 0., 1.0069369709567793e+03, 0., 0.,
      
 1.0978066255792717e+03, 4.5910703830681450e+02, 0., 0., 0., 1.,
      
 0., 0., 0., 0., 1. ]

for the depth calibration:

%YAML:1.0

depthShift: -2.2823243656717089e+01

for the ir calibration

%YAML:1.0

cameraMatrix: !!opencv-matrix
   
rows: 3
   
cols: 3
   
dt: d
   
data: [ 3.6491380051748291e+02, 0., 2.4959685356221763e+02, 0.,
       
3.6438630560775880e+02, 2.0772319014430403e+02, 0., 0., 1. ]

distortionCoefficients: !!opencv-matrix
   
rows: 1
   
cols: 5
   
dt: d
   
data: [ 1.0165096942146126e-01, -2.9299799708217533e-01,
       
-7.8572653956486820e-04, -6.7892402528351662e-04,
       
1.1590297474610570e-01 ]

rotation: !!opencv-matrix
   
rows: 3
   
cols: 3
   
dt: d
   
data: [ 1., 0., 0., 0., 1., 0., 0., 0., 1. ]

projection: !!opencv-matrix
   
rows: 4
   
cols: 4
   
dt: d
   
data: [ 3.6491380051748291e+02, 0., 2.4959685356221763e+02, 0., 0.,
       
3.6438630560775880e+02, 2.0772319014430403e+02, 0., 0., 0., 1.,
       
0., 0., 0., 0., 1. ]

for the pose calibration:

%YAML:1.0

rotation: !!opencv-matrix
   
rows: 3
   
cols: 3
   
dt: d
   data: [ 9.9894326499132058e-01, -3.1952017133869632e-03,
       
-4.5849144097694909e-02, 5.5494116217593434e-03,
       
9.9866726137486050e-01, 5.1311841602970365e-02,
       
4.5624087488219479e-02, -5.1512054356692150e-02,
       
9.9762966620726579e-01 ]

translation: !!opencv-matrix
   
rows: 3
   
cols: 1
   
dt: d
   
data: [ -6.4922819925538416e-02, 1.9525107596072039e-02,
       
3.7529080315411516e-02 ]

essential: !!opencv-matrix
   
rows: 3
   
cols: 3
   
dt: d
   
data: [ 6.8255090272380497e-04, -3.8484842264318329e-02,
       
1.7553140349080749e-02, 4.0451466438663368e-02,
       
-3.4642208107197110e-03, 6.3048254960314049e-02,
       
-1.9864758182739209e-02, -6.4773908118525639e-02,
       
-2.4360999807420910e-03 ]

fundamental: !!opencv-matrix
   
rows: 3
   
cols: 3
   
dt: d
   
data: [ -5.2237669512051990e-07, 2.9496242838306074e-05,
       
-1.0898902090344002e-02, -3.1033795693758629e-05,
       
2.6615488031678916e-06, -1.0457711038294761e-02,
       
3.1504366658227895e-02, 2.3710177483597092e-02, 1. ]

vacky11, 2017-03-28:
I am facing the same problem.
I repeated the calibration process twice and made sure that I am not missing anything.

Please help

lvxingzhe1991, 2017-06-23:
Cly@ly-HP-Pavilion-Notebook:~/catkin_ws$ rosrun kinect2_calibration kinect2_calration chess5x7x0.03 record color
[ INFO] [main] Start settings:
       Mode: record
     Source: color
      Board: chess
 Dimensions: 5 x 7
 Field size: 0.03
Dist. model: 5 coefficients
Topic color: /kinect2/hd/image_mono
   Topic ir: /kinect2/sd/image_ir
Topic depth: /kinect2/sd/image_depth
       Path: ./

[ INFO] [main] starting recorder...
[ INFO] [Recorder::startRecord] Controls:
   [ESC, q] - Exit
 [SPACE, s] - Save current frame
        [l] - decrease min and max value for IR value range
        [h] - increase min and max value for IR value range
        [1] - decrease min value for IR value range
        [2] - increase min value for IR value range
        [3] - decrease max value for IR value range
        [4] - increase max value for IR value range
[ERROR] Tried to advertise a service that is already advertised in this node [/kinect2_calib_1498205553081554212/compressed/set_parameters]
[ERROR] Tried to advertise a service that is already advertised in this node [/kinect2_calib_1498205553081554212/compressed/set_parameters]
Please help

slamidiot, 2018-11-09:
@lvxingzhe1991 ,that is ok. you  can proceed with the following steps.