code-iai/iai_kinect2 #388: Getting and Reading depth values

BHAASKARR, 2017-03-14:
Hi ,

I would like to see the depth values of picture (Distance data from camera to image).
so which of the following will give distance data from camera to image
/kinect2/sd/image_depth
/kinect2/sd/image_depth/compressed
/kinect2/sd/image_depth_rect
/kinect2/sd/image_depth_rect/compressed
/kinect2/sd/image_ir
/kinect2/sd/image_ir/compressed
/kinect2/sd/image_ir_rect
/kinect2/sd/image_ir_rect/compressed

and where can i see the source code for above files ?
what represents each value of below collected data row (EX: data:[73,73,42,0,76,235,2,0,128,0,0,11,......etc..]) ?
/ Below data is from the /kinect2/sd/image_depth_rect/compressed topic /

header:
seq: 2437
stamp:
secs: 1489174612
nsecs: 963329202
frame_id: kinect2_ir_optical_frame
format: 16UC1; tiff compressed
data: [73, 73, 42, 0, 76, 235, 2, 0, 128, 0, 0, 11, 80, 65, 45, 238, 241, 4, 3, 222, 240, 40, 100, 54, 29, 13, 7, 2, 29, 79, 240, 67, 225, 240, 7, 138, 195, 227, 80, 198, 56, 28, 114, 248, 142, 199, 227, 114, 56, 115, 8, 14, 3, 0, 1, 128, 3, 87, 195, 68, 14, 20, 0, 6, 223, 17, 183, 104, 28, 72, 0, 109, 191, 221, 207, 240, 112, 1, 254, 255, 16, 0, 39, 96, 16, 0, 124, 0, 5, 0, 57, 31, 228, 80, 5, 17, 240, 255, 12, 0, 4, 64, 6, 243, 254, 166, 19, 0, 74, 7, 64, 0, 64, 1, 210, 255, 89, 63, 221, 143, 240, 120, 1, 57, 11, 179, 2, 12, 143, 116, 224, 32, 34, 0, 125, 63, 203, 47, 113, 184, 36, 24, 0, 94, 189, 137, 192, 144, 224, 1, 214, 255, 114, 191, 233, 5, 240, .....................some values .......... 0, 238, 138, 2, 0, 209, 155, 2, 0, 242, 171, 2, 0, 75, 188, 2, 0, 169, 203, 2, 0, 39, 219, 2, 0]

Thanks in advance.
Bhaskar.

BHAASKARR, 2017-03-14:
@wiedemeyer 
Please help me on this.

BHAASKARR, 2017-04-05:
@mpatalberta
@SelvamArul

SelvamArul, 2017-04-06:
@BHAASKARR 
The topic ```image_depth_rect``` is the rectified depth and its encoding is ```16UC1```. (Do something like this: ```rostopic echo -n1 /kinect2/hd/image_depth_rect/encoding```) The values represent the distance from the sensor in ```mm```.  When you echo it, the values are not parsed as 16UC1. That's why you see values [0, 255]. You can access the 16UC1 values in your code after you subscribe to the image topic.

BHAASKARR, 2017-04-06:
Thank you @SelvamArul 
Can you give me detailed steps to get the distance data from sensor in mm ?

I observed below
"bhaskar@D125-40806Linux:~$ rostopic echo -n1 /kinect2/hd/image_depth_rect/encoding
16UC1
--- "
How to subscribe to the image topic to access 16UC1 values (distance data from sensor in mm) ?

BHAASKARR, 2017-04-12:
@SelvamArul  
Please can you suggest me solution

JimmyDaSilva, 2017-04-13:
For now the encoding is `16UC1`,  means uint16 and 1 channel because it's a grayscale image.
If you want to read the values they have to be not encoded, so converting the uints in floats is all you need.

```
void callback(sensor_msgs::ImageConstPtr depth_img_ros){
  cv_bridge::CvImageConstPtr depth_img_cv;
  cv::Mat depth_mat;
  // Get the ROS image to openCV
  depth_img_cv = cv_bridge::toCvShare (depth_img_ros, sensor_msgs::image_encodings::TYPE_16UC1);
  // Convert the uints to floats
  depth_img_cv->image.convertTo(depth_mat, CV_32F, 0.001);
}
```
The `0.001` is just a scaling factor (mm to m or the other way around) . It might be required depending on when you are using `openni`, `freenect`, `openni2` ... If not required change it to `1`.

closed 2017-05-23 by wiedemeyer