code-iai/iai_kinect2 #389: cam_calibration error ：Segmentation fault (core dumped)

YoungRainy, 2017-03-19:
when i calibrate my camera, after run the first code "rosrun kinect2_bridge kinect2_bridge _fps_limit:=2",i got the info like this ,can anyone tell me how to fix this problem?THANKS.

`yys@yys-Inspiron-7559:~$ rosrun kinect2_bridge kinect2_bridge _fps_limit:=2
[ INFO] [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: default
        fps_limit: 2
       calib_path: /home/yys/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opengl
     depth_device: -1
       reg_method: default
       reg_device: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 8 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:3 with serial 007139164947
[Info] [Freenect2Impl] found 1 devices
[ INFO] [Kinect2Bridge::initDevice] Kinect2 devices found: 
[ INFO] [Kinect2Bridge::initDevice]   0: 007139164947 (selected)
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[ INFO] [Kinect2Bridge::initDevice] starting kinect2
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
[ INFO] [Kinect2Bridge::initDevice] device serial: 007139164947
[ INFO] [Kinect2Bridge::initDevice] device firmware: 4.0.3916.0
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] canceling rgb transfers...
[Info] [Freenect2DeviceImpl] canceling depth transfers...
[Info] [Freenect2DeviceImpl] stopped
Segmentation fault (core dumped)
`

yvtheja, 2017-06-09:
Hello @YoungRainy, 
I am facing the same problem yet my Protonect works fine.
Did you resolve it?

YoungRainy, 2017-06-09:
@yvtheja  I fix the problem by changing the opencv version.I used Opencv3 and it doesn't work.Opencv2 works fine.

yvtheja, 2017-06-10:
@YoungRainy , Did you edit CMakeLists.txt files to change the version of OpenCV used? I changed the line

`find_package(OpenCV REQUIRED)`

to 

`find_package(OpenCV 2 REQUIRED)`
 
But, by doing this, I couldn't even compile the package.