code-iai/iai_kinect2 #392: How to make sure that the kinect2_bridge is actually using the calibration files

vacky11, 2017-03-28:
I have followed the entire process of calibrating kinect2 and finally have placed the 4 calibration files in kinect2_bridge/data/$serial.
But there is no difference in the pre-calibration and post calibration results 

How can I make sure that the new calibrated files are actually being used?

xuhangxx, 2017-04-23:
I am facing the same problem, I repeated the calibration process for many times and made sure that I am not missing anything. Who can help me!!!

wiedemeyer, 2017-05-23:
While stating up kinect2_bridge, it will tell you if it used the defaults or loaded a calibration.

closed 2017-05-23 by wiedemeyer

TrinhTUHH, 2017-08-21:
@vacky11  and @xuhangxx , have you found the solution? I also create a folder with serial name and place 4 calibrated files there, but it seems like the calibrated files are not loader. When I run kinect2_bridge it always said "sing sensor defaults for..."

vacky11, 2017-08-21:
@TrinhTUHH 
I dont have access to Kinect2 anymore. So I cant verify this. Sorry about that.